openwrt/packages #22240: Aircrack-ng PCRE2 branch needs testing out

trippleflux, 2023-09-28:
Compile-Tested : x86_64
Run-Tested : x86_64
Functionalities-Tested : NONE

My attempt to switch Aircrack-ng into pcre2 : 
```
root@router:~# ldd /usr/bin/aircrack-ng
        /lib/ld-musl-x86_64.so.1 (0x7f6460974000)
        libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0 (0x7f64608cf000)
        libaircrack-osdep-1.7.0.so => /usr/lib/libaircrack-osdep-1.7.0.so (0x7f64608bf000)
        libnl-3.so.200 => /usr/lib/libnl-3.so.200 (0x7f646089d000)
        libnl-genl-3.so.200 => /usr/lib/libnl-genl-3.so.200 (0x7f6460895000)
        libsqlite3.so.0 => /usr/lib/libsqlite3.so.0 (0x7f64606bb000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7f646069c000)
        libssl.so.3 => /usr/lib/libssl.so.3 (0x7f64605f3000)
        libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x7f64601bf000)
        libhwloc.so.15 => /usr/lib/libhwloc.so.15 (0x7f6460155000)
        libpciaccess.so.0 => /usr/lib/libpciaccess.so.0 (0x7f6460149000)
        libbsd.so.0 => /usr/lib/libbsd.so.0 (0x7f6460129000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f645ff4f000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f645ff36000)
        libc.so => /lib/ld-musl-x86_64.so.1 (0x7f6460974000)
```

Test command, as I am yet not that knowledgeable enough to use it.
```
root@router:~# aircrack-ng -u
Vendor          = Intel
Model           = 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
Features        = MMX,SSE,SSE2,SSE3,SSSE3,SSE4.1,SSE4.2,AES-NI,AVX,AVX2,AVX512F
CPU frequency   = 776 MHz (Max: 4700 MHz)
Hyper-Threading = Yes
Logical CPUs    = 8
Threads per core= 2
CPU cores       = 4
SIMD size       = 16 (512 bit)
SIMD size in use= 16 (512 bit)
```

This is a vanilla Aircrack-ng master snapshot from that I have gotten from the Aircrack-ng github at current time.

I am just curious about pentesting apps such as this and currently looking around to about getting mitmproxy working in OpenWrt (but seems beyond my current abilities).

Anyway here is the Aircrack-ng package files :  [Aircrack-ng.zip](https://github.com/openwrt/packages/files/12752182/aircrack-ng.zip)

brada4, 2023-09-28:
You could disable pcre support for essid wildcarding until there is a new release with pcre2 support.

trippleflux, 2023-09-28:
@brada4 
Hmm, have you tested compiling Aircrack-NG with my version of OpenWrt Aircrack-ng package file that is attached above?.

The PCRE and PCRE2 lib differentiation seems coming firstly at this [m4 file](https://github.com/aircrack-ng/aircrack-ng/blob/master/build/m4/aircrack_ng_pcre.m4).

My Aircrack-NG patch that is attached in the archive above is simply disable automatic detection of PCRE, because somehow Aircrack-NG if detecting both libpcre and libpcre2 is available in any build system (such as OpenWrt), it will automatically pick libpcre2 anyway but somehow at the end of it OpenWrt still complaining about missing dependency libpcre.so, so that's why I have removed libpcre detection and linkage with patch attached above.

My preliminary testing with my Aircrack-NG binaries built from simple test seems finely working.

My Patch : 
```
diff -Naur a/build/m4/aircrack_ng_pcre.m4 b/build/m4/aircrack_ng_pcre.m4
--- a/build/m4/aircrack_ng_pcre.m4	2023-09-04 22:11:22.000000000 +0700
+++ b/build/m4/aircrack_ng_pcre.m4	2023-09-28 17:03:49.538867200 +0700
@@ -38,22 +38,6 @@
 dnl program, then also delete it here.
 
 AC_DEFUN([AIRCRACK_NG_PCRE], [
-AC_ARG_ENABLE(static-pcre,
-    AS_HELP_STRING([--enable-static-pcre],
-		[Enable statically linked PCRE libpcre.]),
-    [static_pcre=$enableval], [static_pcre=no])
-
-if test "x$static_pcre" != "xno"; then
-	AC_REQUIRE([AX_EXT_HAVE_STATIC_LIB_DETECT])
-	AX_EXT_HAVE_STATIC_LIB(PCRE, ${DEFAULT_STATIC_LIB_SEARCH_PATHS}, pcre libpcre, pcre_version)
-	if test "x$PCRE_FOUND" = xyes; then
-		HAVE_PCRE=yes
-	else
-		HAVE_PCRE=no
-	fi
-else
-	PKG_CHECK_MODULES(PCRE, libpcre, HAVE_PCRE=yes, HAVE_PCRE=no)
-fi
 
 AC_ARG_ENABLE(static-pcre2,
     AS_HELP_STRING([--enable-static-pcre2],
@@ -72,12 +56,7 @@
 	PKG_CHECK_MODULES(PCRE2, libpcre2-8, HAVE_PCRE2=yes, HAVE_PCRE2=no)
 fi
 
-if test "x$HAVE_PCRE" = "xyes" && test "x$HAVE_PCRE2" = "xyes"; then
-    AC_DEFINE([HAVE_PCRE2], [1], [Define this if you have libpcre2-8 on your system])
-    PCRE2_NOTE="(Pcre and Pcre2 found, using Pcre2)"
-elif test "x$HAVE_PCRE" = "xyes"; then
-    AC_DEFINE([HAVE_PCRE], [1], [Define this if you have libpcre on your system])
-elif test "x$HAVE_PCRE2" = "xyes"; then
+if test "x$HAVE_PCRE2" = "xyes"; then
     AC_DEFINE([HAVE_PCRE2], [1], [Define this if you have libpcre2-8 on your system])
 fi
 ])
\ No newline at end of file
```

brada4, 2023-09-28:
It builds using --no-pcre just fine until v1.8 is released upstream?

trippleflux, 2023-09-28:
Why removing regex features for the case of simplicity of building?,  I am just trying to help the porting into using PCRE2 lib  issue.

The version numbering is only the most probable the possibility of the next version.

Regarding PCRE and PCRE2 code differences in Aircrack-NG seems not much, you could take a look at it in [here ](https://github.com/search?q=repo%3Aaircrack-ng%2Faircrack-ng%20HAVE_PCRE2&type=code). So, I think my patch for that 
aircrack_ng_pcre.m4 should be enough?,  or perhaps you could try the regex / wildcarding functionalities from package?.

So I think until the next stable release of Aircrack-NG comes out, probably this issue is quite useful who love to get up using Aircrack-NG.

brada4, 2023-09-28:
Why dont you bring that concern to upstream? 
Really it is just regex match for essid-s, nothing else, totally optional, not to say on useless side.

brada4, 2023-09-29:
Say current release is v23, it will have pcre1 and aircrack-ng 1.7 with it. Now you remove pcre1 dependency for global good while waiting for aircrack-ng 1.8. Not many users nagging you for that in the meantime. Next step is introducing v1.8 and restoring pcre function in a year or two. In the meantime support burden is minimal and even if you forget to maintain it anyone can  upgrade to v1.8, just leave comment in makefile so they know pcre2 function has to be restored. Indeed snapshot building without major issues means it will be easy ride ahead.

Regarding python modules - you can have pip on router and do the module installation without opkg. If it needs huge gcc installed you may elect to propose small set of packages supporting eg mitmproxy.

trippleflux, 2023-09-29:
Hmm, perhaps my Aircrack-ng above would be better as 'bump' up or probably a helpful base pointer for those going forward really maintaining it or basing it as a Pull request. 

> Regarding python modules - you can have pip on router and do the module installation without opkg. If it needs huge gcc installed you may elect to propose small set of packages supporting eg mitmproxy.

Regarding mitmproxy, it's more about my curiosity on learning more about OpenWrt build system. previous user in forum seems has tried it, from the looks of it looks like too many C/C++ libraries dependencies to manually installed inside a working router.

trippleflux, 2023-09-30:
Closing this one, as it's already filled with a proper PR.

Ansuel, 2023-09-30:
@trippleflux would be good if you guys can test the pr :D

Ansuel, 2023-09-30:
@trippleflux also if you want a clue of why you had the error of linking... check the patch 103 in my pr ;) I bet you are curious why the reason was sneaky :D

trippleflux, 2023-10-01:
@Ansuel reopened hehe, hmm I didn't have a linking error about Aircrack-ng before.

It was about another [mitmproxy ](https://github.com/mitmproxy/mitmproxy)not properly run on my x86-64 OpenWrt router, it was built-in or embedded into the OpenWrt x86-64 image, too much python-fu hehe.