openwrt/packages #22245: strongswan: swanctl init script incorrectly rejects esp proposal combination w/ hash_algo, and doesn't load connections

aleks-mariusz, 2023-09-29:
Maintainer: @pprindeville @Thermi 
Environment: OpenWrt master r24041-74e7f8ebbd on WRT32x 

Description:

The `/var/swanctl/swanctl.conf` file that the init script generates, thinks that if you specify an Authenticated Encryption (AEAD) Algorithms (in my case, aes256gcm16) then you canNOT also specify a hash_algorithm as well.  This seems to not be true for IKE proposals since the connection block then gets skipped being loaded by swanctl (details below):

So for example, the following excerpt from your `/etc/config/ipsec` file will cause the error:

```
config 'crypto_proposal' 'ike_proposal'
        option 'encryption_algorithm' 'aes256gcm16'
        option 'hash_algorithm' 'prfsha512' # <-- init script thinks this shouldn't be here for IKE proposal
        option 'dh_group' 'ecp384'
```

Running `/etc/init.d/swanctl restart` will cause the following warning: `ERROR: Can't have prfsha512 with aes256gcm16`

However, this seems is just a warning, as it's merely removing the hash_algorithm but still generates the config (similarly you could just remove/comment out the `hash_algorithm` for the `crypto_proposal` for the `config remote` block.  However swanctl, whiel having loaded, will not have loaded the offending parent block (the connection object).

From my understanding, the `config-remote` block this proposal is referenced from, is the one that defines the overall IKE parameters served by charon in user-space.

Then this leads to an invalid `swanctl.conf` file being generated in `/var/swanctl`

The contents of the generated file:
```
# generated by /etc/init.d/swanctl
# config for all_remote_ikev2
connections {
  remote_block_name {
[...snip...]
    version = 2
    proposals = aes256gcm16-ecp384
    dpd_delay = 300s
[...snip...]
} 
[...snipping out other authorities/pools blocks, as irrelevant...]   
```

What is seen above is the proposal that are relevant for the IKE first-phase (not the inner children{} block which deals with the ESP proposal).

And as a result, this keeps swanctl from loading the connections-block properly:

```
loaded authority 'all_remote_ikev2'
successfully loaded 1 authorities, 0 unloaded
loaded pool 'ipv4pool'
loaded pool 'ipv6pool'
successfully loaded 2 pools, 0 unloaded
loading connection 'all_remote_ikev2' failed: invalid value for: proposals, config discarded
loaded 0 of 1 connections, 1 failed to load, 0 unloaded
```

If you instead manually edit `/var/swanctl/swanctl.conf` and set the outer IKE (non-children ESP) proposal to the desired: `aes256gcm16-prfsha512-ecp384`, you will instead get swanctl to load correctly:

```
loaded authority 'all_remote_ikev2'
successfully loaded 1 authorities, 0 unloaded
loaded pool 'ipv4pool'
loaded pool 'ipv6pool'
successfully loaded 2 pools, 0 unloaded
loaded connection 'all_remote_ikev2'
successfully loaded 1 connections, 0 unloaded
```

I believe the fix is as simple as getting rid of [these few lines](https://github.com/openwrt/packages/blob/7f8aea6d8682ff3b9a322c588726d150d7317c8f/net/strongswan/files/swanctl.init#L202-L206) (in the config_ike_proposal() function).  I can open a PR tmrw if it's as simple as that, but wanted to get some feedback first and run it by you way-smarter folks.

btw, I also see the following syslog messages:

```
Fri Sep 29 01:53:48 2023 daemon.info ipsec: 10[CFG] a PRF algorithm is mandatory in IKE proposals
```

however my brain is a bit too fried now (2am here) having spent hours on this to figure out if this is related (and a confirmation of all the above) or not.

aleks-mariusz, 2023-09-29:
BTW i have some other ideas for enhancements to expand the settings swanctl.conf supports, that the swanctl init script currently does not know how to handle.. should i open separate issues or add them to this issue since it's about swanctl init script anyway? 

I've more details listed in [this forum post](https://forum.openwrt.org/t/confusion-regarding-setting-up-ikev2-vpn-service-with-strongswan-using-ipsec-and-swanctl/169587/9#:~:text=where%20the%20gaps%20in%20existing%20support%20still%20might%20be).

pprindeville, 2023-10-23:
I'll look at any PR's submitted, but I ask that they be broken up logically so that each one addresses a single individual issue.

kevinoid, 2024-12-01:
I started working on a fix for this, before realizing that it is already supported:  The [PRF](https://docs.strongswan.org/docs/latest/config/IKEv2CipherSuites.html#_pseudo_random_functions) algorithm must be configured using `prf_algorithm` rather than `hash_algorithm`.  Configuring

```uci
config 'crypto_proposal' 'ike_proposal'
        option 'encryption_algorithm' 'aes256gcm16'
        option 'prf_algorithm' 'prfsha512'
        option 'dh_group' 'ecp384'
```

Generates the expected `proposals = aes256gcm16-prfsha512-ecp384` in `/var/swanctl/swanctl.conf` for me.