openwrt/packages #22278: openvpn-openssl: $foreign_option_{n} are not set anymore

IeP4nieF, 2023-10-03:
Maintainer: @\<Magnus Kroken> 
Environment: MediaTek MT7621 ver:1 eco:3, Netgear WAX202, OpenWrt 22.03.5

Description: The environment variables `$foreign_option_{n}` are not set up for use in `up`-, `down`- or hotplug-scripts anymore. I used this to set the pushed DNS-Server. Here is the pushed control message I get from the openvpn-server:

```
PUSH: Received control message: 'PUSH_REPLY,topology p2p,comp-lzo no,route remote_host 255.255.255.255 net_gateway,ping 8,ping-restart 24,route-gateway <ip>,route 0.0.0.0 128.0.0.0,route 128.0.0.0 128.0.0.0,dhcp-option DNS <ip>,dhcp-option DNS <ip>,tun-ipv6,route-ipv6 <ip>,ifconfig-ipv6 <ip>,ifconfig <ip> <ip>,peer-id 16,cipher AES-128-GCM'
```
I checked with a simple `up`-script and also with `/etc/hotplug.d/openvpn/00-resolv`:

    env > /tmp/resolv.tmp

Lot's of environment variables but no `$foreign_option_{n}`.

With an openwrt-version before I could use `$foreign_option_1` and so on to get the IP of the DNS-Server.

egc112, 2023-10-07:
Hmm I did checked it some time ago but for 23.05 snapshot and then it still was there as environment variable.

I am not at home for some week so cannot check it at the moment, that said I use a slightly different approach to get the pushed DNS servers and set those to be used, see: https://github.com/egc112/OpenWRT-egc-add-on/tree/main/stop-dns-leak

egc112, 2023-10-09:
I just made a new build Snapshot 23.05.
root@DL-WRX36:~# ubus call system board
```
{
        "kernel": "5.15.132",
        "hostname": "DL-WRX36",
        "system": "ARMv8 Processor rev 4",
        "model": "Dynalink DL-WRX36",
        "board_name": "dynalink,dl-wrx36",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05-SNAPSHOT",
                "revision": "r23494-e26947993f",
                "target": "ipq807x/generic",
                "description": "OpenWrt 23.05-SNAPSHOT r23494-e26947993f"
        }
}
```

My VPN provider (KeepSolid) pushes a DNS server which can be seen in the environment variables, so I cannot confirm your findings.
```
route_vpn_gateway=10.200.0.69
daemon_log_redirect=0
USER=root
X509_1_emailAddress=support@simplexsolutionsinc.com
script_type=up
proto_1=udp4
ACTION=up
daemon=0
SHLVL=4
foreign_option_1=dhcp-option DNS 10.200.0.1     <<-- FOREIGN OPTION
dev_type=tun
route_network_1=10.200.0.1
user_down=/etc/openvpn/ovpn-up-update-resolv-5
remote_1=ro.vpnunlimitedapp.com
dev=tun1
xormethod_1=0
X509_0_CN=openvpn2.vpnunlimitedapp.com
X509_1_name=server.vpnunlimitedapp.com
remote_port_1=1194
X509_1_CN=server.vpnunlimitedapp.com
X509_1_C=US
INSTANCE=ks_ro
HOTPLUG_TYPE=openvpn
tls_digest_sha256_0=f2:24:b2:97:1b:80:c6:62:91:98:8c:99:5a:9a:ee:3e:97:ea:d5:5d:90:d8:e8:2b:a9:08:bd:82:a6:53:fa:8f
daemon_start_time=1696854594
script_context=init
ifconfig_local=10.200.0.70
common_name=openvpn2.vpnunlimitedapp.com
tls_digest_sha256_1=3a:1f:b1:5b:49:d3:e9:22:1f:e2:fc:46:c2:16:2a:df:9b:be:d4:8d:b5:d4:d8:13:14:98:43:74:7b:48:b8:77
LOGNAME=root
DEVICENAME=
verb=5
X509_1_L=New York
PATH=/usr/sbin:/usr/bin:/sbin:/bin
link_mtu=1553
route_gateway_1=10.200.0.69
trusted_ip=185.144.83.11
tls_serial_hex_0=21:11:b8:d1:a9:5a:18:a0:ac:2f:2f:1d:df:63:83:ac
X509_1_O=Simplex Solutions Inc.
tun_mtu=1500
route_netmask_1=255.255.255.255
tls_serial_hex_1=ab:15:6c:70:5d:6a:45:ee
trusted_port=1194
tls_id_0=CN=openvpn2.vpnunlimitedapp.com
tls_id_1=C=US, ST=NY, L=New York, O=Simplex Solutions Inc., OU=Vpn Unlimited, CN=server.vpnunlimitedapp.com, name=server.vpnunlimitedapp.com, emailAddress=support@simplexsolutionsinc.com
route_net_gateway=192.168.21.1
ifconfig_remote=10.200.0.69
daemon_pid=13488
route_metric_1=1
X509_1_OU=Vpn Unlimited
xormask_1=
user_up=/etc/openvpn/ovpn-up-update-resolv-5
config=ks_ro.ovpn
untrusted_ip=185.144.83.11
tls_serial_0=43956541481962239548369675980772705196
tls_digest_0=45:c7:70:52:ee:03:3f:ef:e8:10:4f:88:ee:07:2d:8c:51:4e:e1:5b
tls_serial_1=12327878784855983598
tls_digest_1=87:3b:fd:90:a2:cc:73:17:6a:23:5f:1b:4f:2a:4b:4b:3f:54:92:6b
X509_1_ST=NY
untrusted_port=1194
PWD=/etc/openvpn
xormasklen_1=0
```
Maybe post in the forum so it can be discussed there?

```