openwrt/packages #22302: Both 23.03 and master have recursive dependencies

sjkhsl, 2023-10-04:
make menuconfig

```
Collecting package info: done
tmp/.config-package.in:34293:error: recursive dependency detected!
tmp/.config-package.in:34293:   symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
feeds/packages/lang/python/pymysql/Config.in:4: symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

ptpt52, 2023-10-05:
same issue on master HEAD.

BKPepe, 2023-10-05:
Next time, please use specific subject instead of vague subject.

Duplicate of https://github.com/openwrt/packages/issues/22250 , closing.

closed 2023-10-05 by BKPepe