openwrt/packages #22304: Error build v17.01.7 kmod-jool

EvgeniyBlinov, 2023-10-04:
Same error https://github.com/openwrt/packages/issues/10590#issuecomment-555783856

Build env https://github.com/mwarning/docker-openwrt-build-env 

```
git status |head -1
HEAD detached at v17.01.7
```

```
grep jool ./.config
CONFIG_PACKAGE_kmod-jool=m
```

```
62cf3bb8b86cc7756ce9d586bd55638a  ./net/jool/patches/010-Add-support-for-kernel-4.17.patch
```

Error
```
make -C "/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/linux-4.4.182" KERNEL_DIR="/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/linux-4.4.182" ARCH="mips" CROSS_COMPILE="mips-openwrt-linux-musl-" SUBDIRS="/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful" V="s" modules
make[4]: Entering directory '/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/linux-4.4.182'
  CC [M]  /home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.o
/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c: In function 'inet6_hash_frag':
/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:96:14: error: 'INETFRAGS_HASHSZ' undeclared (first use in this function)
  return c & (INETFRAGS_HASHSZ - 1);
              ^
/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:96:14: note: each undeclared identifier is reported only once for each function it appears in
/home/user/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:97:1: warning: control reaches end of non-void function [-Wreturn-type]
```

BKPepe, 2023-10-05:
I am so sorry, but LEDE 17.01 is really old and already end-of-life for 4 years. You should not be using that.
Switch to the latest stable release, which is OpenWrt 22.03.

closed 2023-10-05 by BKPepe

EvgeniyBlinov, 2023-10-05:
> I am so sorry, but LEDE 17.01 is really old and already end-of-life for 4 years. You should not be using that. Switch to the latest stable release, which is OpenWrt 22.03.
Is there some manual how to build OpenWrt 22.03 for tp-wr841n?

BKPepe, 2023-10-05:
https://openwrt.org/toh/tp-link/tl-wr841nd