openwrt/packages #22307: luajit2 and luajit conflict

sjkhsl, 2023-10-05:
make V=s -j1

```
Collected errors:
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so
        But that file is already provided by package  * luajit
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so.2
        But that file is already provided by package  * luajit
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so.2.1.0
        But that file is already provided by package  * luajit
 * opkg_install_cmd: Cannot install package luajit2.
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so
        But that file is already provided by package  * luajit
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so.2
        But that file is already provided by package  * luajit
 * check_data_file_clashes: Package luajit2 wants to install file /home/sjkhsl/openwrt-main/build_dir/target-x86_64_musl/root-x86/usr/lib/libluajit-5.1.so.2.1.0
        But that file is already provided by package  * luajit
 * opkg_install_cmd: Cannot install package nginx-mod-lua-resty-lrucache.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/sjkhsl/openwrt-main'
make[1]: *** [package/Makefile:115: /home/sjkhsl/openwrt-main/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sjkhsl/openwrt-main'
make: *** [/home/sjkhsl/openwrt-main/include/toplevel.mk:232: world] Error 2
```

sjkhsl, 2023-10-05:
No error was reported after deleting the lang/luajit package. It is being compiled to see if it can be compiled successfully.

dosmakov, 2025-12-19:
After installation nginx-mod-lua (1.26.1-r2) and luajit2 (2.1.2025.05.29-r1) on [openwrt-24.10 branch (25.333.84481~f8d85b4)](https://github.com/openwrt/luci) / [OpenWrt 24.10.2 (r28739-d9340319c6)](https://openwrt.org/), the following lua scrip has failed with error.

lua script:

local args = ngx.req.get_uri_args();
local test_arg = args["test"]
ngx.say(test_arg)

error:

Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: 2025/12/19 22:27:08 [error] 26457#0: *8 lua entry thread aborted: runtime error: /usr/share/lua/resty/core/request.lua:205: no request found
Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: stack traceback:
Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: coroutine 0:
Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: 	[C]: in function 'error'
Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: 	/usr/share/lua/resty/core/request.lua:205: in function 'get_uri_args'
Fri Dec 19 22:27:08 2025 daemon.err nginx[26430]: 	/etc/nginx/lua/test.lua:1: in function </etc/nginx/lua/test.lua:1>, client: ***

Fixed by force installing luajit (2.1.0-beta3-r7):

/ # opkg --force-overwrite install luajit
Installing luajit (2.1.0-beta3-r7) to root...
Downloading https://downloads.openwrt.org/releases/24.10.2/packages/aarch64_cortex-a53/packages/luajit_2.1.0-beta3-r7_aarch64_cortex-a53.ipk
Configuring luajit.
/ # service nginx restart