openwrt/packages #22372: packages/lang/Python/README.md is out-of-date

cj-obrien, 2023-10-11:
Maintainer: @jefferyto 

The "Host-side Python packages for build" [section](https://github.com/openwrt/packages/tree/master/lang/python#host-side-python-packages-for-build) of `packages/lang/Python/README.md` contains the following language: 

> These can be installed via pip and ideally they should only be installed like this, because it's a bit simpler than running them through the OpenWrt build system. 

According to these instructions, installing a host Python package relies upon on a file specifying the requirements for that package being present in `host-pip-requirements` directory; however, the contents of that directory were removed in 3ee4e7297cf07b644bac3dfafc508da5f31bf63d. The new python-setuptools-scm [Makefile](https://github.com/openwrt/packages/blob/b158d62460e4291bb84411dc74de7445ef7b2020/lang/python/python-setuptools-scm/Makefile) does not seem to use pip at all, but rather downloads source tarballs from Pypi. 

I have two questions:

1. Is the new approach to installing host Python packages to write a Makefile in the style of python-setuptools-scm?
2. Assuming the answer to the question above is yes, if I attempt to write a Makefile for a new host Python package, will I also be required to write a Makefile for each Python package that package depends upon?

jefferyto, 2023-10-12:
> According to these instructions, installing a host Python package relies upon on a file specifying the requirements for that package being present in `host-pip-requirements` directory; however, the contents of that directory were removed in https://github.com/openwrt/packages/commit/3ee4e7297cf07b644bac3dfafc508da5f31bf63d. The new python-setuptools-scm [Makefile](https://github.com/openwrt/packages/blob/b158d62460e4291bb84411dc74de7445ef7b2020/lang/python/python-setuptools-scm/Makefile) does not seem to use pip at all, but rather downloads source tarballs from Pypi.

The readme is quite out of date at this point - hope to have time to update it after 23.05 is done.

> Is the new approach to installing host Python packages to write a Makefile in the style of python-setuptools-scm?

Yes.

> Assuming the answer to the question above is yes, if I attempt to write a Makefile for a new host Python package, will I also be required to write a Makefile for each Python package that package depends upon?

Yes - what Python package are you considering to add as a host package?

cj-obrien, 2023-10-16:
I was trying to figure out how to add the maturin build backend...
...and then I saw this :) 

https://github.com/openwrt/packages/commit/6640ea8cc4d6c35ca587ffe573ca464208c1e0d6