openwrt/packages #22373: stress-ng 0.17 fails to build on musl-1.2.4

oskarirauta, 2023-10-12:
Maintainer: Alexandru Ardelean <alex@shruggie.ro> \ @commodo 
Environment: x86_64, latest_git, musl 1.2.4

Description:

Build failure with musl-1.2.4

```
CC stress-vma.c
CC stress-vnni.c
stress-vnni.c: In function 'stress_vnni_vpdpwssd256':
stress-vnni.c:320:19: warning: AVX vector return without AVX enabled changes the ABI [-Wpsabi]
  320 |                 a = _mm256_loadu_si256((void *)&a_init[i]);
      |                 ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
stress-vnni.c: In function 'stress_vnni_vpdpwssd512':
stress-vnni.c:295:19: warning: AVX512F vector return without AVX512F enabled changes the ABI [-Wpsabi]
  295 |                 a = _mm512_loadu_si512((void *)&a_init[i]);
      |                 ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/immintrin.h:53,
                 from /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/x86intrin.h:32,
                 from core-builtin.h:23,
                 from stress-vnni.c:22:
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/avx512fintrin.h: In function 'stress_vnni_vpaddb512':
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/avx512fintrin.h:6532:1: error: inlining failed in call to 'always_inline' '_mm512_storeu_si512': target specific option mismatch
 6532 | _mm512_storeu_si512 (void *__P, __m512i __A)
      | ^~~~~~~~~~~~~~~~~~~
stress-vnni.c:122:17: note: called from here
  122 |                 _mm512_storeu_si512((void *)&result[i], r);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/immintrin.h:63:
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/avx512bwintrin.h:956:1: error: inlining failed in call to 'always_inline' '_mm512_add_epi8': target specific option mismatch
  956 | _mm512_add_epi8 (__m512i __A, __m512i __B)
      | ^~~~~~~~~~~~~~~
stress-vnni.c:121:21: note: called from here
  121 |                 r = _mm512_add_epi8(a, b);
      |                     ^~~~~~~~~~~~~~~~~~~~~
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/avx512fintrin.h:6499:1: error: inlining failed in call to 'always_inline' '_mm512_loadu_si512': target specific option mismatch
 6499 | _mm512_loadu_si512 (void const *__P)
      | ^~~~~~~~~~~~~~~~~~
stress-vnni.c:120:21: note: called from here
  120 |                 b = _mm512_loadu_si512((void *)&b_init[i]);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.2.0/include/avx512fintrin.h:6499:1: error: inlining failed in call to 'always_inline' '_mm512_loadu_si512': target specific option mismatch
 6499 | _mm512_loadu_si512 (void const *__P)
      | ^~~~~~~~~~~~~~~~~~
stress-vnni.c:119:21: note: called from here
  119 |                 a = _mm512_loadu_si512((void *)&a_init[i]);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [Makefile:621: stress-vnni.o] Error 1
make[4]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/stress-ng-0.17.00'
make[3]: *** [Makefile:50: /usr/src/openwrt/build_dir/target-x86_64_musl/stress-ng-0.17.00/.built] Error 2
make[3]: Leaving directory '/usr/src/openwrt/feeds/packages/utils/stress-ng'
time: package/feeds/packages/stress-ng/compile#160.99#19.54#0.00
    ERROR: package/feeds/packages/stress-ng failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/stress-ng/compile] Error 1
make[2]: Leaving directory '/usr/src/openwrt'
make[1]: *** [package/Makefile:114: /usr/src/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:232: world] Error 2
```

commodo, 2023-10-12:
oh; will check;
must be some stuff added for x86_64;
i tested on plain x86

commodo, 2023-10-14:
It loosk to be a GCC 13 compiler issue.
This builds fine with GCC 12 (the current default).

@ColinIanKing  have you seen this with GCC 13?

For now, I will just disable the `immintrin.h` header check.

commodo, 2023-10-14:
i haven't had time to check;
but this may also happen with older versions of stress-ng and GCC 13;
the AVX intrinsics on x86_64 seem to behave differently on our GCC 13;

one idea would be to check the build flags of the GCC 13 compiler;
i'd guess, this issue may pop-up in other libs (maybe numpy too) with GCC 13

ColinIanKing, 2023-10-14:
Maybe I can add some toolchain detection to avoid this. What is the output from:

gcc -E -dM -xc /dev/null

oskarirauta, 2023-10-14:
Issue didn't exist on 0.15.

output:
```
#define __DBL_MIN_EXP__ (-1021)
#define __UINT_LEAST16_MAX__ 0xffff
#define __FLT16_HAS_QUIET_NAN__ 1
#define __ATOMIC_ACQUIRE 2
#define __FLT128_MAX_10_EXP__ 4932
#define __FLT_MIN__ 1.17549435082228750796873653722224568e-38F
#define __GCC_IEC_559_COMPLEX 2
#define __UINT_LEAST8_TYPE__ unsigned char
#define __SIZEOF_FLOAT80__ 16
#define __BFLT16_DENORM_MIN__ 9.18354961579912115600575419704879436e-41BF16
#define __INTMAX_C(c) c ## L
#define __CHAR_BIT__ 8
#define __UINT8_MAX__ 0xff
#define __SCHAR_WIDTH__ 8
#define __WINT_MAX__ 0xffffffffU
#define __FLT32_MIN_EXP__ (-125)
#define __BFLT16_NORM_MAX__ 3.38953138925153547590470800371487867e+38BF16
#define __BFLT16_MIN_10_EXP__ (-37)
#define __ORDER_LITTLE_ENDIAN__ 1234
#define __WCHAR_MAX__ 0x7fffffff
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
#define __GCC_ATOMIC_CHAR_LOCK_FREE 2
#define __GCC_IEC_559 2
#define __FLT32X_DECIMAL_DIG__ 17
#define __FLT_EVAL_METHOD__ 0
#define __FLT64_DECIMAL_DIG__ 17
#define __GCC_ATOMIC_CHAR32_T_LOCK_FREE 2
#define __UINT_FAST64_MAX__ 0xffffffffffffffffUL
#define __SIG_ATOMIC_TYPE__ int
#define __DBL_MIN_10_EXP__ (-307)
#define __FINITE_MATH_ONLY__ 0
#define __FLT32X_MAX_EXP__ 1024
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __FLT32_HAS_DENORM__ 1
#define __UINT_FAST8_MAX__ 0xff
#define __FLT32_MAX_10_EXP__ 38
#define __INT8_C(c) c
#define __INT_LEAST8_WIDTH__ 8
#define __UINT_LEAST64_MAX__ 0xffffffffffffffffUL
#define __SHRT_MAX__ 0x7fff
#define __LDBL_MAX__ 1.18973149535723176502126385303097021e+4932L
#define __FLT64X_MAX_10_EXP__ 4932
#define __BFLT16_MAX_10_EXP__ 38
#define __BFLT16_MAX_EXP__ 128
#define __LDBL_IS_IEC_60559__ 1
#define __FLT64X_HAS_QUIET_NAN__ 1
#define __UINT_LEAST8_MAX__ 0xff
#define __GCC_ATOMIC_BOOL_LOCK_FREE 2
#define __FLT128_DENORM_MIN__ 6.47517511943802511092443895822764655e-4966F128
#define __UINTMAX_TYPE__ long unsigned int
#define __BFLT16_DECIMAL_DIG__ 4
#define __linux 1
#define __FLT_EVAL_METHOD_TS_18661_3__ 0
#define __UINT32_MAX__ 0xffffffffU
#define __DBL_DENORM_MIN__ ((double)4.94065645841246544176568792868221372e-324L)
#define __FLT128_MIN_EXP__ (-16381)
#define __WINT_MIN__ 0U
#define __FLT128_MIN_10_EXP__ (-4931)
#define __FLT32X_IS_IEC_60559__ 1
#define __INT_LEAST16_WIDTH__ 16
#define __SCHAR_MAX__ 0x7f
#define __FLT128_MANT_DIG__ 113
#define __WCHAR_MIN__ (-__WCHAR_MAX__ - 1)
#define __INT64_C(c) c ## L
[... 284 lines cut ...]
#define __PIE__ 2
#define __FLT32X_DENORM_MIN__ 4.94065645841246544176568792868221372e-324F32x
#define __DBL_MAX_10_EXP__ 308
#define __LDBL_DENORM_MIN__ 3.64519953188247460252840593361941982e-4951L
#define __INT16_C(c) c
#define __STDC__ 1
#define __PTRDIFF_TYPE__ long int
#define __FLT32X_MIN_10_EXP__ (-307)
#define __UINTPTR_TYPE__ long unsigned int
#define __linux__ 1
#define __LDBL_MIN_10_EXP__ (-4931)
#define __SIZEOF_LONG_LONG__ 8
#define __FLT128_DECIMAL_DIG__ 36
#define __GCC_ATOMIC_LLONG_LOCK_FREE 2
#define __FLT32_HAS_QUIET_NAN__ 1
#define __FLT_DECIMAL_DIG__ 9
#define __UINT_FAST16_MAX__ 0xffffffffU
#define __LDBL_NORM_MAX__ 1.18973149535723176502126385303097021e+4932L
#define __GCC_ATOMIC_SHORT_LOCK_FREE 2
#define __SIZE_MAX__ 0xffffffffffffffffUL
#define __UINT_FAST8_TYPE__ unsigned char
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_RELEASE 3

```

closed 2023-10-22 by BKPepe