openwrt/packages #22383: cloudflared does not compile with Go 1.21

BKPepe, 2023-10-13:
Maintainer: @1715173329 
Environment: OpenWrt master and OpenWrt 23.05, all targets

Description:

```
github.com/quic-go/quic-go/internal/qtls
# github.com/quic-go/quic-go/internal/qtls
../../vendor/github.com/quic-go/quic-go/internal/qtls/go121.go:5:13: cannot use "The version of quic-go you're using can't be built on Go 1.21 yet. For more details, please see https://github.com/quic-go/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.21 yet. F...) as int value in variable declaration
google.golang.org/grpc
github.com/cloudflare/cloudflared/management
github.com/coredns/coredns/plugin/pkg/cidr
github.com/coredns/coredns/plugin/pkg/response
github.com/coredns/coredns/plugin/pkg/edns
github.com/coredns/coredns/plugin/pkg/parse
github.com/coredns/coredns/plugin/pkg/nonwriter
github.com/coredns/coredns/plugin/pkg/doh
github.com/coredns/coredns/plugin/pkg/rcode
github.com/coredns/coredns/plugin/pkg/dnstest
github.com/coredns/coredns/plugin/pkg/dnsutil
github.com/coredns/coredns/request
github.com/coredns/coredns/plugin/etcd/msg
github.com/coredns/coredns/plugin
github.com/cloudflare/cloudflared/datagramsession
github.com/cloudflare/cloudflared/edgediscovery/allregions
github.com/cloudflare/cloudflared/logger
github.com/cloudflare/cloudflared/edgediscovery
github.com/coredns/coredns/plugin/pkg/trace
github.com/coredns/coredns/plugin/metrics/vars
github.com/grpc-ecosystem/grpc-opentracing/go/otgrpc
go.opentelemetry.io/proto/otlp/collector/trace/v1
github.com/coredns/coredns/pb
github.com/coredns/coredns/core/dnsserver
github.com/cloudflare/cloudflared/tracing
github.com/coredns/coredns/coremain
github.com/coredns/coredns/plugin/metadata
github.com/coredns/coredns/plugin/metrics
github.com/coredns/coredns/plugin/cache
github.com/cloudflare/cloudflared/tunneldns
github.com/cloudflare/cloudflared/config
github.com/cloudflare/cloudflared/cmd/cloudflared/cliutil
github.com/cloudflare/cloudflared/token
github.com/cloudflare/cloudflared/credentials
github.com/cloudflare/cloudflared/cmd/cloudflared/updater
github.com/cloudflare/cloudflared/cmd/cloudflared/tail
github.com/cloudflare/cloudflared/sshgen
github.com/cloudflare/cloudflared/carrier
zombiezen.com/go/capnproto2/internal/nodemap
zombiezen.com/go/capnproto2/encoding/text
zombiezen.com/go/capnproto2/pogs
github.com/cloudflare/cloudflared/cmd/cloudflared/access
github.com/cloudflare/cloudflared/quic/schema
zombiezen.com/go/capnproto2/std/capnp/rpc
zombiezen.com/go/capnproto2/rpc
github.com/cloudflare/cloudflared/tunnelrpc
github.com/cloudflare/cloudflared/tunnelrpc/pogs

make[3]: *** [Makefile:68: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/cloudflared-2023.7.3/.built] Error 1
time: package/feeds/packages/cloudflared/compile#106.50#27.00#66.73```

Snip from https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/cloudflared/compile.txt

1715173329, 2023-10-13:
I know it, it's an upstream issue. They're really slow.

BKPepe, 2023-10-18:
Let's have reference here: https://github.com/cloudflare/cloudflared/issues/1054

yichya, 2023-10-18:
Try https://github.com/yichya/openwrt-cloudflared when they're still figuring that out.

commit d16377d2 (2024-01-14): cloudflared: Update to 2024.1.2

Finally fixed build with Go 1.21.

Fixes: #22383
commit ebddc07c (2024-01-26): cloudflared: Update to 2024.1.2

Finally fixed build with Go 1.21.

Fixes: #22383

(cherry picked from commit d16377d23fd08e8c5583bba6d93afd89ad3144eb)
closed 2024-01-14 by 1715173329