openwrt/packages #22385: python-psycopg2: pg_config executable not found

BKPepe, 2023-10-13:
Maintainer: @dddaniel
Environment: OpenWrt master, aarch64_cortex-a53

Description:

```
* Getting build dependencies for wheel...
running egg_info
writing psycopg2.egg-info/PKG-INFO
writing dependency_links to psycopg2.egg-info/dependency_links.txt
writing top-level names to psycopg2.egg-info/top_level.txt

Error: pg_config executable not found.

pg_config is required to build psycopg2 from source.  Please add the directory
containing pg_config to the $PATH or specify the full executable path with the
option:

    python setup.py build_ext --pg-config /path/to/pg_config build ...

or with the pg_config option in 'setup.cfg'.

If you prefer to avoid building psycopg2 from source, please install the PyPI
'psycopg2-binary' package instead.

For further information please check the 'doc/src/install.rst' file (also at
<https://www.psycopg.org/docs/install.html>).

ERROR Backend subprocess exited when trying to invoke get_requires_for_build_wheel
make[3]: *** [Makefile:49: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/pypi/psycopg2-2.9.7/.built] Error 1
time: package/feeds/packages/python-psycopg2/compile#3.24#0.36#4.30```

jefferyto, 2023-10-13:
This should be fixed by #22339 (by removing the incorrect `PATH` variable in `PYTHON3_VARS`).

BKPepe, 2023-10-27:
Yup, confirming. It is fixed. Closing, thank you!

closed 2023-10-27 by BKPepe