openwrt/packages #22409: oath-toolkit: Build fails when libpam is not selected

Frederick888, 2023-10-15:
Maintainer: @1715173329
Environment: arm cortex-a9, Linksys WRT3200ACM, OpenWRT v23.05.0

Description:

I only need the command line utilities and I don't have `libpam` selected. `oath-pam` is hidden in `menuconfig`.

However my build failed due to `oath-pam` errors:

```
libtool: relink: ccache arm-openwrt-linux-muslgnueabi-gcc -shared  -fPIC -DPIC  .libs/pam_oath.o   -L/tmp/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-12.3.0_musl_eabi/usr/lib -L/tmp/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_
gcc-12.3.0_musl_eabi/lib -L/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5/ipkg-install/usr/lib -L/usr/lib -loath -lpam  -Os -mfloat-abi=hard -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro   -Wl,-soname -Wl,
pam_oath.so -o .libs/pam_oath.so
make[3]: Leaving directory '/tmp/openwrt/package/system/urandom-seed'
touch /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/stamp/.libext2fs_installed
time: package/system/urandom-seed/compile#0.71#0.36#1.23
echo "libext2fs" >> /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/pkginfo/e2fsprogs.default.install
mkdir -p /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/stamp
/usr/lib/libpam.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
mkdir -p /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/stamp
SHELL= flock /tmp/openwrt/tmp/.root-copy.flock -c 'cp -fpR /tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gptfdisk-1.0.9/.pkgdir/cgdisk/. /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/'
SHELL= flock /tmp/openwrt/tmp/.root-copy.flock -c 'cp -fpR /tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/e2fsprogs-1.47.0/.pkgdir/e2fsprogs/. /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/'
if [ -f "/tmp/openwrt/staging_dir/hostpkg/stamp/python-installer.list" ]; then "/tmp/openwrt/scripts/clean-package.sh" "/tmp/openwrt/staging_dir/hostpkg/stamp/python-installer.list" "/tmp/openwrt/staging_dir/hostpkg" ; rm -f "/tmp/openwrt/s
taging_dir/hostpkg/stamp/python-installer.list" ; fi
touch /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/stamp/.cgdisk_installed
libtool:   error: error: relink 'pam_oath.la' with the above command before installing it
make[8]: *** [Makefile:478: install-pammodLTLIBRARIES] Error 1
make[8]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5/pam_oath'
make[7]: *** [Makefile:869: install-am] Error 2
make[7]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5/pam_oath'
cd "/tmp/openwrt/tmp/host-stage-python-installer" && find ./ > "/tmp/openwrt/tmp/host-stage-python-installer.files"
make[6]: *** [Makefile:565: install-recursive] Error 1
make[6]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5/pam_oath'
make[5]: *** [Makefile:401: install-recursive] Error 1
make[5]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5'
make[4]: *** [Makefile:701: install] Error 2
make[4]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5'
touch /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-mvebu/stamp/.e2fsprogs_installed
make[3]: *** [Makefile:93: /tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/oath-toolkit-2.6.5/.built] Error 2
make[3]: Leaving directory '/tmp/openwrt/feeds/packages/utils/oath-toolkit'
time: package/feeds/packages/oath-toolkit/compile#2.93#1.44#5.98
    ERROR: package/feeds/packages/oath-toolkit failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/oath-toolkit/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
```

The issue was gone after I reverted 80dad35294197d537fe50f4e6143c74acfaf02b0.

1715173329, 2023-10-15:
I am not the maintainer of this package.

Frederick888, 2023-10-15:
> I am not the maintainer of this package.

Sorry I saw you merged #21124, and the issue template said 'find (maintainer) by checking history of the package Makefile'.

Updated to @famz.

Frederick888, 2023-10-15:
My build later failed with (below) after the revert:

```
rm -rf /tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/share/gettext/intl
 /usr/bin/mkdir -p '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/include'
 /tmp/openwrt/staging_dir/host/bin/install -c -m 644 libintl.h '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/include'
make[9]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/intl'
make[8]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/intl'
make[7]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/intl'
Making install in its
make[7]: Entering directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/its'
make[8]: Entering directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/its'
make[8]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/share/gettext-0.21.1/its'
 /tmp/openwrt/staging_dir/host/bin/install -c -m 644 glade.loc glade1.its glade2.its gtkbuilder.its gsettings.loc gsettings.its metainfo.loc metainfo.its '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/share/gettext-0.21.1/its'
make[8]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/its'
make[7]: Leaving directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/its'
Making install in gnulib-lib
make[7]: Entering directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/gnulib-lib'
make  install-am
make[8]: Entering directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/gnulib-lib'
make[9]: Entering directory '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/gnulib-lib'
 /usr/bin/mkdir -p '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/lib'
 /bin/sh ../libtool   --mode=install /tmp/openwrt/staging_dir/host/bin/install -c   libgettextlib.la '/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/ipkg-install/usr/lib'
libtool: warning: relinking 'libgettextlib.la'
libtool: install: (cd /tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/gnulib-lib; /bin/sh "/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1/gettext-tools/libtool"  --tag CC --mode=relink ccache arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/tmp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/gettext-0.21.1=gettext-0.21.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -no-undefined -lacl ../intl/libintl.la -L/tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/lib -lunistring -lattr -release 0.21.1 -L/tmp/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-12.3.0_musl_eabi/usr/lib -L/tmp/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-12.3.0_musl_eabi/lib -znow -zrelro -o libgettextlib.la -rpath /usr/lib libgettextlib_la-acl-errno-valid.lo libgettextlib_la-acl-internal.lo libgettextlib_la-get-permissions.lo libgettextlib_la-set-permissions.lo libgettextlib_la-allocator.lo libgettextlib_la-areadlink.lo libgettextlib_la-argmatch.lo libgettextlib_la-asyncsafe-spin.lo libgettextlib_la-openat-proc.lo libgettextlib_la-backupfile.lo libgettextlib_la-addext.lo libgettextlib_la-basename-lgpl.lo libgettextlib_la-binary-io.lo libgettextlib_la-bitrotate.lo libgettextlib_la-c-ctype.lo libgettextlib_la-c-strcasecmp.lo libgettextlib_la-c-strncasecmp.lo libgettextlib_la-c-strcasestr.lo libgettextlib_la-c-strstr.lo libgettextlib_la-canonicalize.lo libgettextlib_la-canonicalize-lgpl.lo libgettextlib_la-careadlinkat.lo libgettextlib_la-chdir-long.lo libgettextlib_la-classpath.lo libgettextlib_la-clean-temp.lo libgettextlib_la-clean-temp-simple.lo libgettextlib_la-cloexec.lo libgettextlib_la-closeout.lo libgettextlib_la-concat-filename.lo libgettextlib_la-copy-file.lo libgettextlib_la-csharpcomp.lo libgettextlib_la-csharpexec.lo libgettextlib_la-dirname-lgpl.lo libgettextlib_la-stripslash.lo libgettextlib_la-error.lo libgettextlib_la-error-progname.lo libgettextlib_la-execute.lo libgettextlib_la-exitfail.lo libgettextlib_la-fatal-signal.lo libgettextlib_la-fcntl.lo libgettextlib_la-fd-hook.lo libgettextlib_la-fd-safer-flag.lo libgettextlib_la-dup-safer-flag.lo libgettextlib_la-file-set.lo libgettextlib_la-filenamecat-lgpl.lo libgettextlib_la-findprog.lo libgettextlib_la-findprog-in.lo libgettextlib_la-free.lo libgettextlib_la-fstrcmp.lo libgettextlib_la-full-write.lo libgettextlib_la-fwriteerror.lo libgettextlib_la-gcd.lo libgettextlib_la-getcwd.lo libgettextlib_la-getcwd-lgpl.lo libgettextlib_la-getdelim.lo libgettextlib_la-getline.lo libgettextlib_la-getopt.lo libgettextlib_la-getopt1.lo libgettextlib_la-getprogname.lo libgettextlib_la-gettime.lo malloc/libgettextlib_la-scratch_buffer_grow.lo malloc/libgettextlib_la-scratch_buffer_grow_preserve.lo malloc/libg [... cut ...]
make[1]: *** [package/Makefile:114: /tmp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/tmp/openwrt'
make: *** [/tmp/openwrt/include/toplevel.mk:232: world] Error 2
```

Instead of building in my host system (Arch Linux), I moved it into an Ubuntu container and put `oath-pam` back in `Makefile`, finally everything compiled flawlessly.

Maybe it's something similar to https://forum.openwrt.org/t/openwrt-firmware-building-compiling-linking-error/74481?

THLIVSQAZ, 2023-10-17:
Did you try to  clean your build environment? I try to build you device with luci and oath-toolkit, Compiled successfully
the host is  ubuntu 22.04 x86
[image]

```
ubuntu@403b843a380a:/mnt/openwrtZone/opmaster/bin$ tree -L 4
.
|-- packages
|   `-- arm_cortex-a9_vfpv3-d16
|       |-- base
|       |   |-- Packages
|       |   |-- Packages.gz
|       |   |-- Packages.manifest
|       |   |-- Packages.sig
|       |   |-- busybox_1.36.1-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ca-bundle_20230311-1_all.ipk
|       |   |-- dnsmasq_2.89-6_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- firewall4_2023-09-01-598d9fbb-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- getrandom_2022-08-13-4c7b720b-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- hostapd-common_2023-09-08-e5ccbfc6-4_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- index.json
|       |   |-- iw_5.19-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- iwinfo_2023-07-01-ca79f641-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- jansson4_2.14-3_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- jshn_2023-05-23-75a3b870-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- jsonfilter_2018-02-04-c7e938d6-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libblobmsg-json20230523_2023-05-23-75a3b870-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libiwinfo-data_2023-07-01-ca79f641-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libiwinfo20230701_2023-07-01-ca79f641-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libjson-c5_0.17-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libjson-script20230523_2023-05-23-75a3b870-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libmbedtls12_2.28.5-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libmnl0_1.0.5-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libnftnl11_1.2.6-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libnl-tiny1_2023-07-27-bc92a280-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libubox20230523_2023-05-23-75a3b870-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libubus20230605_2023-06-05-f787c97b-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libuci20130104_2023-08-10-5781664d-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libuclient20201210_2023-04-13-007d9454-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libucode20220812_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- libustream-mbedtls20201210_2023-02-25-498f6e26-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- logd_2022-08-13-4c7b720b-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- mwifiex-sdio-firmware_20230804-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- netifd_2023-09-19-7a58b995-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- nftables-json_1.0.8-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- odhcp6c_2023-05-12-bcd28363-20_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- odhcpd-ipv6only_2023-06-24-52112643-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- openwrt-keyring_2022-03-25-62471e69-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- opkg_2022-02-24-d038e5b6-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ppp-mod-pppoe_2.4.9.git-2021-01-04-5_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ppp_2.4.9.git-2021-01-04-5_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- procd-seccomp_2023-06-25-2db83655-3_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- procd-ujail_2023-06-25-2db83655-3_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- procd_2023-06-25-2db83655-3_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- rpcd-mod-file_2023-07-01-c07ab2f9-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- rpcd-mod-iwinfo_2023-07-01-c07ab2f9-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- rpcd-mod-ucode_2023-07-01-c07ab2f9-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- rpcd_2023-07-01-c07ab2f9-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ubox_2022-08-13-4c7b720b-2_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ubus_2023-06-05-f787c97b-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ubusd_2023-06-05-f787c97b-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- uci_2023-08-10-5781664d-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- uclient-fetch_2023-04-13-007d9454-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-fs_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-math_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-nl80211_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-rtnl_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-ubus_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-uci_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
|       |   |-- ucode-mod-uloop_2023-06-06-c7d84aae-1_arm_cortex-a9_vfpv3-d16.ipk
[... 50 lines cut ...]
|       |   |-- Packages.sig
|       |   `-- index.json
|       `-- telephony
|           |-- Packages
|           |-- Packages.gz
|           |-- Packages.manifest
|           |-- Packages.sig
|           `-- index.json
`-- targets
    `-- mvebu
        `-- cortexa9
            |-- config.buildinfo
            |-- feeds.buildinfo
            |-- openwrt-mvebu-cortexa9-linksys_wrt3200acm-initramfs-kernel.bin
            |-- openwrt-mvebu-cortexa9-linksys_wrt3200acm-rootfs.tar.gz
            |-- openwrt-mvebu-cortexa9-linksys_wrt3200acm-squashfs-factory.img
            |-- openwrt-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.bin
            |-- openwrt-mvebu-cortexa9-linksys_wrt3200acm.manifest
            |-- packages
            |-- profiles.json
            |-- sha256sums
            `-- version.buildinfo

12 directories, 123 files
```

Frederick888, 2023-10-17:
@THLIVSQAZ The issue only happens on Arch with a newer toolchain. Like I mentioned I did not have any issues on Ubuntu either.