openwrt/packages #22410: igmpproxy: error in default config file

Tom-Rune, 2023-10-15:
OpenWRT 23.05.0
image: openwrt-23.05.0-mvebu-cortexa9-linksys_wrt1900acs-squashfs-factory
Router: Linksys WRT1900ACSv2

I tried to enable openwrt to work with IPTV over IGMP. So I installed the igmpproxy service but had a bit of a struggle get it to working. 

I eventually found an error in the default config file at /etc/config/igmpproxy
Line 9 read 'list altnet 192.168.1.0/24'. 
I changed the line to 'list altnet 0.0.0.0/0' and rebooted the router. After which IPTV started working on my ISP-provided 'box'. 

Using '0.0.0.0/0' is mentioned both is the doc for setup and a thread where someone had similar issues as I.

I think fixing this may be as simple as editing the standard config-file. I am not sure where this file resides originally, it may be a part of the igmpproxy-package or part of the router-image. 

I also enabled igmp-snooping, which seemed to prevent igmp-traffic from spreading to the rest of my network(the IPTV-box is connected to my routers lan2-port using TP-cable).

Some links:
https://openwrt.org/docs/guide-user/network/wan/udp_multicast
https://forum.openwrt.org/t/solved-iptv-igmpproxy-and-firewall-issues/12890/41?page=2

my igmpproxy file now looks like this, only the altnet adress has been changed be me.
config igmpproxy
	option quickleave 1
#	option verbose [0-3](none, minimal[default], more, maximum)

config phyint
	option network wan
	option zone wan
	option direction upstream
	list altnet 0.0.0.0/0

config phyint
	option network lan
	option zone lan
	option direction downstream

jmv2009, 2023-10-28:
To me opening up to 0.0.0.0/0 looks like a security issue and/or denial of service risk, as the whole world will be able to multicast to your private network.

Tom-Rune, 2023-11-02:
Are you actually opening your network to mulitcast from the whole world? Will traffic enter unless a device on the inside request a multicast stream? Pretty much what NAT do.

Also, I *think* most ISPs generally don't allow mulitcast in their networks, unless it has been approved by them(or they are the source)

The alternative is to manually add approved igmp-sources yourself, firewall style. That is so 1990s :p