openwrt/packages #22418: knot-dig: WARNING: TLS, handshake failed (GnuTLS internal error.)

cherinyy, 2023-10-16:
Maintainer: @salzmdan

Environment: 

- Architecture: `aarch64_cortex-a53`
- Model: `CMCC RAX3000M` (https://github.com/openwrt/openwrt/pull/13513)
- OpenWrt Version: `23.05-SNAPSHOT`

Description:

```
root@RAX3000M:~# kdig @dns.google +tls-ca www.example.com
;; WARNING: TLS, handshake failed (GnuTLS internal error.)
;; WARNING: TLS, handshake failed (GnuTLS internal error.)
Segmentation fault
```

salzmdan, 2023-10-16:
Unfortunately, we are unable to reproduce it. However, we fixed some related things in the upcoming 3.3.2, which will be released in a few weeks.

1715173329, 2023-10-16:
I cannot reproduce this issue either - on both OpenWrt and ImmortalWrt.

Please check the libgnutls library on your device, try to install it from opkg feeds if you're compiling from source, vase versa.

cherinyy, 2023-10-17:
factory-fresh ImmortalWrt device

```
root@ImmortalWrt:~# opkg install knot-dig
Installing knot-dig (3.3.1-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/packages/knot-dig_3.3.1-1_aarch64_cortex-a53.ipk
Installing libatomic1 (12.3.0-4) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/targets/mediatek/filogic/packages/libatomic1_12.3.0-4_aarch64_cortex-a53.ipk
Installing libgnutls (3.8.0-3) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/packages/libgnutls_3.8.0-3_aarch64_cortex-a53.ipk
Installing lmdb (0.9.29-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/packages/lmdb_0.9.29-1_aarch64_cortex-a53.ipk
Installing knot-libs (3.3.1-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/packages/knot-libs_3.3.1-1_aarch64_cortex-a53.ipk
Installing terminfo (6.4-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/base/terminfo_6.4-2_aarch64_cortex-a53.ipk
Installing libncurses6 (6.4-2) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/base/libncurses6_6.4-2_aarch64_cortex-a53.ipk
Installing libedit (20221030-3.1-1) to root...
Downloading https://mirrors.vsean.net/openwrt/releases/23.05-SNAPSHOT/packages/aarch64_cortex-a53/packages/libedit_20221030-3.1-1_aarch64_cortex-a53.ipk
Configuring terminfo.
Configuring lmdb.
Configuring libatomic1.
Configuring libgnutls.
Configuring knot-libs.
Configuring libncurses6.
Configuring libedit.
Configuring knot-dig.
root@ImmortalWrt:~# kdig @8.8.4.4 +tls-ca www.example.com
;; WARNING: TLS, handshake failed (GnuTLS internal error.)
;; ERROR: failed to query server 8.8.4.4@853(TCP)
```

salzmdan, 2023-10-17:
> factory-fresh ImmortalWrt device
>
> ```
> root@ImmortalWrt:~# opkg install knot-dig
> Installing knot-dig (3.3.1-1) to root...
> Downloading [...]

Does it happen also without `+tls-ca`?

cherinyy, 2023-10-17:
> Does it happen also without `+tls-ca`?

No

```
root@RAX3000M:~# kdig @8.8.4.4 www.example.com
;; ->>HEADER<<- opcode: QUERY; status: NOERROR; id: 13766
;; Flags: qr rd ra ad; QUERY: 1; ANSWER: 1; AUTHORITY: 0; ADDITIONAL: 0

;; QUESTION SECTION:
;; www.example.com.             IN      A

;; ANSWER SECTION:
www.example.com.        18585   IN      A       93.184.216.34

;; Received 49 B
;; Time 2023-10-17 18:21:46 CST
;; From 8.8.4.4@53(UDP) in 167.4 ms
root@RAX3000M:~# kdig @8.8.4.4 +tls www.example.com
;; WARNING: TLS, handshake failed (GnuTLS internal error.)
;; ERROR: failed to query server 8.8.4.4@853(TCP)
```

salzmdan, 2023-10-17:
Could you try

`kdig @dns.adguard.com www.example.com +https`

`kdig @dns.adguard.com www.example.com +quic`

?

cherinyy, 2023-10-17:
> kdig @dns.adguard.com www.example.com +https

I don't have the HTTPS feature in my version.

> kdig @dns.adguard.com www.example.com +quic

```
root@RAX3000M:~# kdig @dns.adguard.com www.example.com +quic
;; ERROR: failed to query server dns.adguard.com@853(UDP)
root@RAX3000M:~# kdig @dns.alidns.com www.example.com +quic
;; ERROR: failed to query server dns.alidns.com@853(UDP)
```

salzmdan, 2023-10-17:
I have no idea :thinking: Does kdig with TLS work from your non-owrt device (desktop/server)?

cherinyy, 2023-10-17:
> Does kdig with TLS work from your non-owrt device (desktop/server)?

I can use `kdig` fine in my Armbian device, and because of that I've actually noticed that `knot-dig` doesn't work on my ImmortalWrt devices for a while now, but I haven't given feedback.

```
Phicomm-N1:~:# kdig @94.140.14.14 +quic www.example.com
;; QUIC session (QUICv1)-(TLS1.3)-(ECDHE-X25519)-(ECDSA-SECP256R1-SHA256)-(AES-128-GCM)
;; ->>HEADER<<- opcode: QUERY; status: NOERROR; id: 0
;; Flags: qr rd ra ad; QUERY: 1; ANSWER: 1; AUTHORITY: 0; ADDITIONAL: 1

;; EDNS PSEUDOSECTION:
;; Version: 0; flags: ; UDP size: 0 B; ext-rcode: NOERROR
;; PADDING: 23 B

;; QUESTION SECTION:
;; www.example.com.             IN      A

;; ANSWER SECTION:
www.example.com.        23339   IN      A       93.184.216.34

;; Received 87 B
;; Time 2023-10-17 19:59:02 CST
;; From 94.140.14.14@853(QUIC) in 976.5 ms
Phicomm-N1:~:# kdig @94.140.14.14 +tls-ca www.example.com
;; TLS session (TLS1.3)-(ECDHE-X25519)-(ECDSA-SECP256R1-SHA256)-(AES-128-GCM)
;; ->>HEADER<<- opcode: QUERY; status: NOERROR; id: 64268
;; Flags: qr rd ra ad; QUERY: 1; ANSWER: 1; AUTHORITY: 0; ADDITIONAL: 1

;; EDNS PSEUDOSECTION:
;; Version: 0; flags: ; UDP size: 0 B; ext-rcode: NOERROR
;; PADDING: 3 B

;; QUESTION SECTION:
;; www.example.com.             IN      A

;; ANSWER SECTION:
www.example.com.        23332   IN      A       93.184.216.34

;; Received 67 B
;; Time 2023-10-17 19:59:09 CST
;; From 94.140.14.14@853(TLS) in 1332.1 ms
Phicomm-N1:~:# kdig @94.140.14.14 +https www.example.com
;; TLS session (TLS1.3)-(ECDHE-X25519)-(ECDSA-SECP256R1-SHA256)-(AES-128-GCM)
;; HTTP session (HTTP/2-POST)-(94.140.14.14/dns-query)-(status: 200)
;; ->>HEADER<<- opcode: QUERY; status: NOERROR; id: 0
;; Flags: qr rd ra ad; QUERY: 1; ANSWER: 1; AUTHORITY: 0; ADDITIONAL: 1

;; EDNS PSEUDOSECTION:
;; Version: 0; flags: ; UDP size: 0 B; ext-rcode: NOERROR
;; PADDING: 18 B

;; QUESTION SECTION:
;; www.example.com.             IN      A

;; ANSWER SECTION:
www.example.com.        70334   IN      A       93.184.216.34

;; Received 82 B
;; Time 2023-10-17 19:59:14 CST
;; From 94.140.14.14@443(HTTPS) in 986.5 ms
```

salzmdan, 2023-10-17:
Are you able to execute `gnutls-cli 1.1.1.1 -p 853` on the ImmortalWrt?

cherinyy, 2023-10-17:
Sincere thanks to both of you for your help!