openwrt/packages #22425: ddns-scripts: add support for eurodns

larieu, 2023-10-17:
I have tested and is working 

based on [https://www.eurodns.com/dynamic-dns-documentation](url)

created in **/usr/share/ddns/default/** 
a new file named:   **eurodns.com.json**

with the following content

```
{
	"name": "eurodns.com",
	"ipv4": {
		"url": "https://[username]:[password]@update.eurodyndns.org/update/?hostname=[DOMAIN]&myip=[IP]"
	}
}
```

added the name 
**eurodns.com** 
in the file **/usr/share/ddns/list**

created a ddns profile in 
**/etc/config/ddns**

```
config service 'yourname_here'
	option service_name 'eurodns.com'
	option use_ipv6 '0'
	option enabled '1'
	option lookup_host 'subdomain.example.com'
	option domain 'example.com'
	option username 'user_name'
	option password '***PW***'
	option ip_source 'network'
	option use_syslog '2'
	option check_unit 'minutes'
	option force_unit 'minutes'
	option retry_unit 'seconds'
	option ip_network 'WAN'
	option interface 'WAN'
```

restarted the service

all seems OK

BKPepe, 2023-10-17:
Feel free to create pull request to add support for it. We will not do that for you, I am sorry.

larieu, 2023-10-17:
never used git for this
can you support me to do a pull request?

larieu, 2023-10-18:
Pull request creation failed. Validation failed: must be a collaborator

BKPepe, 2023-10-18:
1. https://openwrt.org/docs/guide-developer/working-with-github-pr
2. https://openwrt.org/submitting-patches

If you can not find it on OpenWrt documentation, use the Internet to help you to do that. Usually, YouTube learning videos could be fine as well. (E.g. this one: https://www.youtube.com/watch?v=XTolZqmZq6s)

Is it possible that you might be stuck, hopefully not, if yes, then you can ask OpenWrt community on forum.

Nothing what we can do here, so closing this issue. This is feature request.

closed 2023-10-18 by BKPepe