openwrt/packages #22428: snmpd 5.9.1-6 shipped with OpenWrt 23.05.0 has bad memory leak

chris-palmer99, 2023-10-17:
Environment: (multiple different environments)

Description:

I have used snmpd, polled by the Zabbix "Network Generic Device SNMP" template, on previous OpenWrt releases without any problems. The routers tend to only be rebooted for OpenWrt upgrades, and there has never been a memory problem.

After upgrading 3 different environments to 23.05.0, snmpd is now leaking memory causing OOM conditions every 2-5 days (depending on router memory size, number of interfaces, etc). I have had to schedule a daily restart of snmpd.

An example from top on one system this morning:

```
Mem: 443756K used, 42280K free, 1016K shrd, 0K buff, 20568K cached
CPU:   0% usr   0% sys   0% nic  98% idle   0% io   0% irq   0% sirq
Load average: 0.03 0.03 0.00 4/82 8073
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1266     1 root     S     3740   1%   1% /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
 1872     1 root     S     2844   1%   0% /usr/sbin/uhttpd -f -h /www -r unify2 -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80
 3344     1 root     S     348m  73%   0% /usr/sbin/snmpd -Lf /dev/null -f -r
  708     1 ubus     S     1380   0%   0% /sbin/ubusd

```

stintel, 2023-10-17:
Please run it through valgrind.

chris-palmer99, 2023-10-18:
I've tried unsuccessfully to run under valgrind on one of the larger routers (with enough space to install valgrind). It was a Ubiquiti UniFi 6 Lite, MediaTek MT7621 ver:1 eco:3, ramips/mt7621, OpenWrt 23.05.0 r23497-6637af95aa.

The first attempt:
```
valgrind /usr/sbin/snmpd -Lf /dev/null -f -r
==11042== Memcheck, a memory error detector
==11042== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==11042== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==11042== Command: /usr/sbin/snmpd -Lf /dev/null -f -r
==11042== 
==11042== Conditional jump or move depends on uninitialised value(s)
==11042==    at 0x4081A1C: ??? (in /lib/libc.so)
==11042==    by 0x4095B10: ??? (in /lib/libc.so)
==11042== 
==11042== Conditional jump or move depends on uninitialised value(s)
==11042==    at 0x4081020: ??? (in /lib/libc.so)
==11042==    by 0x4081540: ??? (in /lib/libc.so)
==11042== 

vex: priv/guest_mips_toIR.c:23432 (disInstr_MIPS_WRK_10): Assertion `mode64' failed.
vex storage: T total 39795360 bytes allocated
vex storage: P total 0 bytes allocated

valgrind: the 'impossible' happened:
   LibVEX called failure_exit().

host stacktrace:
==11042==    at 0x580B688C: show_sched_status_wrk (in /usr/lib/valgrind/memcheck-mips32-linux)
==11042==    by 0x580B6878: show_sched_status_wrk (in /usr/lib/valgrind/memcheck-mips32-linux)

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable (lwpid 11042)
==11042==    at 0x4B21B1D: ??? (in /usr/lib/libnl-tiny.so.1)
==11042==    by 0x4093A40: ??? (in /lib/libc.so)
client stack range: [0x7ED38000 0x7ED38FFF] client SP: 0x7ED38BB0
valgrind stack range: [0x42858000 0x42957FFF] top usage: 9012 of 1048576
```

Then trying using massif:
```
valgrind --tool=massif /usr/sbin/snmpd -Lf /dev/null -f -r
==11171== Massif, a heap profiler
==11171== Copyright (C) 2003-2017, and GNU GPL'd, by Nicholas Nethercote
==11171== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==11171== Command: /usr/sbin/snmpd -Lf /dev/null -f -r
==11171== 

vex: priv/guest_mips_toIR.c:23432 (disInstr_MIPS_WRK_10): Assertion `mode64' failed.
vex storage: T total 24092448 bytes allocated
vex storage: P total 0 bytes allocated

valgrind: the 'impossible' happened:
   LibVEX called failure_exit().

host stacktrace:
==11171==    at 0x580B7420: show_sched_status_wrk (m_libcassert.c:406)
==11171==    by 0x580B765C: report_and_quit (m_libcassert.c:477)
==11171==    by 0x580FB434: UnknownInlinedFun (m_libcassert.c:553)
==11171==    by 0x580FB434: UnknownInlinedFun (m_libcassert.c:558)
==11171==    by 0x580FB434: vgPlain_core_panic (m_libcassert.c:563)
==11171==    by 0x58114914: failure_exit (m_translate.c:761)
==11171==    by 0x5800A6D8: vex_assert_fail (main_util.c:245)
==11171==    by 0x5807DB9C: disInstr_MIPS_WRK_10.constprop.0.isra.0 (guest_mips_toIR.c:23393)
==11171==    by 0x58085660: disInstr_MIPS_WRK (guest_mips_toIR.c:24961)
==11171==    by 0x5808581C: disInstr_MIPS (guest_mips_toIR.c:25135)
==11171==    by 0x580789D8: disassemble_basic_block_till_stop.constprop.0 (guest_generic_bb_to_IR.c:956)
==11171==    by 0x580AAE84: UnknownInlinedFun (guest_generic_bb_to_IR.c:1365)
==11171==    by 0x580AAE84: UnknownInlinedFun (main_main.c:583)
==11171==    by 0x580AAE84: UnknownInlinedFun (main_main.c:1235)
==11171==    by 0x580AAE84: vgPlain_translate (m_translate.c:1831)
==11171==    by 0x581582B8: UnknownInlinedFun (scheduler.c:1136)
==11171==    by 0x581582B8: vgPlain_scheduler (scheduler.c:1526)
==11171==    by 0x5815A4B4: UnknownInlinedFun (syswrap-linux.c:102)
==11171==    by 0x5815A4B4: run_a_thread_NORETURN (syswrap-linux.c:155)

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable (lwpid 11171)
==11171==    at 0x4B11B1D: ??? (in /usr/lib/libnl-tiny.so.1)
==11171==    by 0x4093A40: ??? (in /lib/libc.so)
client stack range: [0x7E84D000 0x7E84DFFF] client SP: 0x7E84DBB0
valgrind stack range: [0x427B3000 0x428B2FFF] top usage: 9012 of 1048576
```

stintel, 2023-10-18:
That's unfortunate. I suspect this is due to 292551163d91d0f6966dbc8abec6a2dff6db320c, which is a backport of #22254, which in turn was done because of #22006. @Ansuel @BKPepe as 23.05 still has PCRE 1, maybe we should revert this?

BKPepe, 2023-10-18:
> Actually there is this ***@***.***

Let's see. I can backport it, but no sooner than in 3 hours.

Ansuel, 2023-10-18:
@BKPepe already created the pr, with @stintel approval I can merge and backport if it's ok for you.

closed 2023-10-18 by stintel

Ansuel, 2023-10-18:
@chris-palmer99 we have backport a patch that in theory should fix this... You should wait 2-3 days for the bot to recompile the package. Please when you notice the update, report back if the leak is fixed or not!

chris-palmer99, 2023-10-22:
That seems to have fixed it. It's been running for a few hours on 2 of the routers, and have just loaded it onto the 3rd. Thank you so much for the fast fix.

stintel, 2023-10-22:
Thanks for the feedback!

mhoran, 2023-10-23:
Unfortunately I still see a memory leak with 5.9.7-1 on an EdgeRouter-X.

[image]

chris-palmer99, 2023-10-23:
Do you mean 5.9.7-1, or 5.9.1-7?
Did you update both snmpd and libnetsnmp packages?

waja, 2023-10-23:
[image]

I can confirm, that it seems to be fixed.

PR #22438 (2023-10-18): net-snmp: backport patch fixing memory leak for PCRE2
Backport patch fixing memory leak for PCRE2 present upstream.

Fixes: #22428

---

@stintel
commit 9f503616 (2023-10-18): net-snmp: backport patch fixing memory leak for PCRE2

Backport patch fixing memory leak for PCRE2 present upstream.

Fixes: #22428
commit fb0c4316 (2023-10-18): net-snmp: backport patch fixing memory leak for PCRE2

Backport patch fixing memory leak for PCRE2 present upstream.

Fixes: #22428

(cherry picked from commit 9f5036169175d853e2e0c76663f0bc98a8645f85)
commit 3eca7825 (2023-10-23): net-snmp: backport patch fixing memory leak for PCRE2

Backport patch fixing memory leak for PCRE2 present upstream.

Fixes: #22428

(cherry picked from commit 9f5036169175d853e2e0c76663f0bc98a8645f85)