openwrt/packages #22461: lm-sensors or luci-app-statistics doesn't show temp

rdeftery, 2023-10-20:
Environment: Asus rt-ac58u OpenWrt 23.05.0, r23497-6637af95aa

Description:
I try show temp.
I successfully installed lm-sensors and luci-app-statistics, and no one of them doesn't show temp

```
root@OpenWrt:~# sensors
ath10k_hwmon-isa-a000000
Adapter: ISA adapter
temp1:            N/A

ath10k_hwmon-isa-a800000
Adapter: ISA adapter
temp1:            N/A
```

rdeftery, 2023-10-20:
I erase config and reinstall openwrt (OpenWrt 23.05.0, r23497-6637af95aa).
I successfully installed luci-app-statistics (from https://openwrt.org/docs/guide-user/luci/luci_app_statistics), but nothing has changed. I still can't see the CPU temperature.

[image]
[image]

brada4, 2023-10-21:
You can try running sensors-detect to get non-functional sensors off the list. Seems particular ath10k shows NA in place of temperature number.

rdeftery, 2023-11-03:
Good day. 
My bad.
I tried reading the temperature sensorы with the wifi turned off.
If the Wi-Fi is turned off, the sensors also do not transmit temperature readings.
After turning on the Wi-Fi, the temperature is read correctly.
[image]

closed 2023-11-03 by rdeftery