openwrt/packages #22463: https-dns-proxy: Crash loop without apparent cause

Ryan-Goldstein, 2023-10-20:
Maintainer: @stangri
Environment: OpenWrt 23.05.0 / mediatek/mt7622 / Belkin RT3200 UBI
Customized image built using the [Firmware Selector](https://firmware-selector.openwrt.org/?version=23.05.0&target=mediatek%2Fmt7622&id=linksys_e8450-ubi):
- **Removed:** `kmod-usb3`, `ppp`, and `ppp-mod-pppoe`
- **Added:** `luci-ssl`, `https-dns-proxy`, `luci-app-https-dns-proxy`, `wireguard-tools`, `luci-app-wireguard`, `ddns-scripts-cloudflare`, and `luci-app-ddns`

I created a [thread](https://forum.openwrt.org/t/https-dns-proxy-crash-loop/175094) on the OpenWrt Forum and got stuck trying to determine the cause of a persistent crash loop of https-dns-proxy.

The output of each Troubleshooting command is as follows:

> service log restart; service dnsmasq restart; service https-dns-proxy restart
```
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, failing
Starting https-dns-proxy 2023-05-25-7 instances ✓
```

> logread -e dnsmasq; netstat -l -n -p | grep -e dnsmasq
```
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.5.100 -- 192.168.5.249, lease time 12h
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 9 names
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 20 names
Fri Oct 20 18:00:19 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
tcp        0      0 10.14.0.1:53            0.0.0.0:*               LISTEN      14338/dnsmasq
tcp        0      0 69.253.xxx.xxx:53       0.0.0.0:*               LISTEN      14338/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      14338/dnsmasq
tcp        0      0 192.168.5.1:53          0.0.0.0:*               LISTEN      14338/dnsmasq
tcp        0      0 2001:558:6027:86:c11b:d34c:xxx:xxx:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fd9f:5b3c:8f49::1:53    :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fe80::ea9f:80ff:feed:599b:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 2601:41:xxx:xxx::1:53  :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fe80::ea9f:80ff:feed:599c:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fe80::ea9f:80ff:feed:599c:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fe80::ea9f:80ff:feed:599e:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 fe80::ea9f:80ff:feed:599d:53 :::*                    LISTEN      14338/dnsmasq
tcp        0      0 ::1:53                  :::*                    LISTEN      14338/dnsmasq
udp        0      0 127.0.0.1:53            0.0.0.0:*                           14338/dnsmasq
udp        0      0 69.253.xxx.xxx:53       0.0.0.0:*                           14338/dnsmasq
udp        0      0 192.168.5.1:53          0.0.0.0:*                           14338/dnsmasq
udp        0      0 10.14.0.1:53            0.0.0.0:*                           14338/dnsmasq
udp        0      0 0.0.0.0:67              0.0.0.0:*                           14338/dnsmasq
udp        0      0 ::1:53                  :::*                                14338/dnsmasq
udp        0      0 fe80::ea9f:80ff:feed:599c:53 :::*                                14338/dnsmasq
udp        0      0 2001:558:6027:86:c11b:d34c:xxx:xxx:53 :::*                                14338/dnsmasq
udp        0      0 fe80::ea9f:80ff:feed:599b:53 :::*                                14338/dnsmasq
udp        0      0 2601:41:xxx:xxx::1:53  :::*                                14338/dnsmasq
udp        0      0 fd9f:5b3c:8f49::1:53    :::*                                14338/dnsmasq
udp        0      0 fe80::ea9f:80ff:feed:599c:53 :::*                                14338/dnsmasq
udp        0      0 fe80::ea9f:80ff:feed:599e:53 :::*                                14338/dnsmasq
udp        0      0 fe80::ea9f:80ff:feed:599d:53 :::*                                14338/dnsmasq
```

> logread -e https-dns; netstat -l -n -p | grep -e https-dns
```
Fri Oct 20 18:00:19 2023 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 146 crashes, 81 seconds since last crash
Fri Oct 20 18:00:20 2023 user.notice https-dns-proxy: Starting service instances ✓
[... 162 lines cut ...]

```
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.995282 main.c:219 Version: 2023-05-25-1
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.995296 main.c:220 Built: Oct 15 2023 17:51:39
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.995302 main.c:221 System c-ares: 1.19.1
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.995353 main.c:222 System libcurl: libcurl/8.4.0 mbedTLS/2.28.5 nghttp2/1.57.0
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [W] 1697844057.995377 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.997542 dns_server.c:50 Listening on 127.0.0.1:5053
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [D] 1697844057.997656 logging.c:40 starting periodic log flush timer
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [D] 1697844057.998122 dns_poller.c:178 Nameservers count: 2
Fri Oct 20 19:20:57 2023 daemon.info https-dns-proxy[10734]: [I] 1697844057.998141 main.c:309 DNS polling initialized for 'cloudflare-dns.com'
Fri Oct 20 19:21:04 2023 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 242 crashes, 6 seconds since last crash
Fri Oct 20 19:21:04 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.711545 main.c:219 Version: 2023-05-25-1
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.711560 main.c:220 Built: Oct 15 2023 17:51:39
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.711564 main.c:221 System c-ares: 1.19.1
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.711952 main.c:222 System libcurl: libcurl/8.4.0 mbedTLS/2.28.5 nghttp2/1.57.0
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [W] 1697844064.711984 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.713624 dns_server.c:50 Listening on 127.0.0.1:5053
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [D] 1697844064.713695 logging.c:40 starting periodic log flush timer
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [D] 1697844064.714346 dns_poller.c:178 Nameservers count: 2
Fri Oct 20 19:21:04 2023 daemon.info https-dns-proxy[10896]: [I] 1697844064.714367 main.c:309 DNS polling initialized for 'cloudflare-dns.com'
```

I'm happy to try any other troubleshooting/debugging steps that anyone proposes!

AcidSlide, 2023-10-21:
I'm also experiencing this on one of my test network.  But it seems its on my test network it happens when ifup/ifdown is triggered due to IPv6 changes especially if there happens to be routing issues in IPv6.

Ryan-Goldstein, 2023-10-21:
> I'm also experiencing this on one of my test network. But it seems its on my test network it happens when ifup/ifdown is triggered due to IPv6 changes especially if there happens to be routing issues in IPv6.

I believe my IPv6 connection is quite stable, and procd_trigger_wan6 is set to the 0 default. Short of disabling IPv6 entirely, any thoughts on what to try to determine whether IPv6 interface changes may be causing these crashes?

stangri, 2023-10-21:
@Ryan-Goldstein Can you try taking wan6 down either with `ifdown wan6` or thru WebUI? 

I've rewritten PROCD triggers sections recently, wonder if this is an unintended consequence.

Ryan-Goldstein, 2023-10-21:
> @Ryan-Goldstein Can you try taking wan6 down either with `ifdown wan6` or thru WebUI?
> 
> I've rewritten PROCD triggers sections recently, wonder if this is an unintended consequence.

Good thought - I did `ifdown wan6`, and it has been well over an hour with no crashes, so it does appear to be caused by IPv6.

Ryan-Goldstein, 2023-10-21:
And strangely, after doing `ifup wan6`, I started seeing a number of 'Starting service instances' messages without any crash messages, then I saw a crash message, as follows:

```
Sat Oct 21 18:45:32 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Sat Oct 21 18:45:36 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Sat Oct 21 18:45:45 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Sat Oct 21 18:45:52 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Sat Oct 21 18:48:03 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
Sat Oct 21 18:48:06 2023 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 3 seconds since last crash
Sat Oct 21 18:48:07 2023 user.notice https-dns-proxy: Starting service instances on_interface_up ✓
```

Unfortunately, I didn't have debug level logging on at the time, but hopefully that's enough to isolate and resolve the issue in a future build! Let me know if there's anything else you'd like me to try.

stangri, 2023-10-22:
@Ryan-Goldstein can you test the init script from [this commit](https://github.com/stangri/source.openwrt.melmac.net/commit/baaf7edac37450758fada4cb8c4a44c9277433ed)? I'd recommend you reboot the router after updating the init script.

Ryan-Goldstein, 2023-10-22:
> @Ryan-Goldstein can you test the init script from [this commit](https://github.com/stangri/source.openwrt.melmac.net/commit/baaf7edac37450758fada4cb8c4a44c9277433ed)? I'd recommend you reboot the router after updating the init script.

I've never done that kind of thing before, so I'm not quite sure what I'd need to do. Would you mind writing up some quick steps? Command line or luci would be fine.

stangri, 2023-10-23:
> I've never done that kind of thing before, so I'm not quite sure what I'd need to do. Would you mind writing up some quick steps? Command line or luci would be fine.

You'd need to replace the `/etc/init.d/https-dns-proxy` file on your router with [this file](https://raw.githubusercontent.com/stangri/source.openwrt.melmac.net/baaf7edac37450758fada4cb8c4a44c9277433ed/https-dns-proxy/files/etc/init.d/https-dns-proxy).

AcidSlide, 2023-10-23:
> You'd need to replace the `/etc/init.d/https-dns-proxy` file on your router with [this file](https://raw.githubusercontent.com/stangri/source.openwrt.melmac.net/baaf7edac37450758fada4cb8c4a44c9277433ed/https-dns-proxy/files/etc/init.d/https-dns-proxy).

Hi @stangri since I've experienced this too and one of my test networks, I'll try to simulate the wan6 related issue.

Ryan-Goldstein, 2023-10-24:
> You'd need to replace the `/etc/init.d/https-dns-proxy` file on your router with [this [...]

I just loaded that new init file, rebooted my router, and started https-dns-proxy and have been using it for the past several hours with no crashes - so that appears to have resolved the issue!

stangri, 2023-10-24:
Should be fixed in https://github.com/openwrt/packages/pull/22489 https://github.com/openwrt/packages/pull/22490 https://github.com/openwrt/packages/pull/22491