openwrt/packages #22466: https-dns-proxy: start is triggered even if the package is DISABLED when ifup is triggered

AcidSlide, 2023-10-21:
Maintainer: @stangri  
Environment: all arch / OpenWRT 22.x and 23.x
Description: Even if the package is **disabled** a start is triggered by the hotplug script: /etc/hotplug.d/iface/90-https-dns-proxy

```
# /etc/hotplug.d/iface/90-https-dns-proxy
#!/bin/sh
# Copied from https://openwrt.org/docs/guide-user/advanced/hotplug_extras
# shellcheck disable=SC1091
. /lib/functions/network.sh
network_flush_cache
network_find_wan NET_IF
network_find_wan6 NET_IF6
[ "$INTERFACE" != "$NET_IF" ] && [ "$INTERFACE" != "$NET_IF6" ] && exit 0
[ "$ACTION" != "ifup" ] && [ "$ACTION" != "ifupdate" ] && exit 0
[ "$ACTION" = "ifupdate" ] && [ -z "$IFUPDATE_ADDRESSES" ] && \
[ -z "$IFUPDATE_DATA" ] && exit 0

sleep 10
/etc/init.d/https-dns-proxy start 'on_hotplug'
```

There should be detection/checking if the package is disabled or not within the hotplug script

stangri, 2023-10-21:
@AcidSlide this file is no longer needed actually with changes to service triggers in 2023-05-25-7 and can be deleted. I'll push an update deleting it soon.

Let me know if everything works as expected with this file deleted.

AcidSlide, 2023-10-21:
> Let me know if everything works as expected with this file deleted.

I actually commented out everything in the script in 1 of 3 of my setups (the one which I've disabled the package) and can confirm that without the script `https-dns-proxy` doesn't randomly start anymore.

stangri, 2023-10-24:
Should be fixed in https://github.com/openwrt/packages/pull/22489 https://github.com/openwrt/packages/pull/22490 https://github.com/openwrt/packages/pull/22491

closed 2023-10-26 by stangri