openwrt/packages #22469: domoticz package does not start in 23.05.0

rolandvans, 2023-10-21:
domoticz package does not work in 23.05.0

It will not start because the Python version 3.11 is not supported and this is the default Python version in 23.05.0.
`Status: PluginSystem: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.`

Changing the file `990-python3.10_fix.patch`, replacing "python3.10" with "python3.11" 
```
#else
+				if (!shared_lib_) FindLibrary("python3.11", true);
 				if (!shared_lib_) FindLibrary("python3.9", true);
 				if (!shared_lib_) FindLibrary("python3.8", true);
 				if (!shared_lib_) FindLibrary("python3.7", true);
@@ -198,6 +192,9 @@ namespace Plugins {
...

```
Fixes the status message, but domoticz does not start properly on boot.
Manually starting domoticz does work:
`domoticz -sslwww 0 -approot /usr/share/domoticz/ -noupdates -userdata /var/lib/domoticz -syslog -daemon`

BKPepe, 2023-10-21:
Ping maintainer @dwmw2 about this issue.

rolandvans, 2023-11-08:
No response from the maintainer until now

WLKe001, 2023-11-14:
Try make symlink for libpython3.11.so and libpython3.11.a to libpython3.9.so and libpython3.9.a
This working for me ;-)

rolandvans, 2023-11-17:
> Try make symlink for libpython3.11.so and libpython3.11.a to libpython3.9.so and libpython3.9.a This working for me ;-)

That trick solves the libpython part, but in my setup, it still does not start on boot. Syslog shows this message:
`Fri Nov 17 07:00:40 2023 daemon.notice procd: /etc/rc.d/S95done: Command failed: ubus call service delete { "name": "domoticz" } (Not found)`
Starting it manually _does_ work. Any idea what could be the problem here?

BKPepe, 2023-11-24:
PIng @dwmw2.

Let's ping previous contributors, who was interested in it as well - @lucize, @stintel.

dwmw2, 2023-11-29:
Apologies for the delayed response. I've updated to 2023.2 on the master branch, will look at backporting just the libpython3.11 fix to stable, once I've got local builds of 23.05 actually working on my hardware.

Not sure about @rolandvans's issue with it not starting though. That seems to be something different.

dwmw2, 2023-12-01:
@rolandvans Could your other startup issue be caused by https://github.com/openwrt/openwrt/issues/9545 ?

dwmw2, 2023-12-05:
For me it ran smoothly anyway; I don't use python plugins. But yes, the update should fix python.

rumburak001, 2024-03-28:
Sorry to be late to the party.
I just upgraded openwrt on my Zyxel NSA325 (Kirkwood) to 23.05.3 and experience the very same behaviour.
Only manual start is possible.