openwrt/packages #22472: openvpn: DCO not enabled on snapshot

asdfjkluiop, 2023-10-22:
# DCO not enabled on OpenWRT snapshot

Maintainer: @AuthorReflex
Environment: amd64 snapshot build r24162-65380dc41c

Description:
As far as I can tell PR #21547 should've added DCO support but it doesn't seem to be present. I spent a bit of time trying to get DCO working to no avail and then realized when I run `openvpn --version` it does not show the [DCO] feature flag like it should. As far as I can tell the openvpn userspace program is not being build with DCO enabled, at least on amd64 builds.

brada4, 2023-10-22:
You need to build package enabling dco option in menuconfig. note no such option is available for 23.05

asdfjkluiop, 2023-10-22:
Well I figured I could build it from source if I wanted to but it seems odd to ship a binary kmod as an official package but then not ship a user space binary which can use it which is why I opened the issue.

brada4, 2023-10-22:
build it and benchmark it on average mips if possible vs non-dco version. 10x speedup looks mpre like marketing pitch/

Sarcasm not welcome, you literarily request removal of dco kmod.

asdfjkluiop, 2023-10-22:
I don't have a mips router as my build information in the original issue states. Also I wasn't being sarcastic? That was a genuine statement, without both packages the kmod is useless. I figured that was a mistake since there is PR enabling it but it wasn't enabled in actual builds.

brada4, 2023-10-22:
Build ipk and validate support. Extensive testing needs mips and aarch  having significant improvement to make essentially bigger package default.

asdfjkluiop, 2023-10-22:
If package size is an issue then what about adding a separate dco enabled package that way users could still use it without causing a size increase for storage constrained platforms.

brada4, 2023-10-23:
The burden of proof and PR is on you.

asdfjkluiop, 2023-10-23:
This is clearly not a productive conversation. I will wait till one of the actual maintainers replies since I'm not looking to submit a PR, this is simply an issue report about a missing feature. I'm not an OpenWRT contributor and based on PR history neither are you and I'm rather tired of your attitude.

meehien, 2023-10-25:
I would also like to have a DCO enabled package for openvpn.  I aggree with @asdfjkluiop as it stands now, the kmod is useless and I would add misleading, especially based on its description.

patrakov, 2025-06-19:
Can this be backported into 24.10?