openwrt/packages #22476: help perl

predators46, 2023-10-22:
MAINTAINER:=Marcel Denia <naoir@gmx.net>
		          Philip Prindeville <philipp@redfish-solutions.com>

freeradius3 add perl

```
configure: loading site script /home/open/sdk64/include/site/aarch64
checking for aarch64-openwrt-linux-gcc... aarch64-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether aarch64-openwrt-linux-musl-gcc accepts -g... yes
checking for aarch64-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... aarch64-openwrt-linux-musl-gcc -E
configure: Calling ExtUtils::Embed to get 'ccopts'
configure: ExtUtil's ccopts were " -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2  -I/home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/CORE "
configure: Sanitized ccopts are "-fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -isystem /usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2 -isystem /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/CORE"
configure: Calling ExtUtils::Embed to get 'ldflags'
configure: ExtUtil's ldopts were "  -fstack-protector-strong -L/usr/local/lib /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/B/B.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Compress/Raw/Bzip2/Bzip2.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Compress/Raw/Zlib/Zlib.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Cwd/Cwd.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Data/Dumper/Dumper.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Devel/Peek/Peek.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Digest/MD5/MD5.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Digest/SHA/SHA.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Encode.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Byte/Byte.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/CN/CN.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/EBCDIC/EBCDIC.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/JP/JP.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/KR/KR.a
/home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Symbol/Symbol.a
/home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/TW/TW.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Unicode/Unicode.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Fcntl/Fcntl.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/File/DosGlob/DosGlob.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/File/Glob/Glob.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Filter/Util/Call/Call.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Hash/Util/Util.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Hash/Util/FieldHash/FieldHash.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/I18N/Langinfo/Langinfo.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/IO/IO.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/IPC/SysV/SysV.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/List/Util/Util.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/MIME/Base64/Base64.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Math/BigInt/FastCalc/FastCalc.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Opcode/Opcode.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/POSIX/POSIX.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/PerlIO/encoding/encoding.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/PerlIO/mmap/mmap.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/PerlIO/scalar/scalar.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/PerlIO/via/via.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/SDBM_File/SDBM_File.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Socket/Socket.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Storable/Storable.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Sys/Hostname/Hostname.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Sys/Syslog/Syslog.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Tie/Hash/NamedCapture/NamedCapture.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Time/HiRes/HiRes.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Time/Piece/Piece.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Unicode/Collate/Collate.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Unicode/Normalize/Normalize.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/arybase/arybase.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/attributes/attributes.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/mro/mro.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/re/re.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/threads/threads.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/threads/shared/shared.a -L/home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/CORE -lperl -lpthread -lnsl -ldl -lm -lcrypt -lutil -lc -lrt"
configure: Sanitized ldopts are "-fstack-protector-strong -L/usr/local/lib /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/B/B.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Compress/Raw/Bzip2/Bzip2.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Compress/Raw/Zlib/Zlib.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Cwd/Cwd.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Data/Dumper/Dumper.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Devel/Peek/Peek.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Digest/MD5/MD5.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Digest/SHA/SHA.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Encode.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Byte/Byte.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/CN/CN.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/EBCDIC/EBCDIC.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/JP/JP.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/KR/KR.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Symbol/Symbol.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/TW/TW.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Encode/Unicode/Unicode.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/Fcntl/Fcntl.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/File/DosGlob/DosGlob.a /home/open/sdk64/staging_dir/hostpkg/usr/lib/perl5/5.28.1/x86_64-linux/auto/File/Glob/Glob.a /home/open/sdk64/staging_dir/hostp [... cut ...]
make[2]: *** [Makefile:992: /home/open/sdk64/build_dir/target-aarch64_cortex-a53_musl/freeradius-server-release_3_0_21/.configured_059ce960b95c68443337f43a915dbd53] Error 1
make[2]: Leaving directory '/home/open/sdk64/feeds/packages/net/freeradius3'
time: package/feeds/packages/freeradius3/compile#34.06#16.35#45.98
    ERROR: package/feeds/packages/freeradius3 failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/freeradius3/compile] Error 1
make[1]: Leaving directory '/home/open/sdk64'
```

voltagex, 2023-10-28:
Maintainer for freeradius3 appears to be @neheb 

This issue should probably be renamed freeradius3: missing Perl module

predators46, 2023-12-01:
it seems difficult to add perl in freeradius. krb5 is all I can add

closed 2024-01-04 by predators46

predators46, 2024-09-07:
@voltagex 

i have added perl in freeradius3

```
root@OpenWrt:/etc/freeradius3# radiusd -X
FreeRADIUS Version 3.2.6
Copyright (C) 1999-2023 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius3/dictionary
including dictionary file /etc/freeradius3/dictionary
including configuration file /etc/freeradius3/radiusd.conf
including configuration file /etc/freeradius3/proxy.conf
including configuration file /etc/freeradius3/clients.conf
including files in directory /etc/freeradius3/mods-enabled/
including configuration file /etc/freeradius3/mods-enabled/always
including configuration file /etc/freeradius3/mods-enabled/attr_filter
including configuration file /etc/freeradius3/mods-enabled/chap
including configuration file /etc/freeradius3/mods-enabled/detail
including configuration file /etc/freeradius3/mods-enabled/digest
including configuration file /etc/freeradius3/mods-enabled/eap
including configuration file /etc/freeradius3/mods-enabled/mschap
including configuration file /etc/freeradius3/mods-enabled/exec
including configuration file /etc/freeradius3/mods-enabled/expiration
including configuration file /etc/freeradius3/mods-enabled/expr
including configuration file /etc/freeradius3/mods-enabled/files
including configuration file /etc/freeradius3/mods-enabled/logintime
including configuration file /etc/freeradius3/mods-enabled/pap
including configuration file /etc/freeradius3/mods-enabled/preprocess
including configuration file /etc/freeradius3/mods-enabled/radutmp
including configuration file /etc/freeradius3/mods-enabled/sradutmp
including configuration file /etc/freeradius3/mods-enabled/realm
including configuration file /etc/freeradius3/mods-enabled/unix
including configuration file /etc/freeradius3/mods-enabled/passwd
including configuration file /etc/freeradius3/mods-enabled/perl
including files in directory /etc/freeradius3/policy.d/
including configuration file /etc/freeradius3/policy.d/accounting
including configuration file /etc/freeradius3/policy.d/filter
including configuration file /etc/freeradius3/policy.d/eap
including files in directory /etc/freeradius3/sites-enabled/
including configuration file /etc/freeradius3/sites-enabled/default
including configuration file /etc/freeradius3/sites-enabled/inner-tunnel
main {
 security {
        allow_core_dumps = no
 }
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
}
main {
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        sbindir = "/usr/sbin"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
        libdir = "/usr/lib/freeradius3"
        radacctdir = "/var/db/radacct"
        hostname_lookups = no
        max_request_time = 30
        proxy_dedup_window = 1
        cleanup_delay = 5
[... 552 lines cut ...]
}
listen {
        type = "acct"
        ipv6addr = ::
        port = 0
   limit {
        max_connections = 16
        lifetime = 0
        idle_timeout = 30
   }
}
listen {
        type = "auth"
        ipaddr = 127.0.0.1
        port = 18120
}
Listening on auth address * port 1812 bound to server default
Listening on acct address * port 1813 bound to server default
Listening on auth address :: port 1812 bound to server default
Listening on acct address :: port 1813 bound to server default
Listening on auth address 127.0.0.1 port 18120 bound to server inner-tunnel
Listening on proxy address * port 56717
Listening on proxy address :: port 52916
Ready to process requests
```