openwrt/packages #22503: i2pd: update to 2.49.0

sysfu, 2023-10-25:
Maintainer: @yangfl 
Environment: arch = mvebu/cortexa53  cpu model = ARMv8 Processor rev 4  OpenWRT = 22.03.4 r20123-38ccc47687

Description:
Please update the i2pd package to the [current 2.49.0 release](https://github.com/PurpleI2P/i2pd/releases/tag/2.49.0).

BKPepe, 2023-10-25:
Feel free to submit pull request to update it.

sysfu, 2023-10-25:
Tried submitting a pull request and got this error "Pull request creation failed. Validation failed: must be a collaborator"

BKPepe, 2023-10-26:
Of course, you need to fork the repository first and send the pull request from your repository to ours. Please follow OpenWrt documentation how you can do it. 

Because this is future request and there is no need to update it urgently (CVEs, security vulnerabilities, mem leaks, etc.), I am closing this. You might want to ask for help on OpenWrt forum.

closed 2023-10-26 by BKPepe

sysfu, 2023-10-26:
Ah, I didn't know the pull request button had to be clicked from _my_ repository, re-submitted from my fork and it worked now, thank you.