openwrt/packages #22517: amnezia-wg: refuses to build

stangri, 2023-10-26:
@jefferyto @1715173329 

I wonder if you could have a look at this, I've made a Makefile for the amnezia-wg package:
```
# This is free software, licensed under the MIT License.

include $(TOPDIR)/rules.mk

PKG_NAME:=amnezia-wg
PKG_VERSION:=0.1.6
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://codeload.github.com/amnezia-vpn/amnezia-wg/tar.gz/v${PKG_VERSION}?
PKG_HASH:=81fb720a97f817d7296e67e218c83487d05615d956ae18b41576c4b2d4f35491

PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

PKG_BUILD_DEPENDS:=golang/host
PKG_BUILD_PARALLEL:=1
PKG_BUILD_FLAGS:=no-mips16

GO_PKG:=github.com/amnezia-vpn/amnezia-wg
GO_PKG_LDFLAGS_X:=\
	main.Build=$(PKG_VERSION)

include $(INCLUDE_DIR)/package.mk
include ../../lang/golang/golang-package.mk

define Package/amnezia-wg
  SECTION:=net
  CATEGORY:=Network
  TITLE:=amnezia-wg
  DEPENDS:=$(GO_ARCH_DEPENDS)
endef

define Build/Compile
  $(call GoPackage/Build/Compile)
endef

define Package/amnezia-wg/description
  AmneziaWG is a contemporary version of the WireGuard protocol. It's a fork of
  WireGuard-Go and offers protection against detection by Deep Packet Inspection
  (DPI) systems. At the same time, it retains the simplified architecture and
  high performance of the original.
endef

define Package/amnezia-wg/install
	$(call GoPackage/Package/Install/Bin,$(PKG_INSTALL_DIR))
	$(INSTALL_DIR) $(1)/usr/sbin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/amnezia-wg $(1)/usr/sbin/
endef

$(eval $(call GoBinPackage,amnezia-wg))
$(eval $(call BuildPackage,amnezia-wg))
```

however it errors out with:
```
Finding targets
package github.com/amnezia-vpn/amnezia-wg/tun/netstack
	imports gvisor.dev/gvisor/pkg/bufferv2
	imports gvisor.dev/gvisor/pkg/atomicbitops
	imports gvisor.dev/gvisor/pkg/state
	imports gvisor.dev/gvisor/pkg/state/wire
	imports gvisor.dev/gvisor/pkg/gohacks: build constraints exclude all Go files in /home/stangri/development/Builder/SDK_x86-64_23.05.0/dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20221203005347-703fd9b7fbc0/pkg/gohacks

Building targets
package github.com/amnezia-vpn/amnezia-wg/tun/netstack
	imports gvisor.dev/gvisor/pkg/bufferv2
	imports gvisor.dev/gvisor/pkg/atomicbitops
	imports gvisor.dev/gvisor/pkg/state
	imports gvisor.dev/gvisor/pkg/state/wire
	imports gvisor.dev/gvisor/pkg/gohacks: build constraints exclude all Go files in /home/stangri/development/Builder/SDK_x86-64_23.05.0/dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20221203005347-703fd9b7fbc0/pkg/gohacks

make[2]: *** [Makefile:62: /home/stangri/development/Builder/SDK_x86-64_23.05.0/build_dir/target-x86_64_musl/amnezia-wg-0.1.6/.built] Error 1
make[2]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_23.05.0/feeds/packages/stangri/amnezia-wg'
time: package/amnezia-wg/compile#0.75#0.28#1.02
    ERROR: package/amnezia-wg failed to build.
make[1]: *** [package/Makefile:120: package/amnezia-wg/compile] Error 1
```

I'm not familiar with Go -- is there anything I can do/patch to make it build for OpenWrt?

Thanks!

jefferyto, 2023-10-27:
You may want to try to contact upstream to pull in changes from their upstream; I believe https://github.com/WireGuard/wireguard-go/commit/42ec952eadc297efadc70b9911d5a59bcd2db4a6 allows both to compile with Go 1.21. You can try applying that as a patch but it may not apply cleanly as upstream has modified go.mod/go.sum as well.

egc112, 2023-10-31:
@Stangri, off-topic but if you are interested in  WireGuard obfuscation see: https://github.com/infinet/xt_wgobfs

closed 2023-10-31 by stangri