openwrt/packages #22530: qemu: 8.1.2 fails to build, --with-git-submodules=ignore removed

voltagex, 2023-10-28:
Maintainer: @yousong  (find it by checking history of the package Makefile)
Environment: master/future - this does not affect OpenWRT yet!

Description: https://gitlab.com/qemu-project/qemu/-/commit/6f3ae23b29ad5831902e3ecdc7e443bbbf295bde removes the --with-git-submodules=ignore option

```diff
diff --git a/utils/qemu/Makefile b/utils/qemu/Makefile
index 4f99c32..5e9851b 100644
--- a/utils/qemu/Makefile
+++ b/utils/qemu/Makefile
@@ -9,10 +9,10 @@
 include $(TOPDIR)/rules.mk

 PKG_NAME:=qemu
-PKG_VERSION:=8.0.2
+PKG_VERSION:=8.1.2
 PKG_RELEASE:=1
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
-PKG_HASH:=f060abd435fbe6794125e2c398568ffc3cfa540042596907a8b18edca34cf6a5
+PKG_HASH:=541526a764576eb494d2ff5ec46aeb253e62ea29035d1c23c0a8af4e6cd4f087
 PKG_SOURCE_URL:=http://download.qemu.org/
 PKG_LICENSE:=GPL-2.0-only
 PKG_LICENSE_FILES:=LICENSE tcg/LICENSE
@@ -392,7 +392,7 @@ CONFIGURE_ARGS +=                   \
        --disable-docs                  \
        --disable-fuse                  \
        --disable-gcrypt                \
-       --with-git-submodules=ignore    \
+       --disable-download              \
        --disable-glusterfs             \
        --disable-gnutls                \
        --disable-guest-agent-msi       \
```

(I'm still building, but this patch should both update to 8.1.2 and fix the configure failure)

vooon, 2023-10-29:
Hi,

Hmm, yeah, i should add --disable-download instead into #22524

voltagex, 2023-10-29:
Whoops, missed that one - thanks.

closed 2023-10-29 by voltagex