openwrt/packages #22572: nginx-full does not support streams in 23.05

npenin, 2023-11-01:
Maintainer: @jmarcet @neheb
Environment: armv7l, wrt3200acm, 23.05

Description:
After upgrading to 23.05, nginx complains about ` unknown directive "stream"` in my file. 

I am using the stream module to reverse proxy some internal web sites + have luci running. I modified the uci.conf.template to listen on 443 and made the luci configuration to listen on 444.

```
stream {
        map $ssl_preread_server_name $name {
                XXX.XXX.XX 10.68.1.2;
                default localhttps;
        }

        upstream localhttps {
                server localhost:444;
        }

        server {
                listen 443;
                proxy_pass $name;
                ssl_preread on;
        }
}
```

Ansuel, 2023-11-02:
Did you install the stream module and correctly include it?

npenin, 2023-11-06:
I installed nginx-full (thus I assume I did). The stream module was working before I upgraded.

brada4, 2023-11-08:
Can you show uci.conf.template after modifications?
You know it is replaced at sysupgrade?

npenin, 2023-11-09:
Here it is. It was not replaced as I upgraded using luci which took and backup and restored my files after upgrade.

```
# Consider using UCI or creating files in /etc/nginx/conf.d/ for configuration.
# Parsing UCI configuration is skipped if uci set nginx.global.uci_enable=false
# For details see: https://openwrt.org/docs/guide-user/services/webserver/nginx

worker_processes auto;

user root;

include module.d/*.module;

events {}

stream {
	map $ssl_preread_server_name $name {
		XXX.XXX.XXX 10.68.1.2;
		default localhttps;
	}

	upstream localhttps {
		server localhost:444;
	}

	server {
		listen 443;
		proxy_pass $name;
		ssl_preread on;
	}
}

http {
	server_names_hash_bucket_size 64;

	access_log off;
	log_format openwrt
		'$request_method $scheme://$host$request_uri => $status'
		' (${body_bytes_sent}B in ${request_time}s) <- $http_referer';

	include mime.types;
	default_type application/octet-stream;
	sendfile on;

	client_max_body_size 128M;
	large_client_header_buffers 2 1k;

	gzip on;
	gzip_vary on;
	gzip_proxied any;

	root /www;
	
	#UCI_HTTP_CONFIG
	include conf.d/*.conf;
}
```

just for the ref, here are my uci config sections for my web server

```
config server 'wrt3200acm'
	list listen '444 ssl default_server'
	list listen '[::]:444 ssl default_server'
	list listen 'unix:/var/nginx.sock ssl default_server'
	option server_name 'wrt3200acm.XXX.XXX'
	list include 'restrict_locally'
	list include 'conf.d/*.locations'
	option ssl_certificate '/etc/acme/wrt3200acm.XXX.XXX_ecc/fullchain.cer'
	option ssl_certificate_key '/etc/acme/wrt3200acm.XXX.XXX_ecc/wrt3200acm.XXX.XXX.key'
	option ssl_session_cache 'shared:SSL:32k'
	option ssl_session_timeout '64m'
	option access_log 'off; # logd openwrt'

config server '_redirect2ssl'
	list listen '80'
	list listen '[::]:80'
	option server_name '_redirect2ssl'
	option return '302 https://$host$request_uri'
```

brada4, 2023-11-09:
Check nginx -V , nginx-full includes  ssl pre-read module. 
then 
nginx -T -c /etc/nginx/nginx.conf
or
nginx -T -c /etc/nginx/uci.conf
id put the stream {} into modules.d to keep it safe from uci editing.

npenin, 2023-11-09:
```
# nginx -V
nginx version: nginx/1.25.0 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.11 19 Sep 2023
TLS SNI support enabled
configure arguments: --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::arm --prefix=/usr --conf-path=/etc/nginx/nginx.conf --modules-path=/usr/lib/nginx/modules --with-compat --with-http_ssl_module --with-ipv6 --with-http_stub_status_module --with-http_flv_module --with-http_dav_module --with-http_auth_request_module --with-http_v2_module --with-http_realip_module --with-http_secure_link_module --with-http_sub_module --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-naxsi/naxsi_src --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/lua-nginx --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-dav-ext-module --with-stream=dynamic --with-stream_ssl_module --with-stream_ssl_preread_module --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-ubus-module --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-headers-more --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-brotli --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-rtmp --add-dynamic-module=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0/nginx-ts --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=arm-openwrt-linux-muslgnueabi-gcc --with-cc-opt='-I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nginx-all-module/nginx-1.25.0=nginx-1.25.0 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/usr/lib -L/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/lib -Wl,--gc-sections -znow -zrelro' --without-http_upstream_zone_module --without-pcre2

# nginx -T -c '/etc/nginx/uci.conf'
2023/11/09 06:44:00 [emerg] 12064#0: unknown directive "stream" in /etc/nginx/uci.conf:14
nginx: configuration file /etc/nginx/uci.conf test failed
```

brada4, 2023-11-09:
And so - what is generated there in 14th line? There is one line inserted in template as original is on 13th line....

npenin, 2023-11-09:
That's because the generated uci.conf file adds 1 comment line at the top
```bash
cat /etc/nginx/uci.conf
# This file is re-created when Nginx starts.
# Consider using UCI or creating files in /etc/nginx/conf.d/ for configuration.
# Parsing UCI configuration is skipped if uci set nginx.global.uci_enable=false
# For details see: https://openwrt.org/docs/guide-user/services/webserver/nginx
```

brada4, 2023-11-09:
check nginx -V -> it must include ssl_preread if full is installed.

npenin, 2023-11-09:
it is there in the output I previously shared:
--with-stream_ssl_module --with-stream_ssl_preread_module

brada4, 2023-11-09:
Something in your module.d opens other section where stream { is no longer permitted.

npenin, 2023-11-10:
The only file I have in module.d is the one for luci:
```
root@WRT3200ACM:~# cat /etc/nginx/module.d/luci.module 
load_module /usr/lib/nginx/modules/ngx_http_ubus_module.so;
```

npenin, 2023-11-10:
That was actually the issue, I had to force load the stream module which is now dynamic and not statically loaded with nginx any longer. Creating a stream.module to load the stream module fixed the problem. Thanks a lot for the hint.

closed 2023-11-10 by npenin

leleb, 2023-12-26:
Me too had problems with stream option
https://forum.openwrt.org/t/nginx-full-package-not-full-featured/180117