openwrt/packages #22574: [23.05] freeradius3: freeradius3-common is missing dependencies on libpcre.so.1

ynezz, 2023-11-01:
Environment: armsr/armv8, OpenWrt 23.05

https://forum.openwrt.org/t/freeradius3-packages-missing/176271 led me to https://downloads.openwrt.org/releases/faillogs-23.05/aarch64_generic/packages/freeradius3/compile.txt

```
Package freeradius3-common is missing dependencies for the following libraries:
libpcre.so.1
```

BKPepe, 2023-11-01:
We switched to PCRE2 in https://github.com/openwrt/packages/pull/22532, but it seems that there needs to be done additional work for this. Thanks for reporting, we will look into it!
cc: @crza

Ansuel, 2023-11-01:
@BKPepe i notice this faillogs from some weeks now and was confused... I'm just checking this and I'm finding multiple conflicts...

current freeradius version have only pcre support... 
stable release 3.23 also have only pcre support...

pcre2 support was added in freeradius only in 4.0.0 alpha... backporting the commit is PITA as they restructured the source and moved lots of code... 

I'm not following how the commit was tested and also why we had this downstream... https://github.com/crza/openwrt-packages/commit/1362b7b694d08f89d78507d67d0b15bbb0c9db3b

(that also doesn't work from my test as the header is still found and autoconfig still try to link with pcre...)

Ansuel, 2023-11-01:
@trippleflux i doubt this work and i'm even confused how this links and compile o.O there are some basic function that changed from pcre and pcre2 so probably those function are not getting compiled at all

trippleflux, 2023-11-01:
w00t, I thought much that PCRE2 has much backward API compatibility with predecessor PCRE. I was just trying to help @predator46 at [here ](https://github.com/openwrt/packages/issues/22476) , but yes freeradius 3 is really PCRE centric codes. My Added mod-perl for PCRE seems working fine (verifiable) with command "radiusd -XC", and the daemon runs.

Ansuel, 2023-11-01:
> I thought much that PCRE2 has much backward API compatibility with its predecessor PCRE

nope not at all... actually quite the opposite. but that was done as PCRE API were flawed and very limiting for evolution of the code itself

brada4, 2023-11-02:
To avoid PCRE1 you may ./configure ..... --without-pcre to use libc regex-es
Or add pcre1 package dependency as far as 23.05 is concerned

clutch2sft, 2023-11-06:
So I'm confused because freeradius3 is now missing in multiple architectres for 23.05.  Will it not carry on in the standard repositories given the noted deficiencies?

brada4, 2023-11-12:
You will need a linux like ubuntu in a virtual machine. Then
https://openwrt.org/docs/guide-developer/toolchain/single.package
Either compile disabling pcre dependency (using libc for regular expressions if you use them at all) or adding pcre as package dependency.

closed 2023-11-15 by neheb

predators46, 2024-09-07:
@trippleflux 

i have added perl in freeradius3

```
root@OpenWrt:/etc/freeradius3# radiusd -X
FreeRADIUS Version 3.2.6
Copyright (C) 1999-2023 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius3/dictionary
including dictionary file /etc/freeradius3/dictionary
including configuration file /etc/freeradius3/radiusd.conf
including configuration file /etc/freeradius3/proxy.conf
including configuration file /etc/freeradius3/clients.conf
including files in directory /etc/freeradius3/mods-enabled/
including configuration file /etc/freeradius3/mods-enabled/always
including configuration file /etc/freeradius3/mods-enabled/attr_filter
including configuration file /etc/freeradius3/mods-enabled/chap
including configuration file /etc/freeradius3/mods-enabled/detail
including configuration file /etc/freeradius3/mods-enabled/digest
including configuration file /etc/freeradius3/mods-enabled/eap
including configuration file /etc/freeradius3/mods-enabled/mschap
including configuration file /etc/freeradius3/mods-enabled/exec
including configuration file /etc/freeradius3/mods-enabled/expiration
including configuration file /etc/freeradius3/mods-enabled/expr
including configuration file /etc/freeradius3/mods-enabled/files
including configuration file /etc/freeradius3/mods-enabled/logintime
including configuration file /etc/freeradius3/mods-enabled/pap
including configuration file /etc/freeradius3/mods-enabled/preprocess
including configuration file /etc/freeradius3/mods-enabled/radutmp
including configuration file /etc/freeradius3/mods-enabled/sradutmp
including configuration file /etc/freeradius3/mods-enabled/realm
including configuration file /etc/freeradius3/mods-enabled/unix
including configuration file /etc/freeradius3/mods-enabled/passwd
including configuration file /etc/freeradius3/mods-enabled/perl
including files in directory /etc/freeradius3/policy.d/
including configuration file /etc/freeradius3/policy.d/accounting
including configuration file /etc/freeradius3/policy.d/filter
including configuration file /etc/freeradius3/policy.d/eap
including files in directory /etc/freeradius3/sites-enabled/
including configuration file /etc/freeradius3/sites-enabled/default
including configuration file /etc/freeradius3/sites-enabled/inner-tunnel
main {
 security {
        allow_core_dumps = no
 }
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
}
main {
        name = "radiusd"
        prefix = "/usr"
        localstatedir = "/var"
        sbindir = "/usr/sbin"
        logdir = "/var/log"
        run_dir = "/var/run/radiusd"
        libdir = "/usr/lib/freeradius3"
        radacctdir = "/var/db/radacct"
        hostname_lookups = no
        max_request_time = 30
        proxy_dedup_window = 1
        cleanup_delay = 5
[... 552 lines cut ...]
}
listen {
        type = "acct"
        ipv6addr = ::
        port = 0
   limit {
        max_connections = 16
        lifetime = 0
        idle_timeout = 30
   }
}
listen {
        type = "auth"
        ipaddr = 127.0.0.1
        port = 18120
}
Listening on auth address * port 1812 bound to server default
Listening on acct address * port 1813 bound to server default
Listening on auth address :: port 1812 bound to server default
Listening on acct address :: port 1813 bound to server default
Listening on auth address 127.0.0.1 port 18120 bound to server inner-tunnel
Listening on proxy address * port 56717
Listening on proxy address :: port 52916
Ready to process requests
```