openwrt/packages #22632: tang: running but not doing anything?

tew42, 2023-11-09:
Maintainer: @Tiboris  (copying @nmav due to recent commits)
Environment: Archer A7 v5, OpenWRT 23.05

Description:
tang starts fine, keys are generated, but: default permissions are wrong (process running as user tang can't see jwk files that are generated by/owned by/readable by root)

Everything works as expected when removing `procd_set_param user tang` from the init script

Logs:
nothing, until I added following to init.d (_this should be implemented as standard in any case, perhaps?_)
procd_set_param stdout 1
procd_set_param stderr 1

access attempt http://192.168.1.1:9090
```
Thu Nov  9 15:41:21 2023 daemon.err tangd[3596]: <unknown> GET / => 404 (../src/http.c:128)
Thu Nov  9 15:41:22 2023 daemon.err tangd[3596]: <unknown> GET / => 404 (../src/http.c:128)
```

access attempt http://192.168.1.1:9090/adv
```
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]: <unknown> GET /advCannot load JSON file (/usr/share/tang/db/xxx.jwk); skipping
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]: error text unable to open /usr/share/tang/db/xxx.jwk: Permission denied, line -1, col -1, pos 0
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]: Cannot load JSON file (/usr/share/tang/db/xxx.jwk); skipping
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]: error text unable to open /usr/share/tang/db/xxx.jwk: Permission denied, line -1, col -1, pos 0
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]: Error saving JWK to file (/usr/share/tang/db/xxx.jwk)
Thu Nov  9 15:41:55 2023 daemon.err tangd[3596]:  => 500 (../src/tangd.c:78)
```

```
top
3076     1 tang     S     5136   4%   0% /usr/sbin/tangd -p 9090 -l /usr/share/tang/db
```

```
/usr/share/tang/db# ls -l
-r--r-----    1 root     root           349 Nov  9 15:37 xxx.jwk
-r--r-----    1 root     root           354 Nov  9 15:37 xxx.jwk
```

brada4, 2023-11-09:
chown -R tang db/

nmav, 2023-11-09:
Could you verify that https://github.com/openwrt/packages/pull/22634 fixes the issue?

tew42, 2023-11-09:
Thanks for the quick response!

Weirdly enough, it doesn't work, however - neither with the updated init script, nor by directly issuing chown. It seems related to the fact that the tang user got created as:
```
cat /etc/passwd
tang:x:65536:65536:tang:/var/run/tang:/bin/false
```

If I manually update the UID to 65530, everything works fine, however?! I thought that's only supposed to be an issue for really old kernels...

closed 2023-11-10 by nmav

brada4, 2023-11-10:
@tew42 i see you opened other issue? Is it related?

tew42, 2023-11-10:
in part, yes. the issue with tang itself is (theoretically) fixed in this issue/PR #22634. but there appears to be an underlying and more general issue with openwrt user permission handling (unrelated to tang itself) that I report in openwrt/openwrt#13927 (linked also in the PR)

rugk, 2024-10-31:
Gosh wtf is that. Running `chown` on `/usr/share/tang/db` does not change anything. 

So yeah, as per https://github.com/trinidude4/openwrt/commit/0a32627e3e82c473ffb2ed2463da7175f4202de3 maybe limit yourself to 16bit numbers, which somehow still got a 32bit one generated.

So changing this manually now…

brada4, 2024-10-31:
Every service with auto-generated user and persistent data is affected (well fixed, so that chown does not change to overflown uids)

PR #22634 (2023-11-10): tang: set the right permissions to keys
Maintainer: me
Compile tested: -
Run tested: -

Description:
This corrects the permissions of generated files and sets the right maintainer.

Resolves: #22632
commit 7d77c4d4 (2023-11-10): tang: set the right permissions to keys

Resolves: #22632
commit 8cf81bcc (2023-11-10): tang: set the right permissions to keys

Resolves: #22632
commit 8282c0f8 (2023-11-10): tang: set the right permissions to keys

Resolves: #22632