openwrt/packages #22657: openconnect: no specific route and wrong MTU

A1EF, 2023-11-13:
Maintainer: @nmav
Environment: aarch64, Xiaomi Redmi Router AX6S(mediatek/mt7622), OpenWrt 23.05.0 r23497-6637af95aa

Description:
I using my router as client for remote Openconnect server. It worked perfect with OpenWrt 22.03.5 but stopped after upgrade firmware to 23.05.0. My investigation bring me two issues:

1) no specific route to the ocserv's IP-address after connection is up
2) wrong MTU on the openconnect interface (always 1500 whatever set in the config)

I managed to working connection by manual add the route and change MTU to 1436 but I don't know why it's not working automatically.

nmav, 2023-11-13:
I suggest you provide the logs from ocserv during a connection to identify what goes wrong.

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:59:40 2023 user.notice openconnect: initializing...
Tue Nov 14 18:59:40 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--mtu' '1436' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 19:02:48 2023 daemon.notice openconnect[6938]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 19:05:00 2023 daemon.notice openconnect[6938]: Failed to reconnect to host oc.example.net: Operation timed out
[... cut ...]
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c8:94:02:7d:89:df brd ff:ff:ff:ff:ff:ff
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:53:15 2023 user.notice openconnect: initializing...
Tue Nov 14 18:53:15 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 18:56:21 2023 daemon.notice openconnect[6113]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 18:58:30 2023 daemon.notice openconnect[6113]: Failed to reconnect to host oc.tdm-tech.ru: Operation timed out
[... cut ...]
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c8:94:02:7d:89:df brd ff:ff:ff:ff:ff:ff
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:53:15 2023 user.notice openconnect: initializing...
Tue Nov 14 18:53:15 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 18:56:21 2023 daemon.notice openconnect[6113]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 18:58:30 2023 daemon.notice openconnect[6113]: Failed to reconnect to host oc.tdm-tech.ru: Operation timed out
[... cut ...]
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c8:94:02:7d:89:df brd ff:ff:ff:ff:ff:ff
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:53:15 2023 user.notice openconnect: initializing...
Tue Nov 14 18:53:15 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 18:56:21 2023 daemon.notice openconnect[6113]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 18:58:30 2023 daemon.notice openconnect[6113]: Failed to reconnect to host oc.example.net': Operation timed out
[... cut ...]
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c8:94:02:7d:89:df brd ff:ff:ff:ff:ff:ff
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:53:15 2023 user.notice openconnect: initializing...
Tue Nov 14 18:53:15 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 18:56:21 2023 daemon.notice openconnect[6113]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 18:58:30 2023 daemon.notice openconnect[6113]: Failed to reconnect to host oc.example.net': Operation timed out
[... cut ...]
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c8:94:02:7d:89:df brd ff:ff:ff:ff:ff:ff
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

A1EF, 2023-11-14:
I need to explain more detailed. For example, my ocserv has `192.0.2.1` IP-address and advertise to clients `192.0.2.0/24` network. I have `203.0.113.10` IP-address on my side and `203.0.113.1` is my gateway. I can connect to the ocserv and I see nothing suspicious in the logs. No server's:
```
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: skipping 'pid-file' config option
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:53:16 oc.example.net ocserv[821184]: note: setting 'file' as supplemental config option
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: 6Vmg8H)
Nov 14 18:53:16 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:53:16 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:53:16 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:53:16 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: 6Vmg8H)
Nov 14 18:56:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 18:56:50 oc.example.net ocserv[821261]: GnuTLS error (at worker-vpn.c:883): Error in the pull function.
Nov 14 18:57:16 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 18:57:46 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 18:58:26 oc.example.net ocserv[821184]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 18:58:26 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: 6Vmg8H)
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: skipping 'pid-file' config option
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 18:59:40 oc.example.net ocserv[821341]: note: setting 'file' as supplemental config option
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: sec-mod instance 0 issue cookie
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: using 'pam' authentication to authenticate user (session: R+L4nq)
Nov 14 18:59:40 oc.example.net ocserv[126284]: PAM-auth conv: echo-off, msg: 'Password: '
Nov 14 18:59:40 oc.example.net ocserv[126284]: pam_sss(ocserv:auth): authentication success; logname= uid=0 euid=0 tty= ruser= rhost=203.0.113.10 user=my_username
Nov 14 18:59:40 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 18:59:40 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: skipping 'pid-file' config option
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:00:12 oc.example.net ocserv[821360]: note: setting 'file' as supplemental config option
Nov 14 19:00:22 oc.example.net ocserv[821360]: GnuTLS error (at worker-vpn.c:883): Error decoding the received TLS packet.
Nov 14 19:03:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (210 secs)
Nov 14 19:03:40 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (240 secs)
Nov 14 19:04:10 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (270 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 have not received TCP DPD for long (310 secs)
Nov 14 19:04:50 oc.example.net ocserv[821341]: worker[my_username]: 203.0.113.10 connection timeout (DPD); tearing down connection
Nov 14 19:04:50 oc.example.net ocserv[126284]: sec-mod: temporarily closing session for my_username (session: R+L4nq)
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: skipping 'pid-file' config option
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: vhost:default: setting 'pam' as primary authentication method
Nov 14 19:05:10 oc.example.net ocserv[821470]: note: setting 'file' as supplemental config option
Nov 14 19:05:10 oc.example.net ocserv[126284]: Loading group configuration '/etc/ocserv/user_groups//vpnusers'
Nov 14 19:05:10 oc.example.net ocserv[126284]: sec-mod: initiating session for user 'my_username' (session: R+L4nq)
```
neither client's:
```
Tue Nov 14 18:53:15 2023 user.notice openconnect: initializing...
Tue Nov 14 18:53:15 2023 user.notice openconnect: executing 'openconnect 'oc.example.net' '-i' 'vpn-oc' '--non-inter' '--syslog' '--script' '/lib/netifd/vpnc-script' '--no-dtls' '--protocol' 'anyconnect' '--authgroup' 'vpnusers' '-u' 'my_username' '--passwd-on-stdin''
Tue Nov 14 18:56:21 2023 daemon.notice openconnect[6113]: CSTP Dead Peer Detection detected dead peer!
Tue Nov 14 18:58:30 2023 daemon.notice openconnect[6113]: Failed to reconnect to host oc.example.net': Operation timed out
[... cut ...]
In other hand, I can to connect toward my ocserv from my laptop without any problems: I get specific route and right MTU automatically:
```bash
:~ $ ip l sh dev tun0
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1434 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 500
    link/none
~ $ ip r
default via 172.16.0.1 dev wlan0 proto dhcp src 172.16.0.74 metric 600
172.16.0.0/24 dev wlan0 proto kernel scope link src 172.16.0.74 metric 600
192.0.2.0/24 dev tun0 scope link
192.0.2.1 via 172.16.0.1 dev wlan0 src 172.16.0.74 metric 600
```

brada4, 2023-11-14:
mtu fix location will be corrected in future firewall4 release. Quick glimpse here.
https://github.com/openwrt/openwrt/issues/12112#issuecomment-1712739131

nmav, 2023-11-16:
From what I see I'm not sure it is ocserv who is creating the wrong routing table.

A1EF, 2023-11-16:
But I cant to connect from my laptop (Arch Linux installed) without any problem: I get right route and right MTU from the server by my openconnect client. May be it's my unluck, but on previous OpenWrt release all works fine.

nr23730, 2024-03-25:
I'm probably facing the same MTU problem on my OpenWrt. Routes work fine, but my linux machine sets mtu to 1400 while OpenWrt sets it to 1500.

I'm using r25661-bf4c04a4d0 on a GL.inet GL-X3000.

kor7win, 2024-10-30:
It seems that this problem is still not solved. MTU is still always set to 1500. I have to solve the problem now with a crutch, set MTU as soon as the status changes to enabled. I hope that this will be fixed.

zengxinhui, 2024-11-14:
My solution is to update `/lib/netifd/vpnc-script` and add a line after ip address is set:

`ip li set dev $TUNDEV up mtu $MTU`

```bash
do_connect() {
        if [ -n "$CISCO_BANNER" ]; then
                logger -t openconnect "Connect Banner:"
                echo "$CISCO_BANNER" | while read LINE ; do logger -t openconnect "|" "$LINE" ; done
        fi

        proto_init_update "$TUNDEV" 1

        if [ -n "$INTERNAL_IP4_MTU" ]; then
                MTU=$INTERNAL_IP4_MTU
        fi

        if [ -z "$MTU" ]; then
                MTU=1412
        fi

        proto_add_ipv4_address "$INTERNAL_IP4_ADDRESS" 32 "" "$INTERNAL_IP4_ADDRESS"

        ip li set dev $TUNDEV up mtu $MTU

```