openwrt/packages #22658: mwan3: Can't run java program in mwan3 use

250king, 2023-11-13:
Maintainer: Florian Eckert <fe@dev.tdt.de> \ Aaron Goodman <aaronjg@alumni.stanford.edu>
Environment: Intel(R) Celeron(R) CPU 3865U @1.80GHz \ LuCI openwrt-22.03 branch git-23.292.78392-9f66674
Description:
I want to use run a java program without mwan3's rule, but unluckly
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fb1bb4a77c7, pid=3976, tid=3977
#
# JRE version:  (21.0.1+12) (build )
# Java VM: OpenJDK 64-Bit Server VM (21.0.1+12-29, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0xe997c7]  Thread::SpinAcquire(int volatile*, char const*)+0x7
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
#

---------------  S U M M A R Y ------------

Command Line: /root/app/client.jar

Host: Intel(R) Celeron(R) CPU 3865U @ 1.80GHz, 2 cores, 3G, iStoreOS 22.03.5
Time: Mon Nov 13 17:49:59 2023 UTC elapsed time: 0.016777 seconds (0d 0h 0m 0s)

---------------  T H R E A D  ---------------

Current thread is native thread

Stack: [0x00007fb1ba422000,0x00007fb1ba522000],  sp=0x00007fb1ba520ab8,  free space=1018k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
V  [libjvm.so+0xe997c7]  Thread::SpinAcquire(int volatile*, char const*)+0x7
V  [libjvm.so+0xe98d3f]  Thread::Thread()+0x8f
V  [libjvm.so+0x8f5879]  JavaThread::JavaThread()+0x19
V  [libjvm.so+0xea64b6]  Threads::create_vm(JavaVMInitArgs*, bool*)+0x256
V  [libjvm.so+0x977a8e]  JNI_CreateJavaVM+0x4e
C  [libjli.so+0x42cb]  JavaMain+0x8b
C  [libjli.so+0x7e49]  ThreadJavaMain+0x9

siginfo: si_signo: 11 (SIGSEGV), si_code: 2 (SEGV_ACCERR), si_addr: 0x00007fb1bbcd76bc

Registers:
RAX=0x0000000000000000, RBX=0x00007fb1b4028928, RCX=0x0000000000000002, RDX=0x0000000000000001
RSP=0x00007fb1ba520ab8, RBP=0x00007fb1ba520ae0, RSI=0x0000000000000000, RDI=0x00007fb1bbcd76bc
R8 =0x00007fb1b40286e0, R9 =0x00007fb1b40286e0, R10=0x00007fb1ba520aa0, R11=0x00007ffd1eff4080
R12=0x0000000000000000, R13=0x00007fb1bbcd73c0, R14=0x00007fb1bbcd76bc, R15=0x0000000000000000
RIP=0x00007fb1bb4a77c7, EFLAGS=0x0000000000010246, CSGSFS=0x002b000000000033, ERR=0x0000000000000007
  TRAPNO=0x000000000000000e

Top of Stack: (sp=0x00007fb1ba520ab8)
0x00007fb1ba520ab8:   00007fb1baf68521 00007fb1b40286e0
0x00007fb1ba520ac8:   00007fb1b40286e0 00007fb1b40286e0
0x00007fb1ba520ad8:   00007fb1ba520ba0 00007fb1ba520b00
0x00007fb1ba520ae8:   00007fb1bb4a6d3f 00007fb1b40286e0
0x00007fb1ba520af8:   00007fb1ba520b80 00007fb1ba520b40
0x00007fb1ba520b08:   00007fb1baf03879 00007fb1b4028440
0x00007fb1ba520b18:   00007fb1ba520d3f 00007fb1ba520b80
0x00007fb1ba520b28:   00007fb1b40286e0 00007fb1ba520ba0
0x00007fb1ba520b38:   0000000000000000 00007fb1ba520d20
0x00007fb1ba520b48:   00007fb1bb4b44b6 0000000000000000
0x00007fb1ba520b58:   0000000000000000 0000000000000000
0x00007fb1ba520b68:   0000000000000000 00007fb1ba520c68
0x00007fb1ba520b78:   00007fb1ba520ca9 0000000000000001
0x00007fb1ba520b88:   0000018bc9cdf382 0000000000000000
0x00007fb1ba520b98:   0000000000000000 0000000000000100
0x00007fb1ba520ba8:   0000000000000000 0000000000000000
0x00007fb1ba520bb8:   0000000000000000 0000000000000000
0x00007fb1ba520bc8:   00007fb1bb66ba62 0000000000000000
0x00007fb1ba520bd8:   0000000000000000 0000000000000000
[... 330 lines cut ...]

Steal ticks since vm start: 0
Steal ticks percentage since vm start:  0.000

CPU: total 2 (initial active 2) 
CPU Model and flags from /proc/cpuinfo:
model name	: Intel(R) Celeron(R) CPU 3865U @ 1.80GHz
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust smep erms invpcid mpx rdseed smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm arat pln pts hwp hwp_notify hwp_act_window hwp_epp

Online cpus: 0-1
Offline cpus: 
BIOS frequency limitation: <Not Available>
Frequency switch latency (ns): 0
Available cpu frequencies: <Not Available>
Current governor: powersave
Core performance/turbo boost: <Not Available>

Memory: 4k page, physical 3961376k(3458424k free), swap 0k(0k free)
Page Sizes: 4k

vm_info: OpenJDK 64-Bit Server VM (21.0.1+12-29) for linux-amd64 JRE (21.0.1+12-29), built on 2023-10-05T13:27:09Z by "mach5one" with gcc 11.2.0

END.
```
I have made sure that these is no any problem with the java program.

feckert, 2023-11-14:
Have you ever tried to run the programme without the mwan3? Does it work then?

250king, 2023-11-14:
It works without mwan3. But I want to the jar to sent the request with specific network interface so I need to use the jar with mwan3.

brada4, 2023-11-14:
> OPENWRT_RELEASE="iStoreOS 22.03.5 2023102711"

This is OpenWRT bug tracker, not random fork back patting forum.

feckert, 2023-11-15:
I can't test this and therefore can't help to reproduce your problem as I am not useing java. I can only imagine that it might take too long because the call is invoked in the hotplug context. Maybe it would be better if you have a java service and send a short command in the hotplug via an interface and not load the whole JAVAVM right away!!! Why don't you ask in the OpenWrt forum?

closed 2023-11-15 by feckert

250king, 2023-11-16:
Will, now I give up to use Java and use Curl, but now I have a problem.
```
curl "http://xxx.xxx.xxx.xxx/ticket.cgi?wlanuserip=xxx.xxx.xxx.xxx&wlanacip=xxx.xxx.xxx.xxx&portal_node=http%3A%2F%2Fxxx.xxx.xxx.xxx%3A10002" --header "User-Agent: CCTP/Android8_vpn/2082" --header "Client-ID: 00000000-0000-0000-0000-000000000000" --header "CDC-Checksum: 9f89c84a559f573636a47ff8daed0d33" --header "Algo-ID: 00000000-0000-0000-0000-000000000000" --data "00000000-0000-0000-0000-000000000000"
```
It can run in terminal. But if I add ```mwan3 use wan0```, it always parases wrong.
```
Running 'curl "http://xxx.xxx.xxx.xxx/ticket.cgi?wlanuserip=xxx.xxx.xxx.xxx&wlanacip=xxx.xxx.xxx.xxx&portal_node=http%3A%2F%2Fxxx.xxx.xxx.xxx%3A10002" --header "User-Agent: CCTP/Android8_vpn/2082" --header "Client-ID: 00000000-0000-0000-0000-000000000000" --header "CDC-Checksum: 9f89c84a559f573636a47ff8daed0d33" --header "Algo-ID: 00000000-0000-0000-0000-000000000000" --data "00000000-0000-0000-0000-000000000000"' with DEVICE=eth1mac1 SRCIP=xxx.xxx.xxx.xxx FWMARK=0x3f00 FAMILY=ipv4
curl: (3) URL using bad/illegal format or missing URL
curl: (6) Could not resolve host: CCTP
curl: (3) URL using bad/illegal format or missing URL
curl: (3) URL using bad/illegal format or missing URL
curl: (3) URL using bad/illegal format or missing URL
curl: (6) Could not resolve host: text
```

250king, 2023-11-16:
Are you sure it isn't app's problem and is system's?
Or you can give me a solution to execute this command.

brada4, 2023-11-16:
both are not from here according to your own report. Feel free to open a fresh issue if you can repeat problem like in x86 VM and openwrt.