openwrt/packages #22664: The following problems occur when compiling and running make defconfig locally with version 23.05.0 code on Ubuntu 22.04.3 system

Jewell1024, 2023-11-15:
### Describe the bug

tmp/.config-package.in:33662:error: recursive dependency detected!
tmp/.config-package.in:33662:	symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
/home/ay/openwrt23/feeds/packages/lang/python/pymysql/Config.in:4:	symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

### OpenWrt version

. /etc/openwrt_release && echo $DISTRIB_REVISION

### OpenWrt target/subtarget

. /etc/openwrt_release && echo $DISTRIB_TARGET

### Device

cat /tmp/sysinfo/model

### Image kind

Official downloaded image

### Diffconfig
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jefferyto, 2023-11-15:
Ping @commodo, @cotequeiroz - perhaps https://github.com/openwrt/packages/pull/22301 should be cherry picked for 23.05?

commodo, 2023-11-15:
> Ping @commodo, @cotequeiroz - perhaps #22301 should be cherry picked for 23.05?

https://github.com/openwrt/packages/pull/22668

jefferyto, 2023-11-16:
Should be fixed by #22668 - closing.

closed 2023-11-16 by jefferyto