openwrt/packages #22698: ddns not starting on boot

geraudcr, 2023-11-18:
Looks lots of people (including me) are having issues to get ddns to start on boot:
https://forum.openwrt.org/t/ddns-no-start-after-boot/94769/2
https://forum.openwrt.org/t/dynamic-dns-is-not-running-upon-power-on-or-reboot/150807/7
And many more..

Based on https://openwrt.org/docs/techref/initscripts, the boot function is claled at boot once and by default is calling the start one. But for some reason the ddns init script boot() is overwriten and not returning right away, causing the issue.
https://github.com/openwrt/packages/blob/7fd3364f028abd470baa4f242ffe70b954ef0c2b/net/ddns-scripts/files/etc/init.d/ddns#L5

I am not familiar enough with github to fix it myself, but I hope someone can submit the fix.

Thank you!!!

feckert, 2023-12-21:
The ddns in openwrt is event based. A service does not have to be started when booting.

closed 2023-12-21 by feckert

qianxuu, 2024-12-06:
I encountered the same issue. Based on my testing, the IP address source in the advanced settings needs to remain set to the default "Network." If it is set to "Interface," DDNS will not start automatically on boot. After reverting it to the default setting, the auto-start on boot works normally.