openwrt/packages #22699: ddns-scripts: package using wget instead of curl for providers asking for https result in "badauth" with all settings correct

larieu, 2023-11-18:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Hostname	OpenWrt-pi4
Model	Raspberry Pi 4 Model B Rev 1.5
Architecture	ARMv8 Processor rev 3
Target Platform	bcm27xx/bcm2711
Firmware Version	OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256
Kernel Version	5.15.137

Description:

if the provider of service asks to use https instead of http and if the ddclient use wget instead of curl then the result of the update IP is "badauth"

as a a workaround uninstall the "wget-ssl" package and the service will start to work ( because ddns will ask for curl instead ) 

```
Format code blocks by wrapping them with pairs of ```
```

brada4, 2023-11-19:
Which of 50 providers asks for that? Like error messago or how do you know so?

larieu, 2023-11-20:
I am using eurodns
( I have tried to add it in the mainstream but my git abilities are not up
to scratch - I'll try at some point to go again)
which is asking for https
( which should be the way to go in the future - as DNS is inching toward
...)

the fact that openwrt implementation is going preferentially to wget-nossl
instead of curl when both are available ... no clue why
As a curious monkey I have a clue why wget-nossl is performing badly but
needs testing.
When time will permit, I'll try to install the wget-ssl (only, then with
both ssl and non ssl and then with curl together in all combinations and
see how ddns package responds to the combinations )
- as all the changes takes time ( to wait to see if the change of IP is
handled correctly without intervention and go through all combinations ...
) I need to find a proper time for it

I found this problem because I have used a spare Pi to implement the
routing in my home and then I ordered a new one with POE HAT and
transferred it to this new one
In the process I have added wget-nossl to transfer some configs through an
internal web site
On the "old" one the ddns worked flawless and on the "new" one ... no (
with almost identical configs )
Then noticed in the logs of ddns that the old one used curl and the new one
wget
By simply uninstalling wget-nossl the problem was solved

As a good practice in linux world I wrote it down on the place where
somewhere else will find it when needed and probably save a lot of other
people time with it

brada4, 2023-11-20:
Default busybox wget supports ssl?

larieu, 2023-11-20:
https://openwrt.org/packages/pkgdata/wget-ssl

brada4, 2023-11-20:
https://github.com/openwrt/packages/blob/f52f4370700afeba5a7b085ab6a6693745c89eba/net/ddns-scripts/files/usr/lib/ddns/dynamic_dns_functions.sh#L717

Seems you need to set use_https because it is not derived from url ?

PS your git abilities are in line, just some patience needed.