openwrt/packages #22727: mwan3: rules with source/dest IP doesn't take effect

eddiehsu88, 2023-11-22:
mwan3: rules with source/dest IP doesn't take effect 

Maintainer: @feckert
Environment: OpenWrt 23.05, x86, mwan3 2.11.8-1

Description:
rules with source IP or destination IP doesn't work. 
Once add this kind of rules in configuration file, the command "mwan3 rules" shows records without name and assigned IP and the corresponding iptables rule incorrect as well. 

```
➜  ~ more /etc/config/mwan3 

config globals 'globals'
	option mmx_mask '0x3F00'

config interface 'wan'
	option family 'ipv4'
	option reliability '2'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option timeout '4'
	option interval '60'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '223.5.5.5'
	list track_ip '180.76.76.76'

config interface 'wan6'
	option enabled '0'
	list track_ip '2606:4700:4700::1001'
	list track_ip '2606:4700:4700::1111'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
	option family 'ipv6'
	option reliability '2'

config interface 'wanb'
	option family 'ipv4'
	option reliability '1'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option timeout '4'
	option interval '60'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '180.76.76.76'
	list track_ip '223.5.5.5'

config interface 'wanb6'
	option enabled '1'
	option family 'ipv6'
	option reliability '1'
	option initial_state 'online'
	list track_ip '100.65.0.1'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
[... 336 lines cut ...]
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 MARK xset 0x100/0x3f00
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x100/0x3f00 ! match-set mwan3_rule_ipv4_OVPN src,src MARK and 0xffffc0ff
    0     0 mwan3_policy_wan_wanb  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
    0     0 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 del-set mwan3_rule_ipv4_OVPN src,src
    0     0 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 add-set mwan3_rule_ipv4_OVPN src,src

Chain mwan3_rule_https (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  521 33300 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 MARK xset 0x300/0x3f00
    5   816 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x300/0x3f00 ! match-set mwan3_rule_ipv4_https src,src MARK and 0xffffc0ff
    5   816 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 MARK xset 0x100/0x3f00
    5   816 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x100/0x3f00 ! match-set mwan3_rule_ipv4_https src,src MARK and 0xffffc0ff
    5   816 mwan3_policy_balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
  521 33300 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 del-set mwan3_rule_ipv4_https src,src
  521 33300 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 add-set mwan3_rule_ipv4_https src,src

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 mwan3_rule_OVPN  udp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 9443,953 mark match 0x0/0x3f00
  521 33300 mwan3_rule_https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 443 mark match 0x0/0x3f00
 7744  498K mwan3_policy_balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
    0     0 mwan3_policy_wanb_wan  all  --  *      *       0.0.0.0/0            0.0.0.0              mark match 0x0/0x3f00
    0     0 mwan3_policy_wanb_wan  all  --  *      *       0.0.0.0              0.0.0.0/0            mark match 0x0/0x3f00
```

brada4, 2023-11-22:
`iptables --version`
`nft list ruleset`
You need to be on one side between nftables and iptables kernel modules.

eddiehsu88, 2023-11-23:
> `iptables --version` `nft list ruleset` You need to be on one side between nftables and iptables kernel modules.

Please find following. btw, the rules with ports are good. 
```
➜  ~ iptables --version
iptables v1.8.8 (nf_tables)
➜  ~ nft list ruleset
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy drop;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname { "eth1", "eth2" } jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
		iifname "docker0" jump input_docker comment "!fw4: Handle docker IPv4/IPv6 input traffic"
		jump handle_reject
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname { "eth1", "eth2" } jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		iifname "docker0" jump forward_docker comment "!fw4: Handle docker IPv4/IPv6 forward traffic"
		jump upnp_forward comment "Hook into miniupnpd forwarding chain"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
		oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
		oifname { "eth1", "eth2" } jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
		oifname "docker0" jump output_docker comment "!fw4: Handle docker IPv4/IPv6 output traffic"
	}

	chain prerouting {
		type filter hook prerouting priority filter; policy accept;
		iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
		iifname "docker0" jump helper_docker comment "!fw4: Handle docker IPv4/IPv6 helper assignment"
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		jump accept_to_docker comment "!fw4: Accept lan to docker forwarding"
		jump accept_to_lan
	}

	chain helper_lan {
	}
[... 337 lines cut ...]
}
# Warning: table ip nat is managed by iptables-nft, do not touch!
table ip nat {
	chain DOCKER {
		iifname "docker0" counter packets 0 bytes 0 return
		iifname "br-119147ac2f74" counter packets 0 bytes 0 return
	}

	chain POSTROUTING {
		type nat hook postrouting priority srcnat; policy accept;
		oifname != "docker0" ip saddr 172.17.0.0/16 counter packets 0 bytes 0 xt target "MASQUERADE"
		oifname != "br-119147ac2f74" ip saddr 172.18.0.0/16 counter packets 0 bytes 0 xt target "MASQUERADE"
	}

	chain PREROUTING {
		type nat hook prerouting priority dstnat; policy accept;
		xt match "addrtype" counter packets 5207754 bytes 332091664 jump DOCKER
	}

	chain OUTPUT {
		type nat hook output priority -100; policy accept;
		ip daddr != 127.0.0.0/8 xt match "addrtype" counter packets 0 bytes 0 jump DOCKER
	}
}
```

brada4, 2023-11-23:
config looks relistic, same seems emitted into iptables, just that if nftables pick on some packet half way.

eddiehsu88, 2023-11-25:
> config looks relistic, same seems emitted into iptables, just that if nftables pick on some packet half way.

updated in last comment

brada4, 2023-11-25:
it mixes xt mark and nft mark whic+ do not exchange markings between them, you need to either install or uninstall nft mark and conntrack  modules

brada4, 2023-11-25:
i.e. 
opkg install kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-nat kmod-nf-nat6

check after if any more `xt` bridge rules get emitted, and try to figure out respective kmod-nf\*
You need to keep iptables kmod-s even for iptables-nft that you are using.

alex3594, 2024-03-07:
Hello!
I have the same issue with the source/dest ip addresses in mwan3 rules. Is there any solution found to make it work?

brada4, 2024-03-07:
Somehow you need to get all mark statements either to xt xor nft, mix assumes zero mark usually.
That is keep iptables-nft , install modules in one side and uninstall in other until all mark statements are xt or no xt.
Nice if you share findings to add dependencies and conflicts in the package.

litori, 2024-05-13:
Can not get source/destination IP to work as well.  I'm not sure how to get things on one side or another.  Any help would be greatly appreciated.  Thanks.

brada4, 2024-05-13:
You need to 
` nft list ruleset | grep xt `

Then install all respective nftables modules:
https://wiki.nftables.org/wiki-nftables/index.php/Supported_features_compared_to_xtables

litori, 2024-05-13:
``` nft list ruleset | grep xt
# Warning: table ip mangle is managed by iptables-nft, do not touch!
                xt match "set" counter packets 0 bytes 0 xt target "MARK"
                xt match "set" counter packets 5484 bytes 521021 xt target "MARK"
                xt match "set" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 counter packets 26364 bytes 6398902 xt target "CONNMARK"
                counter packets 26412 bytes 6402934 xt target "CONNMARK"
                iifname "eth0" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                iifname "eth0" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                iifname "eth0" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                iifname "eth0" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 478 bytes 118477 xt target "MARK"
                iifname "eth1" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                iifname "eth1" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 6 bytes 375 xt target "MARK"
                iifname "eth1" xt match "set" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                iifname "eth1" meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "statistic" xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
# Warning: table ip6 mangle is managed by iptables-nft, do not touch!
                xt match "set" counter packets 0 bytes 0 xt target "MARK"
                xt match "set" counter packets 11 bytes 940 xt target "MARK"
                xt match "set" counter packets 0 bytes 0 xt target "MARK"
                meta l4proto ipv6-icmp xt match "icmp6" counter packets 3 bytes 168 return
                meta l4proto ipv6-icmp xt match "icmp6" counter packets 4 bytes 536 return
                meta l4proto ipv6-icmp xt match "icmp6" counter packets 22 bytes 1544 return
                meta l4proto ipv6-icmp xt match "icmp6" counter packets 22 bytes 1448 return
                meta l4proto ipv6-icmp xt match "icmp6" counter packets 0 bytes 0 return
                meta mark & 0x00003f00 == 0x00000000 counter packets 177 bytes 21219 xt target "CONNMARK"
                counter packets 177 bytes 21219 xt target "CONNMARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
                meta mark & 0x00003f00 == 0x00000000 xt match "comment" counter packets 0 bytes 0 xt target "MARK"
```
Sorry i'm not very good at this.  this is my output of `nft list ruleset | grep xt`, what modules do i need?  Thank you in advance.

Oniicyan, 2024-08-16:
`cat /var/run/mwan3/iptables_log/iptables-set_user_rules-ipv4.dump`

You can see the src/dest IPs are correct.

See also
https://github.com/openwrt/openwrt/issues/11169
https://bugzilla.netfilter.org/show_bug.cgi?id=1642
https://git.netfilter.org/iptables/commit/?id=eddbb27651b93ac6f329bf8113223e7360ea7613

spadesoface, 2024-08-20:
> https://bugzilla.netfilter.org/show_bug.cgi?id=1642

is there any way to implement this patch( 39589b3edb ) without fully rebuilding openwrt from sources?