openwrt/packages #22744: openvswitch: daemon fails to start

AnttiNykanen, 2023-11-24:
Maintainer: @yousong 
Environment: x86_64, OpenWRT 23.05.2

Description:

OVS package versions:
```
kmod-openvswitch - 5.15.137+2.17.0-3
kmod-openvswitch-gre - 5.15.137+2.17.0-3
openvswitch - 2.17.0-3
openvswitch-common - 2.17.0-3
openvswitch-libofproto - 2.17.0-3
openvswitch-libopenvswitch - 2.17.0-3
openvswitch-libovsdb - 2.17.0-3
openvswitch-ovsdb - 2.17.0-3
openvswitch-vswitchd - 2.17.0-3
```

I have been using OVS with the following config

```
config ovs 'ovs'
        option ca '/etc/openvswitch/example_ca.crt'
        option cert '/etc/openvswitch/example_cert.crt'
        option key '/etc/openvswitch/example_key.crt'

config ovn_northd 'north'
        option disabled '1'

config ovn_controller 'controller'
        option disabled '1'

config ovs_bridge
        option disabled '1'
        option name 'my-bridge'
        option controller 'tcp:192.168.0.1'
        option fail_mode 'standalone'

config ovs_port
        option disabled '1'
        option bridge 'my-bridge'
        option port 'ovs-port1'
        option ofport '1'
        option tag '123'
        option type 'internal'
```

... so in other words, I want the init script to just start the ovs daemon, all configuration is then done with the ovs tools.

This used to work but it doesn't anymore. OVS doesn't start from the init script. The init script seems to do nothing except to write the following **in syslog**

```
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch: Syntax: /etc/init.d/openvswitch [command]
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch: Available commands:
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  start           Start the service
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  stop            Stop the service
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  restart         Restart the service
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  reload          Reload configuration files (or restart if service does not implement reload)
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  enable          Enable service autostart
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  disable         Disable service autostart
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  enabled         Check if service is started on boot
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:  status          Get status information
Fri Nov 24 12:22:42 2023 daemon.notice procd: /etc/init.d/openvswitch:
```

No matter which command I try to give to the init script, the result is always that the usage information is written into syslog - nothing else happens.

alex3594, 2023-12-21:
Hello! I observe the same issue with the OVS. Is there any workaround?

astrcomp, 2023-12-31:
@alex3594 @AnttiNykanen Hello! Is file /usr/share/openvswitch/scripts/ovs-ctl empty?

alex3594, 2023-12-31:
> @alex3594 @AnttiNykanen Hello! Is file /usr/share/openvswitch/scripts/ovs-ctl empty? 
Hello! Yes, the file is empty. I made a workaround by putting /usr/share/openvswitch/scripts/ files into the "files" directory in my Openwrt build.

AnttiNykanen, 2024-01-02:
Hi,

@astrcomp yes - as I explained, in my application openwrt init scripts just start ovs daemon(s), all other configuration is done via external tools. 

I will experiment with @alex3594 workaround

I wonder what changed that causes this?

AnttiNykanen, 2024-01-02:
Investigated a bit more: while running` make -j1 V=sc package/openvswitch/compile`, this happens

```
PYTHONPATH=$PYTHONPATH":"./python python3.11 ./build-aux/soexpand.py -I. < utilities/ovs-ctl.in | \
  python3.11 ./build-aux/dpdkstrip.py --nodpdk | \
  sed \
    -e 's,[@]PKIDIR[@],/var/lib/openvswitch/pki,g' \
    -e 's,[@]LOGDIR[@],/var/log/openvswitch,g' \
    -e 's,[@]DBDIR[@],/etc/openvswitch,g' \
    -e 's,[@]PYTHON3[@],python3.11,g' \
    -e 's,[@]RUNDIR[@],/var/run/openvswitch,g' \
    -e 's,[@]VERSION[@],2.17.0,g' \
    -e 's,[@]localstatedir[@],/var,g' \
    -e 's,[@]pkgdatadir[@],/usr/share/openvswitch,g' \
    -e 's,[@]sysconfdir[@],/etc,g' \
    -e 's,[@]bindir[@],/usr/bin,g' \
    -e 's,[@]sbindir[@],/usr/sbin,g' \
    -e 's,[@]abs_builddir[@],/home/annykane/gregw/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.17.0,g' \
    -e 's,[@]abs_top_srcdir[@],/home/annykane/gregw/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.17.0,g' \
  > utilities/ovs-ctl.tmp
Traceback (most recent call last):
  File "/home/annykane/gregw/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.17.0/./build-aux/soexpand.py", line 17, in <module>
    from build import soutil
ImportError: cannot import name 'soutil' from 'build' (/home/annykane/gregw/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/build/__init__.py)
mv -f utilities/ovs-ctl.tmp utilities/ovs-ctl
```

soutil is at `openvswitch-2.17.0/python/build/soutil.py`, but the pythonpath seems incorrect as this isn't found. thus we end up with empty ovs-ctl and other scripts

drrobot333, 2024-01-31:
> @alex3594 @AnttiNykanen Hello! Is file /usr/share/openvswitch/scripts/ovs-ctl empty? @AnttiNykanen in config all disabled

I have same problem that /usr/share/openvswitch/scripts/ovs-ctl is empty. 
Is there any way to mitigate this problem?

yousong, 2024-02-21:
> ImportError: cannot import name 'soutil' from 'build' (/home/annykane/gregw/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/build/__init__.py)

Hi @AnttiNykanen , it seems `build` in the hostpkg dir overrides what what should be in the source code.  Please consider give #23490 a try to see if it fixes the issue for you.

closed 2024-02-22 by yousong