openwrt/packages #22745: https-dns-proxy: SIGSEGV (libcurl)

systemcrash, 2023-11-24:
So I've had about a week of uptime on 23.05.2. And for whatever reason, when I got up this morning, I had no Internet connection. A restart of the network service fixed that. 

I don't know whether this is symptomatic or causal (it seemed occur around the start of the network loss), but the below was in the kernel log. Seems to indicate that libcurl can crash if there's no Internet. 

Could potentially be related to the other bug report #22674

Maintainer: @stangri 
Environment: 

Hostname	asdf
Model	TP-Link Archer C7 v2
Architecture	Qualcomm Atheros QCA9558 ver 1 rev 0
Target Platform	ath79/generic
Firmware Version	OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256
Kernel Version	5.15.137
Local Time	2023-11-24 13:36:30
Uptime	7d 12h 3m 38s

Event occurred around Uptime	7d 9h 28m 11s

Description:

```
[635291.037571] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 72003aa4
[635291.047311] epc = 77cef10f in libcurl.so.4.8.0[77ccc000+5f000]
[635291.053380] ra  = 77cef105 in libcurl.so.4.8.0[77ccc000+5f000]
[635316.628610] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 006874a8
[635316.638339] epc = 77d3910f in libcurl.so.4.8.0[77d16000+5f000]
[635316.644750] ra  = 77d39105 in libcurl.so.4.8.0[77d16000+5f000]
```

brada4, 2023-11-24:
few lines leading/tailing THE EVENT would be nice to have from logread.

systemcrash, 2023-11-24:
Absolutely not a thing to do with it. JUST those lines from those modules.
Everything else unrelated.

stangri, 2023-11-25:
@systemcrash what's the version of the proxy binary?

systemcrash, 2023-11-25:
That would be the one included when 23.05.2 was released.

2023-10-25-1

systemcrash, 2023-11-28:
```
# curl --version
-ash: curl: not found
# ls -l /usr/lib/libcurl.so.4*
lrwxrwxrwx    1 root     root            16 Nov 14 14:38 /usr/lib/libcurl.so.4 -> libcurl.so.4.8.0
-rwxr-xr-x    1 root     root        336868 Nov 14 14:38 /usr/lib/libcurl.so.4.8.0
```

Built my system using firmware-selector.

brada4, 2023-11-28:
opkg list-installed | grep -e curl -e s-p

systemcrash, 2023-11-28:
libcurl4
8.4.0-2

brada4, 2023-11-28:
Can you provide information on BOTH involved packages?

stangri, 2023-11-28:
> libcurl4 8.4.0-2

If you can, please install curl, because the `--version` command also outputs information (the SSL library and ng-libs installed/used) which may be useful for the upstream devs.

Also, the package config information (specifically resolvers you use) might also be of interest.

Once it's all here, I'll tag upstream devs.

systemcrash, 2023-11-29:
Now that I have shell access:
```
# opkg list-installed | grep -e curl -e s-p
https-dns-proxy - 2023-10-25-1
libcurl4 - 8.4.0-2
luci-app-https-dns-proxy - 2023-10-25-1

# curl --version
curl 8.4.0 (mips-openwrt-linux-gnu) libcurl/8.4.0 mbedTLS/2.28.5 nghttp2/1.57.0
Release-Date: 2023-10-11
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
```

stangri, 2023-11-29:
@aarond10 @baranyaib90 any insight into this binary error?

baranyaib90, 2023-11-29:
From this much information I have no idea. But my guess is, that it is not a libcurl issue.
The truth is, that I'm using the proxy on a 64bit raspberry and not on OpenWRT.
Finding the root cause of this would need debug build and coredump analyzation with gdb (but at least a usable backtrace).

aarond10, 2023-11-29:
I have an archer c7. Let me update and see what happens to mine.

aarond10, 2023-11-29:
```
# curl --version
curl 8.4.0 (mips-openwrt-linux-gnu) libcurl/8.4.0 mbedTLS/2.28.5 nghttp2/1.57.0
Release-Date: 2023-10-11
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
# https-dns-proxy -V
2023-10-25-4
```

I'll see how I go.

aarond10, 2023-11-29:
```
# ifdown wan
# ping www.facebook.com
ping: bad address 'www.facebook.com'
# dmesg
<nothing in logs>
# ps | grep https
 4701 nobody    2844 S    /usr/sbin/https-dns-proxy -r https://cloudflare-dns.
 4702 nobody    2844 S    /usr/sbin/https-dns-proxy -r https://dns.google/dns-
...
# ifup wan
<wait>
# ping www.facebook.com
<redacted but all good>
```

Can't reproduce.

systemcrash, 2023-11-29:
> ```
> # curl --version
> curl 8.4.0 (mips-openwrt-linux-gnu) libcurl/8.4.0 mbedTLS/2.28.5 nghttp2/1.57.0
> Release-Date: 2023-10-11
> Protocols: file ftp ftps http https mqtt
> Features: alt-svc HSTS HTTP2 [...]

As long as our versions differ it may never be reproducible. I hypothesized that this traceback was due to the other "crash" I logged. The one that triggered the "ubus syntax" in the logs.

stangri, 2023-11-30:
> I hypothesized that this traceback was due to the other "crash" I logged. The one that triggered the "ubus syntax" in the logs.

The ubus call error was from the init script and the SIGSERV is from the binary itself. If you disable forcing local DNS the ubus call error should disappear if you want to make sure there's no connection.

systemcrash, 2023-12-01:
I meant to say "precipitated by". I see no plausible direct connection, but the earlier crashes may have had an effect.

closed 2023-12-28 by systemcrash