openwrt/packages #22760: syncthing: duplicate command options due to missing option_cb cases

efahl, 2023-11-26:
Maintainer: @brvphoenix @BKPepe 
Environment: generic, introduced in 22.03

Description:
Introduced in 
https://github.com/openwrt/packages/commit/94f7097207653adba1ffc002d1255ebce1022357

Bug visible at
https://github.com/openwrt/packages/blob/master/utils/syncthing/files/syncthing.init#L17

Missing case values in syncthing init script generate duplicate parameters during service construction, causing errors at startup (--home is in the wrong place or not created; --gui-address is not as specified).  Following ps output line has been wrapped so you can more clearly see the duplicates.

```
$ uci show syncthing
syncthing.syncthing=syncthing
syncthing.syncthing.enabled='1'
syncthing.syncthing.gui_address='http://10.1.2.1:8384'
syncthing.syncthing.home='/mnt/sda2/syncthing'
...

$ ps ww | grep syncthing
11996 syncthin 1220m SN   /usr/bin/syncthing serve 
--gui-address=http://0.0.0.0:8384 --home=/etc/syncthing --logfile=/mnt/sda2/syncthing/syncthing.log --no-browser 
--gui-address=http://10.1.2.1:8384 --home=/mnt/sda2/syncthing --log-max-old-files=7 --log-max-size=1048576 --no-default-folder
```

If we edit `/etc/init.d/syncthing` and change line 17, adding both gui_address and home, from
```
	enabled|macprocs|nice|user|logfile)
```
to
```
	enabled|gui_address|home|logfile|macprocs|nice|user)
```

Then we get proper behavior with those parameters being supplied just once, and --home directory is created, chowned and accessed correctly.
```
$ /etc/init.d/syncthing restart

$ ps ww | grep syncthing
30786 syncthin 1220m SN   /usr/bin/syncthing serve --gui-address=http://10.1.2.1:8384 --home=/mnt/sda2/syncthing --logfile=/mnt/sda2/syncthing/syncthing.log --no-browser --log-max-old-files=7 --log-max-size=1048576 --no-default-folder
```

BKPepe, 2023-11-27:
> 94f7097

Hey!

Your issue report is clear and understable. You seems to know what is going on and that you took time to write it down. Thanks! :)

I am thinking would you mind to send PR to fix it?

efahl, 2023-11-27:
Sure, I'm trying to get a handle on the OpenWrt PR workflow, so let me know if anything is amiss.

https://github.com/openwrt/packages/pull/22767

jefferyto, 2023-12-13:
I believe this is complete, closing.

closed 2023-12-13 by jefferyto