openwrt/packages #22777: rtorrent-rpc: Install doesn't reate an initscript

racingcirc, 2023-11-27:
Maintainer: @neheb 
Environment: ARMV7, IPQ806x, 23.05.2

Description: opkg install of rtorrent-rpc doesn't result in the creation of an initscript, which is anomalous to most other official packages available for 23.05.2

brada4, 2023-11-28:
have not found any init script for this ncurses ui.

racingcirc, 2023-12-01:
Hey, thanks for looking at this! To answer your question, here's an example of a very similar package with a complex procd initscript https://github.com/openwrt/packages/blob/openwrt-23.05/net/transmission/files/transmission.init

And here's some other examples of some other packages in this category along with their initscripts: https://github.com/search?q=repo%3Aopenwrt%2Fpackages+path%3A.init&type=code

Since this is the heavier, RPC-enabled rtorrent package, it's probably a safe assumption that the ncurses TUI is not the most common way it'll be used.

One might argue that it's totally doable for even a non-developer to spend 2 or 3 days reading through the (incomplete) rtorrent documentation to learn the particulars of the script that is the main configuration file, then adapt it for use with Busybox/Ash (since their documentation is for Bash). Once this config file is saved as rtorrent.rc, then one can create a new Initscript before proceeding to set up all the remainder of the RPC client and whatever other tools (perhaps screen, as is being suggested above) that one would need to, in order to use a headless bittorrent client.

But is there any benefit to leaving all these tasks to every individual user instead of making them part of the OpenWRT package build system like one would expect any fully-functioning system to be?

I understand that developer capacity is super limited, and perhaps it's more important to fix bugs and/or support more devices. In that case, I would recommend pruning the gigantic repository of "official" packages that are available in OPKG. One can still compile one's own packages, after all!

As things stand right now someone unfamiliar with code has no way of knowing how much time it's going to take to get this rtorrent package to the point of it being usable after installing it from OPKG.

With all that out of the way, I really want to say thank you to all the devs for your stellar work in enabling this community!

Please feel free to disposition this issue in whichever way is most appropriate (=

racingcirc, 2023-12-01:
If I knew how to do any of that I gladly would have, before even creating this issue. I'm slowly learning though, so maybe in a few months from now I'll have the chops necessary...