openwrt/packages #22828: curl: libcurl only depends on libmbedtls

zero13th, 2023-12-06:
Maintainer: @stangri
Environment: OpenWrt 23.05.2, mediatek, ramips

Description:
libcurl4 only depends on libmbedtls12
Ref: #16674 
> mbedTLS and OpenSSL is widely used.

Need curl to be able to detect if libopenssl (or other lib-ssl) is installed so it will not install libmbedtls as a dependency.
This will benefit users who only use libopenssl (or other lib-ssl) on their router.
Will save another ~250kB storage space.

```
:~# opkg list-installed | grep 'ssl\|tls'
libopenssl-conf - 3.0.12-1
libopenssl-legacy - 3.0.12-1
libopenssl3 - 3.0.12-1
libustream-openssl20201210 - 2023-02-25-498f6e26-1
luci-ssl-openssl - git-23.024.33244-34dee82
openssl-util - 3.0.12-1
wpad-openssl - 2023-09-08-e5ccbfc6-6

:~# opkg install curl
Installing curl (8.4.0-2) to root...
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/packages/curl_8.4.0-2_aarch64_cortex-a53.ipk
Installing libmbedtls12 (2.28.5-2) to root...
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/base/libmbedtls12_2.28.5-2_aarch64_cortex-a53.ipk
Installing libnghttp2-14 (1.57.0-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/packages/libnghttp2-14_1.57.0-1_aarch64_cortex-a53.ipk
Installing libcurl4 (8.4.0-2) to root...
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/packages/libcurl4_8.4.0-2_aarch64_cortex-a53.ipk
Configuring libmbedtls12.
Configuring libnghttp2-14.
Configuring libcurl4.
Configuring curl.

:~# opkg info libmbedtls12
Package: libmbedtls12
Version: 2.28.5-2
Depends: libc
Provides: libmbedtls
Status: install ok installed
Section: libs
Architecture: aarch64_cortex-a53
Size: 252922
Filename: libmbedtls12_2.28.5-2_aarch64_cortex-a53.ipk
Description: The aim of the mbedtls project is to provide a quality, open-source
 cryptographic library written in C and targeted at embedded systems.
 This package contains the mbedtls library.
Installed-Time: 1701886853

:~# opkg info curl
opkg info curl
Package: curl
Version: 8.4.0-2
Depends: libc, libcurl4
Status: install user installed
Section: net
Architecture: aarch64_cortex-a53
Size: 66838
Filename: curl_8.4.0-2_aarch64_cortex-a53.ipk
Description: A client-side URL transfer utility
Installed-Time: 1701886858

 :~# opkg info libcurl4
Package: libcurl4
Version: 8.4.0-2
Depends: libc, libmbedtls12, libnghttp2-14, ca-bundle
Provides: libcurl
Status: install ok installed
Section: libs
Architecture: aarch64_cortex-a53
Size: 204933
Filename: libcurl4_8.4.0-2_aarch64_cortex-a53.ipk
Description: A client-side URL transfer library
Installed-Time: 1701886858
```
Thank you.

stangri, 2023-12-06:
If you want curl to depend on a different library, you'll need to set such a dependency for `libcurl` thru `make menuconfig` and rebuild `curl`.

closed 2023-12-06 by stangri

bam80, 2024-10-06:
> Feel free to contribute code, but from what has been discussed, it's not possible.

@stangri 
What is not possible exactly, you mean curl needs to be rebuilt against the right ssl library?
Can we have both `curl-openssl` and `curl-mbedtls`, then?

I wonder how one could swap mbedtls to openssl currently, without rebuild anything and ending up with only one ssl library in the system, to minimize space required?

bam80, 2024-10-06:
> If you want curl to depend on a different library, you'll need to set such a dependency for `libcurl` thru `make menuconfig` and rebuild `curl`.

@stangri you mean "rebuild `libcurl`", maybe?

stangri, 2024-10-06:
@bam80 I'm no longer a maintainer for curl, please direct your questions to the current maintainer.