openwrt/packages #22850: clamav: Invalid character in name field `config/clamav-milter`

pradyparanjpe, 2023-12-09:
Maintainer: @dhewg
Environment: Linux OpenWrt 5.15.134 #0 Mon Oct 9 21:45:35 2023 mips GNU/Linux

Description:
An error is thrown every time I try to upgrade ```opkg upgr```.
```
uci: Parse error (invalid character in name field) at line 1, byte 37
```

I narrowed this error down to `clamav-milter` config after scanning. The following scanning for-loop was gratefully picked from [here](https://forum.openwrt.org/t/uci-parse-error-invalid-command-at-line-1-byte-0-should-contain-the-file-name-as-well-as/90859).

``` bash
for CONF in /etc/config/* ; do uci show "${CONF##*/}" > /dev/null || echo "${CONF}"; done
```

```
uci: Parse error (invalid character in name field) at line 1, byte 37
/etc/config/clamav-milter
```

**Clamav-milter configuration appears to have invalid an character.**

Upon uninstallation of clamav (and hence, removal of the corresponding config file), the above error is not thrown.
Upon fresh installation with the `default config`, the error reappears.

brada4, 2023-12-09:
please restore file https://github.com/openwrt/packages/blob/master/net/clamav/files/clamav.config and compare to one on your system. Looks like some text editor destryed tick'ark in 1st line and slash in 2nd file path turns up after seectiontitle where newline is expected. Post the file, compressed on the system

BKPepe, 2023-12-18:
Closing this due to inactivity.

closed 2023-12-18 by BKPepe

pradyparanjpe, 2024-01-02:
Sorry for replying late, I was travelling.
Thanks for your suggesting, but
I have uninstalled clamav as I found myself not using it much...

vincejv, 2024-03-14:
The issue is malformed config file in `clamav-milter`, I've raised it already before in the forums: https://forum.openwrt.org/t/uci-parse-problem/143302/5

https://github.com/openwrt/packages/blob/09feb1885c136774e64cc743147910036a31c5b8/net/clamav/files/clamav-milter.config#L1