openwrt/packages #22882: libxcrypt-4.4.33 on 23.05 fails to build; known fixed bug, can we bump the version?

Wabuo, 2023-12-12:
Maintainer: @neheb &  @graysky2
Environment: (put here arch, model, OpenWrt version)

Description: libxcrypt-4.4.33 fails to build, known bug fixed upstream, fixed in libxcrypt 4.4.35

It fails with:
```
BEGIN failed--compilation aborted at ./build-aux/scripts/expand-selected-hashes line 28.
configure: error: bad value 'solaris' for --enable-hashes
```

See my [forum post](https://forum.openwrt.org/t/build-error-libxcrypt-configure-error-bad-value-solaris-for-enable-hashes/180382), which lead to this [gentoo](https://bugs.gentoo.org/909650) bug report.
Thanx again, [etpolanne](https://forum.openwrt.org/u/retpolanne) for digging that up.

BKPepe, 2023-12-18:
Feel free to create pull request to fix that.

Wabuo, 2023-12-20:
Sure but to which version?

I've been using the one from master, as in this one https://github.com/openwrt/packages/pull/21767, and it's been working fine so far.

Do we just grab that one as in what are the rules and best practices on updating a stable release branch?
In my opinion, it makes sense to just go for the one in master because 4.4.36 isn't far off and seems to work fine.

I also don't know how to make a pull request towards a specific branch, but I'm sure I can figure that one out.

BKPepe, 2023-12-20:
Fixed by https://github.com/openwrt/packages/commit/5b4067ccb6fedc4308a54dc05c21749ee7cb1c05, closing.

closed 2023-12-20 by BKPepe