openwrt/packages #22883: QoS Nftables restrictions not taking effect

xh0823, 2023-12-12:
target platform x86/64

Firmware version OpenWrt SNAPSHOT r25055-0bc3d2e0cf

Kernel version 6.1.67

QoS Nftables, Both IP and mac speed limits have no effect

brada4, 2023-12-12:
Which package you are talking about?

xh0823, 2023-12-14:
> 你说的是哪个包？

nft-qos

xh0823, 2023-12-14:
> Which package you are talking about?

nft-qos

brada4, 2023-12-14:
check `nft list ruleset` for QoS rules
(luci/status/firewall also works)
lucky if QoS rules have 'counter' so you can see if limiting activity occurs.
firewall (flowtable) offload if enabled bypasses any rules once the state is established, disable offload if you want nft-qos, if you need offload for performance you must use other QoS.

xh0823, 2023-12-14:
> check `nft list ruleset` for QoS rules (luci/status/firewall also works) lucky if QoS rules have 'counter' so you can see if limiting activity occurs. firewall (flowtable) offload if enabled [...]

Whether it is enabled or disabled, disabling uninstallation is invalid. Thank you.

brada4, 2023-12-14:
Are you talking about this particular issue? 
- do you find counters at qos rules
- and disable firewall offload as this is incompatible with nft-qos
- invalid what?

brada4, 2023-12-14:
if you dont speak English, write in 1x English and  1x your language, someone can help.