openwrt/packages #22889: rust/host: undefined reference to `zmemcpy` in libz-sys while compiling cargo tool

egorenar, 2023-12-13:
Maintainer: @1715173329
Environment: ipq806x, ARMv7, OpenWrt commit `9b598ec8d558`

Description:

```
make V=sc -j$(($(nproc)+1)) package/rust/host/compile
....

  = note: /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `flush_pending':
          deflate.c:(.text.flush_pending+0x69): undefined reference to `zmemcpy'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `deflate':
          deflate.c:(.text.deflate+0x8d3): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate+0x9e4): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate+0x1052): undefined reference to `zmemzero'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `read_buf':
          deflate.c:(.text.read_buf+0x63): undefined reference to `zmemcpy'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `lm_init':
          deflate.c:(.text.lm_init+0x61): undefined reference to `zmemzero'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `fill_window':
          deflate.c:(.text.fill_window+0x83): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.fill_window+0x3c7): undefined reference to `zmemzero'
          /usr/bin/ld: deflate.c:(.text.fill_window+0x479): undefined reference to `zmemzero'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(deflate.o): in function `deflate_stored':
          deflate.c:(.text.deflate_stored+0x22e): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate_stored+0x385): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate_stored+0x423): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate_stored+0x4ae): undefined reference to `zmemcpy'
          /usr/bin/ld: deflate.c:(.text.deflate_stored+0x64c): undefined reference to `zmemcpy'
          /usr/bin/ld: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/build/x86_64-unknown-linux-gnu/stage1-tools/x86_64-unknown-linux-gnu/release/deps/liblibz_sys-ca7901d93337d4cf.rlib(trees.o):trees.c:(.text._tr_stored_block+0x24b): more undefined references to `zmemcpy' follow
          collect2: error: ld returned 1 exit status

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `cargo` (bin "cargo") due to previous error
Build completed unsuccessfully in 0:19:45
make[3]: *** [Makefile:110: /home/egorenar/Repositories/openwrt-test/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.0-src/.built] Error 1
make[3]: Leaving directory '/home/egorenar/Repositories/openwrt-test/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#491.57#105.92#1263.64
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:126: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/egorenar/Repositories/openwrt-test'
make[1]: *** [package/Makefile:122: /home/egorenar/Repositories/openwrt-test/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/egorenar/Repositories/openwrt-test'
make: *** [/home/egorenar/Repositories/openwrt-test/include/toplevel.mk:232: world] Error 2
```

egorenar, 2023-12-13:
I checked the output of `objdump -d /path/to/libz-sys/rlib` and` zutil.o` contains no `zmemcpy`.

From zutil.h

```
#ifdef HAVE_MEMCPY
#  ifdef SMALL_MEDIUM /* MSDOS small or medium model */
#    define zmemcpy _fmemcpy
#    define zmemcmp _fmemcmp
#    define zmemzero(dest, len) _fmemset(dest, 0, len)
#  else
#    define zmemcpy memcpy
#    define zmemcmp memcmp
#    define zmemzero(dest, len) memset(dest, 0, len)
#  endif
#else
   void ZLIB_INTERNAL zmemcpy(Bytef* dest, const Bytef* source, uInt len);
   int ZLIB_INTERNAL zmemcmp(const Bytef* s1, const Bytef* s2, uInt len);
   void ZLIB_INTERNAL zmemzero(Bytef* dest, uInt len);
#endif
```

Looks like it should be mapped to the regular `memcpy` but for some reason it's not.
Maybe HAVE_MEMCPY check failed ?

egorenar, 2023-12-13:
hmm, libz-sys uses cmake to build zlib library, and OpenWrt updated cmake recently, let me check this.

egorenar, 2023-12-13:
no, reverting cmake didn't help.

egorenar, 2023-12-13:
reverting back to rust `1.73` doesn't help.

egorenar, 2023-12-14:
solved, my fault, `RUSTC_WRAPPER` was set in my shell.
for some reason, cached build artifacts caused this problem.
removing sccache's cache solved the problem.

closed 2023-12-14 by egorenar

1715173329, 2023-12-14:
Probably we should unexport this env variable if sccache is not enabled explicitly.

commit 9f010109 (2024-01-01): rust: unexport host sccache env variable

Users might configure their own env variables on the host, and sometimes
it can lead build failure or unexpected behavior.

Fixes: #22889
commit cbc8d7bc (2024-01-10): rust: unexport host sccache env variable

Users might configure their own env variables on the host, and sometimes
it can lead build failure or unexpected behavior.

Fixes: #22889

(cherry picked from commit 9f01010958b9f57ef51903f69c7223eaefabf139)