openwrt/packages #22895: modemmanager: Quectel modem RM502Q-AE is not working with modemmanager, but modem is fully working outside of modemmanager

bengeek2, 2023-12-15:
Maintainer: @feckert
Environment: Raspberry Pi 4B 4gb, OpenWRT 23.05

Description:
I am using a Quectel RM502Q-AE with a usb3 board and a T-Mobile sim. The modem/board is fully functional using a raspberry pi running the ROOTer project, and this modem/board is fully working with drivers from quectel when connected to my windows machine.

Steps to replicate:
1. Attempt to connect the modem via mmcli -m 9 -e
2. Open syslog in Luci

```
Thu Dec 14 18:26:27 2023 authpriv.info dropbear[11336]: Child connection from 192.168.10.29:63208
Thu Dec 14 18:26:27 2023 authpriv.notice dropbear[11336]: Password auth succeeded for 'root' from 192.168.10.29:63208
Thu Dec 14 18:27:00 2023 daemon.info [8783]: <info>  [modem9] state changed (disabled -> enabling)
Thu Dec 14 18:27:05 2023 daemon.warn [8783]: <warn>  [modem9] failed checking whether autoconnect is disabled or not: Transaction timed out
Thu Dec 14 18:27:10 2023 daemon.warn [8783]: <warn>  [modem9] port cdc-wdm0 timed out 2 consecutive times
Thu Dec 14 18:27:10 2023 daemon.info [8783]: <info>  [modem9] power state updated: on
Thu Dec 14 18:27:20 2023 daemon.warn [8783]: <warn>  [modem9] port cdc-wdm0 timed out 3 consecutive times
Thu Dec 14 18:27:20 2023 daemon.warn [8783]: <warn>  [modem9] could not load SIM identifier: QMI operation failed: Transaction timed out (3 retries left)
Thu Dec 14 18:27:31 2023 daemon.warn [8783]: <warn>  [modem9] port cdc-wdm0 timed out 4 consecutive times
Thu Dec 14 18:27:31 2023 daemon.warn [8783]: <warn>  [modem9] could not load SIM identifier: QMI operation failed: Transaction timed out (2 retries left)
Thu Dec 14 18:27:42 2023 daemon.warn [8783]: <warn>  [modem9] port cdc-wdm0 timed out 5 consecutive times
Thu Dec 14 18:27:42 2023 daemon.warn [8783]: <warn>  [modem9] could not load SIM identifier: QMI operation failed: Transaction timed out (1 retries left)
Thu Dec 14 18:27:53 2023 daemon.warn [8783]: <warn>  [modem9] port cdc-wdm0 timed out 6 consecutive times
Thu Dec 14 18:27:53 2023 daemon.warn [8783]: <warn>  [modem9] could not load SIM identifier: QMI operation failed: Transaction timed out
Thu Dec 14 18:27:53 2023 daemon.info [8783]: <info>  [modem9] SIM identifier has changed: 8901260484745831208 -> <none>
```

brada4, 2023-12-15:
> running the ROOTer

is it via luci-proto-qmi or modemmanager in presence or absence of modeswitch?

`opkg list-installed | grep qmi`

feckert, 2024-02-12:
No reaction from  issue creator