openwrt/packages #22966: adblock 4.1.5 with knot-resolver works inconsistently

ghinlon, 2023-12-22:
I'm trying to understand how adblock works with knot-resolver, but I'm running into some issues.

Here's what I know so far:

- Adblock seems to generate a file called `/etc/kresd/adb_list.overall`.
- My adblock was working, then stopped, then started again randomly.
- I'm not sure if I need to manually add `policy.add(policy.rpz(policy.DENY, '/etc/kresd/adb_list.overall'))` to knot-resolver's config, or if adblock handles that automatically.

Can anyone shed some light on:

- How adblock and knot-resolver interact under the hood?
- Whether I need to manually configure knot-resolver for adblocking?

brada4, 2023-12-23:
is there a bug reported  in package or you need help in forums?

ghinlon, 2023-12-24:
My bad. It's not a bug. I'm just seeking for help.

brada4, 2023-12-24:
either can be forwarder for other. Benchmark in place.

closed 2023-12-29 by ghinlon