openwrt/packages #22968: sing-box: building with embedded Tor support should require libevent2

szescxz, 2023-12-23:
Maintainer: @brvphoenix 
Environment: _no precision necessary_

Description:
See also https://github.com/SagerNet/sing-box/issues/614#issuecomment-1558536203
```
Finding targets

Building targets
berty.tech/go-libtor/libtor
# berty.tech/go-libtor/libtor
In file included from ../../../../../dl/go-mod-cache/berty.tech/go-libtor@v1.0.385/libtor/linux_tor_src_app_main_main.go:10:
../../../../../dl/go-mod-cache/berty.tech/go-libtor@v1.0.385/libtor/../linux/tor/src/../src/app/main/main.c:87:10: fatal error: event2/event.h: No such file or directory
   87 | #include <event2/event.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

szescxz, 2023-12-23:
Hmm, seems like the bulid is still broken anyway, as I've also encountered another error mentioned in the linked issue.
```
Finding targets

Building targets
berty.tech/go-libtor/libtor
# berty.tech/go-libtor/libtor
In file included from ../../../../../dl/go-mod-cache/berty.tech/go-libtor@v1.0.385/libtor/../linux/tor/src/../src/lib/tls/tortls_openssl.c:48,
                 from ../../../../../dl/go-mod-cache/berty.tech/go-libtor@v1.0.385/libtor/linux_tor_src_lib_tls_tortls_openssl.go:10:
../../../../../dl/go-mod-cache/berty.tech/go-libtor@v1.0.385/libtor/../linux/tor/src/../src/lib/tls/tortls_openssl.c:1176:3: error: missing binary operator before token "1"
 1176 |   SSL_OP_ALLOW_UNSAFE_LEGACY_RENEGOTIATION != 0
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
I'll leave this issue opened anyway since we do need `libevent2` as the dependency.

<details>
<summary>This second build error should probably be fixed at the upstream side.</summary>

Apparently this error is due to [https://github.com/berty/go-libtor being unmaintained](https://github.com/berty/go-libtor/issues/8). For explanation on the error itself, see https://gitlab.torproject.org/tpo/core/tor/-/merge_requests/390.
We should probably wait for `sing-box` switching to another fork of `go-libtor` with this issue fixed, e.g. https://github.com/ooni/go-libtor which [they also used for embedded Tor on mobile](https://github.com/SagerNet/sing-box/blob/5cd230bea07747699c8ef0abfd76f04d8fb42baf/outbound/tor_embed_mobile.go#L7).

</details>

brada4, 2023-12-23:
What do you want to do with problematic upstream package?

szescxz, 2023-12-23:
> What do you want to do with problematic upstream package?

_In case you're not aware of this GitHub feature - any line that begins with a ▶ (after formatting of course) can be expanded by clicking on it._
tl;dr Upstream should probably fix the dependency (either use another active fork or maintain their own fork).

Please also note that everything I mentioned in this issue only applies when the embedded Tor build option is selected.

brada4, 2023-12-23:
Yeah, type keyboard, letters show on TV, i know that. What is your proposed solution? ooni mothership supports just ancient openssl.

szescxz, 2023-12-24:
> ooni mothership supports just ancient openssl.

That would beyond both my knowledge and the original scope of this issue, so besides waiting for someone capable to take it over, I don't have any proposal.

brada4, 2023-12-24:
sing-box embeddedtor does  not function upstream. links you threw here link ti IA babble. You can open a ne@ issu€ if upstream fixes it.

szescxz, 2023-12-24:
I know, that's exactly why [I said](https://github.com/openwrt/packages/issues/22968#issuecomment-1868207240)
> I'll leave this issue opened anyway since we do need `libevent2` as the dependency.

The [original issue](https://github.com/openwrt/packages/issues/22968#issue-2054646079) is still valid if upstream fixes it.

brada4, 2023-12-24:
your link blames upstream, it is worth opening issue or PR if upstream fixes a problem, not to main%ain eternal grie,ance. There is zero to fix at present moment. Please remove all serverfault ai generated references. Thanks

szescxz, 2023-12-25:
You're totally missing the point when I say "original issue". Let me rephrase myself in simple words then.

When I select [`SINGBOX_WITH_EMBEDDED_TOR`](https://github.com/openwrt/packages/blob/5e87484968a9bae1a11acb256fe233457c4c9b62/net/sing-box/Makefile#L59), I expect `menuconfig` should automatically select `PACKAGE_libevent2` as well.
This is sufficient enough to close this issue.

Anything else, like you said, should be done upstream by either submitting an issue or PR there.

brada4, 2023-12-25:
that can be done when ssl issue is fixed upstream, kind of indifferent if dependencies are less or more correct moving towards the inevitable build fail.