openwrt/packages #22971: i2pd: bad `chown` call not setting pidfile ownership, causing service to fail

Sir-Photch, 2023-12-23:
Maintainer: @r4sas
Environment: Qualcomm Atheros QCA9558 ver 1 rev 0, TP-Link Archer C7 v1, OpenWrt 23.05.2 r23630-842932a63d

https://github.com/openwrt/packages/blob/ad5a83c4a4f90b8b170db2780c9ddb9239759037/net/i2pd/files/i2pd.init#L60

The group `adm` may not exist, shouldn't it be `$USER:$GROUP` here?
On my device, the pidfile has ownership `root:root`, which might be because of a failed `chown` call.

Thus, on service start, I get:
```
daemon.err i2pd[1234]: i2pd: Could not create pid file /var/run/i2pd.pid: Permission denied
```

This issue does not persist when I change ownership to `:$GROUP` in the `init.d` script.

r4sas, 2023-12-23:
Hello. Thank you for report.
First at all, package maintainer in OpenWRT is @yangfl, not me. I'm PurpleI2P team internal maintainer, but can help here too.
Secondly, that is strange that there is no `adm` group in your firmware... I have it on 23.05.0-rc3 firmware.

Sir-Photch, 2023-12-23:
I just checked, and the `adm`-group does exist...

Maybe, the ownership is changed back to `root:root` on restart of the service?

r4sas, 2023-12-23:
May be. But script must touch pidfile and change rights on every service start.