openwrt/packages #22977: mosquitto-ssl - Segmentation fault on subscribe

ASerbinski, 2023-12-23:
The following config works perfectly on mosquitto 1.6.15 (Rocky Linux 8) and mosquitto 2.0.14 (Fedora), but results in frequent Segmentation fault on OpenWRT 23.05.2;

```
max_inflight_messages 1

listener 8883
cafile /etc/mosquitto/ca.crt
certfile /etc/mosquitto/dom.crt
keyfile /etc/mosquitto/dom.key

require_certificate true

log_dest syslog

log_type error
log_type warning
log_type notice
log_type information
log_type all

allow_anonymous false
password_file /etc/mosquitto/pwfile
```

The segfault seems to always happen when the client sends a SUBSCRIBE;

```
Sat Dec 23 14:14:19 2023 daemon.debug mosquitto[12688]: 1703358859: Received SUBSCRIBE from {redacted}
Sat Dec 23 14:14:19 2023 daemon.debug mosquitto[12688]: 1703358859: 	prop1/thermostat/furnace/log-ac1/$format (QoS 1)
```

The SUBSCRIBES are sent while the client is connecting. It does some wildcard and recursive subscriptions, and somewhere in the process it often crashes.

I'm not sure how to debug this, since segmentation fault doesn't really give much to go on. Its also not going to be an upstream bug since the same configuration works on other machines without the segmentation fault.

brada4, 2023-12-23:
backtrace or at least a kernel message of the crash?

brada4, 2023-12-23:
(and which architecture, and versions of all packages in backtrace)

ASerbinski, 2023-12-23:
I'll need instructions on how to collect those. Nothing is appearing in the kernel or system logs with respect to the crash, and if I start it from terminal, it outputs simply "Segmentation fault".

brada4, 2023-12-23:
kernel log via dmesg command shows segfaults, including disk file and offset in file it happened.

ASerbinski, 2023-12-23:
Nothing there.
And wouldn't that normally be just for crashes within the kernel?

brada4, 2023-12-23:
qualcommax/ipq807x
Dynalink DL-WRX36

did you install package from default repos? Like opkg
`opkg list-installed` will show versions.

ASerbinski, 2023-12-23:
Everything is from default repos.

brada4, 2023-12-24:
Looks sane, like all currentoackages intact.

ASerbinski, 2023-12-24:
Exactly. So where do I go from here? Any way you know of to coerce it into handing over a trace?

brada4, 2023-12-24:
strace, gdb, check wiki.

ASerbinski, 2023-12-24:
Hmm. It may have been an issue with mosquitto 2.0.17. I installed 2.0.18 from the snapshot repository and I'm having a hard time reproducing the problem now.

brada4, 2023-12-24:
ping @neheb

neheb, 2023-12-24:
You want it back ported?

ASerbinski, 2023-12-24:
It would probably make a lot of sense. I've been trying for over an hour to reproduce the crash with 2.0.18 and have failed. With 2.0.17, I could make it happen within 3 tries over maybe 30 seconds.

neheb, 2023-12-24:
done.

BKPepe, 2023-12-29:
Btw, why are you trying to post something or everything, where there isnt anything related to mosquitto? You can see it even yourself that it was not helpful.

Backported even to OpenWrt 22.03. 

Closing as I think 4 days was enough to reproduce it and test the version 2.0.18 on the stable branches.

closed 2023-12-29 by BKPepe