openwrt/packages #22980: avahi: Warning about nss-mdns

cherinyy, 2023-12-24:
Maintainer: @thess
Environment: OpenWrt 23.05.2 on mediatek/filogic

Description:

I noticed a warning in the logs:

```
WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
```

`nss-mdns` does not appear to be present in the OpenWrt packages feed now.

So maybe consider disabling this warning for OpenWrt until `nss-mdns` is supported?

predators46, 2023-12-26:
@cherinyy 

can you try this and build it

https://github.com/predators46/nss-mdns

cherinyy, 2024-07-12:
No response was received for a long time.

closed 2024-07-12 by cherinyy

Vitaliy86, 2024-08-11:
Problem not solved

luxflow, 2024-11-24:
as i know nss-mdns cannot be used because of musl which openwrt depends on
https://wiki.musl-libc.org/future-ideas.html#mDNS_and_alternate_hostname_database_backends