openwrt/packages #22994: Dnsdist: Base package does not support setting server policy

killgufo, 2023-12-27:
Maintainer: @Habbie @rgacogne
Environment: OpenWRT 23.05.0 and 23.05.2 on ath79 (f9k1115-v2)

Description:
Since the upgrade from 22.03 my dnsdist configuration can't include the _setServerPolicy_ directive. 
Unfortunately I can't run the dnsdist-full package as I only have 9Mb free for the overlay FS and dnsdist-full takes about 7.7Mb which doesn't allow me to install all the other packages I need.

The problem is simple to reproduce, I have 3 servers

> newServer({address="10.100.170.5", pool="allsrv", checkInterval=1, maxCheckFailures=2, checkType="A", checkName="google.com", mustResolve=true, order=1})    -- main srv
> newServer({address="9.9.9.9", pool="allsrv", checkInterval=2, checkType="A", checkName="google.com", mustResolve=true, order=2})       -- quad9
> newServer({address="209.244.0.3", pool="allsrv", checkInterval=2, checkType="A", checkName="google.com", mustResolve=true, order=2})   -- level3

which are accepted, but when I try to _setServerPolicy(firstAvailable)_ I get the following error:

> dnsdist 1.8.2 comes with ABSOLUTELY NO WARRANTY. This is free software, and you are welcome to redistribute it according to the terms of the GPL version 2
> Allowing remote access to the console while libsodium support has not been enabled is not secure, and will result in cleartext communications
> Calling setKey() while libsodium support has not been enabled is not secure, and will result in cleartext communications
> Added downstream server 10.100.170.5:53
> Added downstream server 9.9.9.9:53
> Added downstream server 209.244.0.3:53
> Fatal Lua error: [string "chunk"]:33: attempt to call global 'setServerPolicy' (a nil value)
> stack traceback:
>	[string "chunk"]:33: in main chunk

As I am not running the full version it's difficult to troubleshoot as I can't connect interactively to the daemon and see more than what's in the output.

An LDD over the binary might help (?) 
`# ldd /usr/bin/dnsdist
	/lib/ld-musl-mips-sf.so.1 (0x77d6c000)
	libluajit-5.1.so.2 => /usr/lib/libluajit-5.1.so.2 (0x77ce4000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x77b0e000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x77aec000)
	libssl.so.3 => /usr/lib/libssl.so.3 (0x77a58000)
	libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x77750000)
	libh2o-evloop.so.1.0 => /usr/lib/libh2o-evloop.so.1.0 (0x776ec000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x776ba000)
	libatomic.so.1 => /lib/libatomic.so.1 (0x77698000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77666000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x77d6c000)
	libz.so.1 => /usr/lib/libz.so.1 (0x77644000)`

rgacogne, 2023-12-28:
This is happening because the `dnsdist` package is now built with a small subset of the most important features, to save disk space and memory, and custom load-balancing policies have not been selected. We could include them back by removing the `-DDISABLE_POLICIES_BINDINGS` flag, as we were apparently wrong in assuming these were not enabled by OpenWrt users.
Out of curiosity, did you have any trouble with the package size on 22.03? Because before the `dnsdist`/`dnsdist-full` split `dnsdist` was quite big already.

killgufo, 2023-12-28:
Thanks for the change to include load balancing policies, it will allow me to stay on 23.05 without reverting back to 22.03!

I didn't have trouble with the size of the package before, but didn't check how close I've been to a disk-full either as it "just worked"™ and I didn't question the status until the upgrade this week where I ended up without free space and started chasing down this issue.

For the record, the reason why I'm using this policy is that I want a specific internal DNS (e.g. pihole) to be used exclusively unless unreachable, where I revert back to public DNSs. 
Maybe there's a different way to implement the same but this worked for me for the past years without troubles.

Cheers and thanks again for the quick fix!

closed 2023-12-29 by neheb