openwrt/packages #22996: sqm-scripts: Stopping/Starting 100+ times a day

ervee, 2023-12-27:
Maintainer: @tohojo
Environment: aarch64_generic, FriendlyElec NanoPi R2S (rockchip/armv8), OpenWrt 23.05.2

Description:
SQM is stopping/starting 100+ times a day with these messages in the log:
```
Wed Dec 27 19:41:10 2023 user.notice SQM: Stopping SQM on eth0.34
Wed Dec 27 19:41:10 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Wed Dec 27 19:41:10 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Wed Dec 27 19:41:10 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Wed Dec 27 19:41:10 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Wed Dec 27 19:41:10 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
```
I can't find anything useful around these entries in the log. My WAN DHCP renews every 30 minutes but does not match up with these SQM restarts.

My question/report is actually 2-fold:
1. Why does this happen about every 16-17 minutes, but also sometimes after only 2 minutes or 5...
2. What about those ERROR messages?

This is my config:
```
root@nanort:~# cat /etc/config/sqm
config queue 'eth0'
        option enabled '1'
        option interface 'eth0.34'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'
        option debug_logging '0'
        option verbosity '5'
#       option download '68408' # 85% of 80.48 Mbps
#       option upload '73840'   # 85% of 86.87 Mbps
#       option download '72432' # 90% of 80.48 Mbps
#       option upload '78183'   # 90% of 86.87 Mbps
#       option download '76456' # 95% of 80.48 Mbps
#       option upload '82527'   # 95% of 86.87 Mbps
        option download '85000'
        option upload '85000'
        option linklayer 'ethernet'
        option overhead '44'
```
Please let me know what else I can supply or troubleshoot.

brada4, 2023-12-28:
@tohojo may know better. What happens stop removes qdisc then start removes(nothing and errs) and adds it back. In principle should suffice for ip(route2) for last 10 years to just replace root qdisc destroying past hierarchy. You can try replac(ng remove/add to replace in qos script in use.

brada4, 2023-12-28:
why it restarts - check ethtool -S - usually cable or sub-par phy config, normally hotplug scripts get triggered on actual cable unplug or real ip c+ange from dhcp.

ervee, 2023-12-31:
> lac(ng remove/add to replace in qos script

> @tohojo may know better. What happens stop removes qdisc then start removes(nothing and errs) and adds it back. In principle should suffice for ip(route2) for last 10 years to just replace root qdisc destroying past hierarchy. You can try replac(ng remove/add to replace in qos script in use.

There are no ```remove``` statements in piece_of_cake.qos but there are 4 ```add``` statements in there. Can I just replace them all by ```replace```? or specific ones? Is there a specific command output I could check before and after this change to make sure I don't break functionality?

ervee, 2023-12-31:
> why it restarts - check ethtool -S - usually cable or sub-par phy config, normally hotplug scripts get triggered on actual cable unplug or real ip c+ange from dhcp.

The DHCP renews do not line up with the SQM restarts. The DHCP renew also does not change the actual IP.
I don't see any disconnects/connects in the logs.

What am I looking for with ```ethtool -S```? eth0 is the physical WAN port of the NanoPi R2S.
```
root@router:~# ethtool -S eth0
NIC statistics:
     mmc_tx_octetcount_gb: 4225059748
     mmc_tx_framecount_gb: 109989600
     mmc_tx_broadcastframe_g: 0
     mmc_tx_multicastframe_g: 0
     mmc_tx_64_octets_gb: 0
     mmc_tx_65_to_127_octets_gb: 0
     mmc_tx_128_to_255_octets_gb: 0
     mmc_tx_256_to_511_octets_gb: 0
     mmc_tx_512_to_1023_octets_gb: 0
     mmc_tx_1024_to_max_octets_gb: 0
     mmc_tx_unicast_gb: 0
     mmc_tx_multicast_gb: 0
     mmc_tx_broadcast_gb: 0
     mmc_tx_underflow_error: 0
     mmc_tx_singlecol_g: 0
     mmc_tx_multicol_g: 0
     mmc_tx_deferred: 0
     mmc_tx_latecol: 0
     mmc_tx_exesscol: 0
     mmc_tx_carrier_error: 0
     mmc_tx_octetcount_g: 4225059748
     mmc_tx_framecount_g: 109989600
     mmc_tx_excessdef: 0
     mmc_tx_pause_frame: 0
     mmc_tx_vlan_frame_g: 0
     mmc_rx_framecount_gb: 246930023
     mmc_rx_octetcount_gb: 3652580616
     mmc_rx_octetcount_g: 3652580616
     mmc_rx_broadcastframe_g: 0
     mmc_rx_multicastframe_g: 4246
     mmc_rx_crc_error: 0
     mmc_rx_align_error: 0
     mmc_rx_run_error: 0
     mmc_rx_jabber_error: 0
     mmc_rx_undersize_g: 0
     mmc_rx_oversize_g: 0
     mmc_rx_64_octets_gb: 0
     mmc_rx_65_to_127_octets_gb: 0
     mmc_rx_128_to_255_octets_gb: 0
     mmc_rx_256_to_511_octets_gb: 0
     mmc_rx_512_to_1023_octets_gb: 0
     mmc_rx_1024_to_max_octets_gb: 0
     mmc_rx_unicast_g: 0
     mmc_rx_length_error: 0
     mmc_rx_autofrangetype: 0
     mmc_rx_pause_frames: 0
     mmc_rx_fifo_overflow: 0
     mmc_rx_vlan_frames_gb: 0
     mmc_rx_watchdog_error: 0
     mmc_rx_ipc_intr_mask: 3221078013
     mmc_rx_ipc_intr: 0
     mmc_rx_ipv4_gd: 190163515
     mmc_rx_ipv4_hderr: 0
     mmc_rx_ipv4_nopay: 0
     mmc_rx_ipv4_frag: 0
     mmc_rx_ipv4_udsbl: 0
     mmc_rx_ipv4_gd_octets: 0
     mmc_rx_ipv4_hderr_octets: 0
     mmc_rx_ipv4_nopay_octets: 0
     mmc_rx_ipv4_frag_octets: 0
     mmc_rx_ipv4_udsbl_octets: 0
[... 119 lines cut ...]
     mac_tx_frame_ctrl_pause: 0
     mac_gmii_tx_proto_engine: 0
     mtl_rx_fifo_fill_level_full: 0
     mtl_rx_fifo_fill_above_thresh: 0
     mtl_rx_fifo_fill_below_thresh: 0
     mtl_rx_fifo_fill_level_empty: 0
     mtl_rx_fifo_read_ctrl_flush: 0
     mtl_rx_fifo_read_ctrl_read_data: 0
     mtl_rx_fifo_read_ctrl_status: 0
     mtl_rx_fifo_read_ctrl_idle: 0
     mtl_rx_fifo_ctrl_active: 0
     mac_rx_frame_ctrl_fifo: 0
     mac_gmii_rx_proto_engine: 0
     tx_tso_frames: 0
     tx_tso_nfrags: 0
     mtl_est_cgce: 0
     mtl_est_hlbs: 0
     mtl_est_hlbf: 0
     mtl_est_btre: 0
     mtl_est_btrlm: 0
     q0_tx_pkt_n: 109989600
     q0_tx_irq_n: 0
     q0_rx_pkt_n: 246929451
     q0_rx_irq_n: 0
```

brada4, 2023-12-31:
some err counters counted something , but not up/down. 
yes, you can use r#placefor add as it replaces noqueue after deletion too.

tohojo, 2024-01-01:
Using `replace` instead of `add` shouldn't hurt, but I don't expect it'll help either. 

My guess would be that there is some system event that causes the network device to come and go periodically, causing an SQM restart due to hotplug events firing. It looks like the device has already disappeared by the time the sqm stop hook is running, which is what causes the errors.

Overall, this is most likely relatively harmless, so if you're not seeing any impact on network traffic it should be fine to ignore the errors. I don't have a good guess as to why this is happening in the first place, the only suggestion I have is to look for other log messages around the SQM errors, to see if there is anything providing any hints...

ervee, 2024-01-11:
Hi @tohojo,

TL;RD: There is an IPv6 event restarting SQM. But why... Check the last part of this post if you are in a hurry.

Looking in dmesg the last ethx event was 25 days ago the script reloads at least every 5 minutes. I really have no clue what could be causing this.

I changed the 4 "add" statements in /usr/lib/sqm/piece_of_cake.qos to "replace":
```
# diff /usr/lib/sqm/piece_of_cake.qos /usr/lib/sqm/piece_of_cake.qos.org
24c24
<     $TC qdisc replace dev $IFACE root $( get_stab_string ) cake \
---
>     $TC qdisc add dev $IFACE root $( get_stab_string ) cake \
33c33
<     $TC qdisc replace dev $IFACE handle ffff: ingress
---
>     $TC qdisc add dev $IFACE handle ffff: ingress
38c38
<     $TC qdisc replace dev $DEV root $( get_stab_string ) cake \
---
>     $TC qdisc add dev $DEV root $( get_stab_string ) cake \
45c45
<     $TC filter replace dev $IFACE parent ffff: protocol all prio 10 u32 \
---
>     $TC filter add dev $IFACE parent ffff: protocol all prio 10 u32 \
```

After starting sqm this is the logread output:
```
Thu Jan 11 19:31:44 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Thu Jan 11 19:31:45 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
```

And ```tc qdisc``` output:
```
# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc mq 0: dev eth0 root
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth1 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc cake a25d: dev eth0.34 root refcnt 2 bandwidth 85Mbit besteffort triple-isolate nonat nowash no-ack-filter split-gso rtt 100ms noatm overhead 44
qdisc ingress ffff: dev eth0.34 parent ffff:fff1 ----------------
qdisc noqueue 0: dev eth1.2 root refcnt 2
qdisc noqueue 0: dev eth1.3 root refcnt 2
qdisc noqueue 0: dev eth1.5 root refcnt 2
qdisc noqueue 0: dev wg0 root refcnt 2
qdisc cake a25e: dev ifb4eth0.34 root refcnt 2 bandwidth 85Mbit besteffort triple-isolate nonat wash no-ack-filter split-gso rtt 100ms noatm overhead 44
```

After a while I get a stop/start event:
```
Thu Jan 11 19:47:52 2024 user.notice SQM: Stopping SQM on eth0.34
Thu Jan 11 19:47:52 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Thu Jan 11 19:47:53 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully

```

At 19:35 and 20:05 there are WAN DHCP renews, but again, this is a renew of the IP my router already had and it is 12 minutes before and 18 minutes after the SQM restart.

The ```tc qdisc``` output changes after the stop/start:

```
# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2
qdisc mq 0: dev eth0 root
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth1 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5ms interval 100ms memory_limit 4Mb ecn drop_batch 64
qdisc noqueue 0: dev eth0.34 root refcnt 2
qdisc noqueue 0: dev eth1.2 root refcnt 2
qdisc noqueue 0: dev eth1.3 root refcnt 2
[... 7 lines cut ...]
I did some searching on troubleshooting hotplug and found that you can create ```/etc/hotplug.d/iface/00-logger``` and ```/etc/hotplug.d/net/00-logger``` with ```logger -t hotplug $(env)``` in it and then it will log stuff! :)

After a wait I saw this:
```
Thu Jan 11 20:04:27 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Thu Jan 11 20:04:32 2024 user.notice SQM: Stopping SQM on eth0.34
Thu Jan 11 20:04:32 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Thu Jan 11 20:04:32 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Thu Jan 11 20:04:32 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Thu Jan 11 20:04:32 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Thu Jan 11 20:04:32 2024 user.notice hotplug: USER=root ACTION=remove SHLVL=1 HOME=/ SEQNUM=54056 IFINDEX=8812 HOTPLUG_TYPE=net DEVPATH=/devices/virtual/net/ifb4eth0.34 LOGNAME=root DEVICENAME=ifb4eth0.34 TERM=linux SUBSYSTEM=net PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=ifb4eth0.34 PWD=/
Thu Jan 11 20:04:32 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Thu Jan 11 20:04:32 2024 user.notice hotplug: USER=root ACTION=add SHLVL=1 HOME=/ SEQNUM=54057 IFINDEX=8814 HOTPLUG_TYPE=net DEVPATH=/devices/virtual/net/SQM_IFB_a6e82 LOGNAME=root DEVICENAME=SQM_IFB_a6e82 TERM=linux SUBSYSTEM=net PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=SQM_IFB_a6e82 PWD=/
Thu Jan 11 20:04:32 2024 user.notice hotplug: USER=root ACTION=remove SHLVL=1 HOME=/ SEQNUM=54062 IFINDEX=8814 HOTPLUG_TYPE=net DEVPATH=/devices/virtual/net/SQM_IFB_a6e82 LOGNAME=root DEVICENAME=SQM_IFB_a6e82 TERM=linux SUBSYSTEM=net PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=SQM_IFB_a6e82 PWD=/
Thu Jan 11 20:04:33 2024 user.notice hotplug: USER=root ACTION=add SHLVL=1 HOME=/ SEQNUM=54063 IFINDEX=8815 HOTPLUG_TYPE=net DEVPATH=/devices/virtual/net/SQM_IFB_888bb LOGNAME=root DEVICENAME=SQM_IFB_888bb TERM= [... cut ...] RM=linux SUBSYSTEM=net PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=SQM_IFB_2c414 PWD=/
Thu Jan 11 20:04:33 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
```

So there is an ifupdate event on eth0.34 caused by wan6 perhaps? As far as I can tell my IPv6 is rock solid and the prefix never changes (if my IPv4 does not change). And the ERROR messages when SQM restarts are still there after changing to "replace".

Do you have IPv6 and not have this event once in a while?

brada4, 2024-01-11:
Thats ifb device (ex dummy) being recreated (rolling id is interface id, like you see in `ip link` the frontmost number like 1 for loopback)
In principle you can drop events with ^ifb.* as it is qos piece, not genuine interface you police traffic on.
Still what makes it recreating is a question.

tohojo, 2024-01-12:
Andrew ***@***.***> writes:

> Thats ifb device (ex dummy) being recreated (rolling id is interface id, like you see in `ip link` the frontmost number like 1 for loopback)
> In principle you can drop events with ^ifb.* as it is qos piece, not genuine interface you police traffic on.
> Still what makes it recreating is a question.

The ifb devices are created by sqm-scripts for feature probing, so that
part is normal. It's the reason that an sqm restart is triggered in the
first place that is elusive here...

ervee, 2024-01-12:
I kept the hotplug iface logging and added a ```logger -t hotplug $(env)``` line to /etc/hotplug.d/iface/11-sqm (just after the first line).

This is the output of logread now:
```
Fri Jan 12 20:34:23 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:34:23 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:34:27 2024 user.notice SQM: Stopping SQM on eth0.34
Fri Jan 12 20:34:27 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Fri Jan 12 20:34:27 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Fri Jan 12 20:34:27 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Fri Jan 12 20:34:27 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Fri Jan 12 20:34:28 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Fri Jan 12 20:34:28 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
Fri Jan 12 20:35:32 2024 daemon.notice netifd: wan (3449): udhcpc: sending renew to server 217.19.17.85
Fri Jan 12 20:35:32 2024 daemon.notice netifd: wan (3449): udhcpc: lease of 82.197.218.30 obtained from 217.19.17.85, lease time 3600
Fri Jan 12 20:51:03 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:51:03 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:51:08 2024 user.notice SQM: Stopping SQM on eth0.34
Fri Jan 12 20:51:08 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Fri Jan 12 20:51:08 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Fri Jan 12 20:51:08 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Fri Jan 12 20:51:08 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Fri Jan 12 20:51:08 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Fri Jan 12 20:51:09 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
Fri Jan 12 20:58:00 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:58:00 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 20:58:04 2024 user.notice SQM: Stopping SQM on eth0.34
Fri Jan 12 20:58:04 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Fri Jan 12 20:58:04 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Fri Jan 12 20:58:04 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Fri Jan 12 20:58:04 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Fri Jan 12 20:58:05 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Fri Jan 12 20:58:05 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
Fri Jan 12 21:05:32 2024 daemon.notice netifd: wan (3449): udhcpc: sending renew to server 217.19.17.85
Fri Jan 12 21:05:32 2024 daemon.notice netifd: wan (3449): udhcpc: lease of 82.197.218.30 obtained from 217.19.17.85, lease time 3600
Fri Jan 12 21:07:43 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 21:07:43 2024 user.notice hotplug: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Fri Jan 12 21:07:48 2024 user.notice SQM: Stopping SQM on eth0.34
Fri Jan 12 21:07:48 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Fri Jan 12 21:07:48 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Fri Jan 12 21:07:48 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Fri Jan 12 21:07:48 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Fri Jan 12 21:07:48 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Fri Jan 12 21:07:49 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
```

So the SQM restart does seem to be triggered by that hotplug event. The first ```user.notice hotplug``` is from /etc/hotplug.d/iface/00-logger and the second is from the /etc/hotplug.d/iface/11-sqm script. Why it takes 4 to 5 seconds before there is an actual "Stopping SQM" event I don't know.

My IPv6 prefix does not change and things like the firewall do not restart. So can this event be filtered out somehow?

brada4, 2024-01-12:
It is wan6 rtadv which is usually notlogged.

ervee, 2024-01-14:
Amy idea how to exclude this event from the SQM restart?

brada4, 2024-01-14:
Filter in front of script or disable wan6 interface.

tohojo, 2024-01-14:
Ralf ***@***.***> writes:

> I kept the hotplug iface logging and added a ```logger -t hotplug
> $(env)``` line to /etc/hotplug.d/iface/11-sqm (just after the first
> line).

The hotplug script will definitely be executed, but it has logic in it
to only react to ifup and ifdown events. So I don't think this is the
source of the restart (also thinking about the five-second delay).

ervee, 2024-01-14:
> (also thinking about the five-second delay).

I know, but it is darn consistent. I changed the logger commands in ```iface/00-logger``` and ```iface/11-sqm``` to log ```hotplug-iface``` and ```logger-sqm```. Now you can see that the ```iface/00-logger``` fires and ```iface/11-sqm``` fires _**at the same time**_. Then it just takes 4 to 5 seconds for the rest of the ```iface/11-sqm``` script to actually stop/start SQM.

```
Sun Jan 14 15:08:18 2024 user.notice hotplug-iface: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:08:18 2024 user.notice logger-sqm: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:08:23 2024 user.notice SQM: Stopping SQM on eth0.34
Sun Jan 14 15:08:23 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Sun Jan 14 15:08:23 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Sun Jan 14 15:08:23 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Sun Jan 14 15:08:23 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Sun Jan 14 15:08:23 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Sun Jan 14 15:08:24 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully

Sun Jan 14 15:08:40 2024 user.notice hotplug-iface: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:08:40 2024 user.notice logger-sqm: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:08:45 2024 user.notice SQM: Stopping SQM on eth0.34
Sun Jan 14 15:08:45 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Sun Jan 14 15:08:45 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Sun Jan 14 15:08:45 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Sun Jan 14 15:08:45 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Sun Jan 14 15:08:45 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Sun Jan 14 15:08:46 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully

Sun Jan 14 15:17:57 2024 user.notice hotplug-iface: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:17:57 2024 user.notice logger-sqm: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:18:02 2024 user.notice SQM: Stopping SQM on eth0.34
Sun Jan 14 15:18:02 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Sun Jan 14 15:18:02 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Sun Jan 14 15:18:02 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Sun Jan 14 15:18:02 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Sun Jan 14 15:18:02 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Sun Jan 14 15:18:03 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully

Sun Jan 14 15:22:33 2024 user.notice hotplug-iface: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:22:33 2024 user.notice logger-sqm: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sun Jan 14 15:22:38 2024 user.notice SQM: Stopping SQM on eth0.34
Sun Jan 14 15:22:38 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Sun Jan 14 15:22:38 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Sun Jan 14 15:22:38 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Sun Jan 14 15:22:38 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Sun Jan 14 15:22:38 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Sun Jan 14 15:22:39 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully

[... cut ...]
Sun Jan 14 15:25:26 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
```

> Filter in front of script or disable wan6 interface.

I'm not going to disable wan6.
I added something in 11-sqm just after restart_sqm() {
```
    [ "$ACTION" = "ifupdate" -a "$INTERFACE" = "wan6" ] && {
        logger -t logger-sqm ifupdate on wan6
        exit 0
    }
```
This is pretty ugly but let's see what it does. I could also add this to the beginning of the script. That would be better perhaps.

ervee, 2024-01-14:
> I added something in 11-sqm just after restart_sqm() {
> 
> ```
>     [ "$ACTION" = "ifupdate" -a "$INTERFACE" = "wan6" ] && {
>         logger -t logger-sqm ifupdate on wan6
>         exit 0
>     }
> ```

Ignore that. It did not work :)

ervee, 2024-01-27:
I have added logger configuration now to the start of the ```hotplug.d/iface/11-sqm script``` created a new part in the script:
```
[ "$ACTION" = "ifupdate" -a "$INTERFACE" = "wan6" ] && {
    logger -t logger-sqm ifupdate on wan6
    exit 0
}
```
And added logger lines to the ```restart_sqm``` part and the ```"$ACTION" = ifup``` and the ```"$ACTION" = ifdown``` parts. Just to see what gets triggered.

The resulting logs:
```
Sat Jan 27 16:28:32 2024 user.notice logger-sqm: USER=root ACTION=ifupdate SHLVL=1 HOME=/ HOTPLUG_TYPE=iface IFUPDATE_PREFIXES=1 LOGNAME=root DEVICENAME= TERM=linux PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan6 PWD=/ DEVICE=eth0.34
Sat Jan 27 16:28:32 2024 user.notice logger-sqm: ifupdate on wan6
Sat Jan 27 16:28:37 2024 user.notice SQM: Stopping SQM on eth0.34
Sat Jan 27 16:28:37 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 ingress
Sat Jan 27 16:28:37 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
Sat Jan 27 16:28:37 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev eth0.34 root
Sat Jan 27 16:28:37 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Sat Jan 27 16:28:37 2024 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.34, in: 85000 Kbps, out: 85000 Kbps
Sat Jan 27 16:28:38 2024 user.notice SQM: piece_of_cake.qos was started on eth0.34 successfully
```

The first ```user.notice logger-sqm:``` is at the start of the script. So the ```hotplug.d/iface/11-sqm script``` script is triggered by something every now and then, but we already knew that.

The second ```user.notice logger-sqm: ifupdate on wan6``` is in the part that tests for ACTION" = "ifupdate" and "$INTERFACE" = "wan6" but we also kind of knew that. I just wanted to see if I could trigger it that way. But as you can see above I have added an ```exit 0``` after the logger line. This _should_ exit the script is it not? And if it does, 4-5 seconds later there is still a stop/start of SQM. Without any of the other loggers logging anything. @tohojo noticed this behavior earlier and it could suggest that the stop/start is not directly triggered by the hotplug event. But it must be triggered indirectly somehow because it is without exception always triggered 5 to 6 seconds after the hotplug event.

Udayanithig, 2024-06-03:
I'm not very technical, but I'm using a NanoPi R2S. I have a similar problem where, when the PPPoE connection goes down and comes back up, the SQM log shows some failures.

`Mon Jun  3 15:42:11 2024 daemon.notice netifd: Interface 'wan' is now up
Mon Jun  3 15:42:11 2024 daemon.notice netifd: Interface 'wan_6' is enabled
Mon Jun  3 15:42:11 2024 daemon.notice netifd: Interface 'wan_6' has link connectivity
Mon Jun  3 15:42:11 2024 daemon.notice netifd: Interface 'wan_6' is setting up now
Mon Jun  3 15:42:11 2024 user.notice SQM: Stopping SQM on pppoe-wan
Mon Jun  3 15:42:11 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev pppoe-wan ingress
Mon Jun  3 15:42:11 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot find specified qdisc on specified device.
Mon Jun  3 15:42:11 2024 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /sbin/tc qdisc del dev pppoe-wan root
Mon Jun  3 15:42:11 2024 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.`

Also, it increased the buffer bloat. I need to manually restart the SQM service to get it working properly again